gitbls/sdm #125: How do i fail a build from a plugin?

justin-robinson, 2023-11-06:
Is there a way to get sdm to fail from a plugin? I’m doing everything I can think of and it’s still just chugging along assuming the toucan plugin succeeded. I’m explicitly writing to stderr and exiting with a non 0 exit code

```
if [ $? -ne 0 ]; then >&2 echo "Failed to get claim.pem.crt from AWS" && exit 1; fi
```

#### sdm output
```
Failed to get claim.pem.crt from AWS
* Phase 0 Completed
* Enter image '../2023-05-03-raspios-bullseye-arm64-lite.img' for Phase 1
```

This is problematic because sdm will build an image that doesn't have what we require. Worse, it's burying the error message in the massive amount of output.

wreedamz, 2023-11-06:
Similarly it would be nice to be able to have the sdm execution fail if the image runs out of space or apt fails to install a package. If i'm not staring at stdout or looking through the logs afterwards, it's incredibly easy to miss these errors.

gitbls, 2023-11-06:
@justin-robinson No, there is currently no way to fail from within a plugin. If a plugin has an error, things just keep chugging along.

I originally thought that it was best to let things go, because that would expose the most errors in a run. But I can understand that in certain environments that may not be desirable.

I can think of at least two possible solutions. What are your thoughts on these?
* Change your failure path to provide a better, perhaps coded error message. By "coded" I mean something that you can grep the log for (e.g, FAIL or AWS-KEY-FAIL, or whatever) to determine if there are any problems.

  It's possible that some things currently characterized as warnings should be errors, but I think I've been careful about this.
* Implement a way for a plugin to signal to the upper levels of sdm that something has gone seriously awry, and have the upper levels notice this and stop sdm. The signal could either be a real signal (invoked via a function that the plugin calls) or a flag file in /etc/sdm.
  * A real signal is perhaps cleaner, but would likely be much more definitive, in that the trap handler wouldn't do anything besides clean up (unmount disk, delete loop device), and exit. The code raising the signal could also provide some text to be output immediately before exiting.
  * Flag file existence would indicate an error and cause sdm to cease processing. This check would likely only be done in the loops that run the plugins in phase0, phase1, and post-install. The contents of the file could include additional information to be displayed as sdm stops processing and exits.

I'm not wed to either of those, and open to other ideas, as long as they aren't too complex/costly.

Also, in case you weren't aware, messages output by sdm and plugins are preceded with a single character:
  * `*` Start of a new 'section' (e.g., `* Start Phase 1 image customization`)
  * `>` Informing you of something that sdm is doing
  * `%` Warning you that something isn't quite right, but forging ahead
  * `?` An error was identified
This idea was that you grep the log for '%' and '?' to decide if things worked or not. I use this all the time.

@wreedamz are you aware that the end of the log can have information in it such as
```
%%%
% IMG '2023-10-10-raspios-bookworm-arm64-lite.img' is 100% full (0 1K-blocks free)
% Review /etc/sdm/apt.log in the image for insufficient disk space errors
% If needed use --extend --xmb nnnn with --customize to create more space
%%%
? apt reported errors; review /etc/sdm/apt.log
* Enter Shell Command Prompt
  'exit' to exit back to host system
...
sdm~# 
```
As an aside, sdm checks for apt errors by 
```
    grep -q "^E:" /etc/sdm/apt.log && logtoboth "? apt reported errors; review /etc/sdm/apt.log"
```
which you can obviously do yourself as well. If this information is not sufficient for you, what would you propose?

gitbls, 2023-11-07:
After further thought on this, I think the best way is to implement this via real signals. There are still a few nuances to work through, so premature to predict when this will pop out, but I think it's important so it is at the top of the ToDo list.

gitbls, 2023-11-09:
Ignore previous comment. Here's sketch of what I'm thinking. I'd appreciate your thoughts on it. 

* New functions for plugins to use:
   * Take a note: Write a line of text to a file that is displayed at the end of a customization.
   * Abort: print provided text and abort the customization

sdm will check for an abort request every time a plugin returns from being called for a phase. It will also abort via this mechanism if out of disk space or if apt logs a message starting with 'E:' in /etc/sdm/apt.log

sdm will print the contents of the take a note file at the end of the customization, if it exists. Content and format are up to you.

Will this work for you?  Thx

justin-robinson, 2023-11-09:
I really appreciate the thought you put into this. I think your idea with "real signals" was the best. It's the expected behavior from a bash script and requires less documentation from you and learning from us. Principle of least surprise would support failing if a plugin exiting with a non zero exit code.

justin-robinson, 2023-11-09:
I also understand this is a breaking change so gating this with a cli flag or env variable would be best

gitbls, 2023-11-09:
> I really appreciate the thought you put into this. I think your idea with "real signals" was the best. It's the expected behavior from a bash script and requires less documentation from you and [...]

I agree with you, and that was my first inclination. Until I thought about how it would work, and it won't.

When sdm is running phase 1 and post-install code it's running in a systemd-nspawn container so the plugin can't signal the sdm code running in the parent.

justin-robinson, 2023-11-09:
I understand and that sounds like a hard stop. If you think that's the best way then I'l defer to your judgement.

gitbls, 2023-11-09:
> I understand and that sounds like a hard stop. If you think that's the best way then I'l defer to your judgement.

I wish it wasn't, but as often happens in software development it's trading off "purity of programming" vs "keeping focused".

It will seem natural for people writing sdm plugins (using one new call), and it will be definitive, in that calling *abort* will not return, keeping the plugin code simple.

And lastly, it solves your problem sooner 😎

gitbls, 2023-11-13:
> I understand and that sounds like a hard stop. If you think that's the best way then I'l defer to your judgement.

I reconsidered this during my short San Juans trip and am now moving down the path to properly handle failure codes from plugins. This is not "real signals"; those are a complexity I'd prefer to avoid for now.

Rather, if a plugin exits any of the phases with a non-zero status the sdm customization will be terminated.

gitbls, 2023-11-18:
My main concern with honoring plugin return status was that the sdm world as we know it would be broken. That was not the case (none of the sdm-provided plugins broke), so this will be in the next release which will likely arrive the week after Thanksgiving.

So much for my "hard stop" 🤣

You may find this new command line switch useful as well: The `--plugilist` switch. It takes a list of files (e.g., `--pluglist file1,file2,file3`), and can also be invoked on the command line multiple times. Each file is a list of plugins. For example:
```
user:deluser=pi
user:adduser=bls
system:fstab=fstab.extension|eeprom=stable|sysctl=/path/to/some-sysctl-config.conf
network:nmconn=/path/to/eth0.nmconnection,/path/to/wlan0.nmconnection
trim-enable
quiet:consoleblank=300|noquiet|nosplash
vnc:tigervnc=2540x1350,1880x960,1700x1200,1880x1100
apps:name=myapps|apps=@myappslist
```

gitbls, 2023-11-29:
This is complete in V10.0. Please give it a go and LMK of any issues. Thx

justin-robinson, 2023-11-30:
Thank you! It'll be a couple months before we touch this again but I'll reply back when I have feedback.

gitbls, 2023-11-30:
> Thank you! It'll be a couple months before we touch this again but I'll reply back when I have feedback.

Thanks for inspiring this change. Looking forward to hearing how it works for you when you are back at it.

gitbls, 2023-12-05:
Closing this issue. if/when you get around to testing it, please feel free to re-open this issue or start a new one for any problems you run into. Thx

wreedamz, 2024-03-05:
I'm getting around to testing this now and I'm having some issues.

1. If a plugin fails with an error, the error code that it fails with is not logged. Perhaps `Plugin '<name>' exited with failure` could be modified to include this
2. If an app installation fails with `apt`, `sdm` execution exits successfully rather than with an error code

wreedamz, 2024-03-05:
Also looks like I don't have permission to re-open this issue @gitbls, let me know if you would like me to start a new issue

gitbls, 2024-03-05:
> I'm getting around to testing this now and I'm having some issues.
>
> 1. If a plugin fails with an error, the error code that it fails with is not logged. Perhaps `Plugin '<name>' exited with failure` [...]

1) is reasonable. Will look into it for next release
2) I don't understand. Exactly how is apt invoked in this failing scenario? Is it in your plugin or something else?

gitbls, 2024-03-05:
> Also looks like I don't have permission to re-open this issue @gitbls, let me know if you would like me to start a new issue

should be open now

wreedamz, 2024-03-05:
In the scenario I was testing, one of the apps specified no longer exists in a particular repository. You can reproduce this issue with something like `--plugin apps:"apps=fakeappdoesntexist"`

gitbls, 2024-03-05:
Ah, so the failure is that the `apps` plugin doesn't check status. Yes, indeed, it doesn't now, but it will for the next release, which I'm planning for the week of March 18th due to some impending out-of-pocketness.

gitbls, 2024-03-05:
> In the scenario I was testing, one of the apps specified no longer exists in a particular repository. You can reproduce this issue with something like `--plugin apps:"apps=fakeappdoesntexist"`

Doing:
```
 sdm --customize --poptions noupdate,noupgrade,noautoremove --plugin apps:apps=foobaz virgin-image.img
```
on 11.4 I get:
```
* Start [default] Application Installs
> [default] Application Installs: foobaz
? apt returned an error; review /etc/sdm/apt.log
? Plugin 'apps' exited with failure
umount: /mnt/sdm/boot/firmware unmounted
umount: /mnt/sdm unmounted
```
So I'm a little confused. Is the scenario more complex than this?

 In spite of my confusion, I've corrected the involved code to correctly handle status, as well as providing the status code per your suggestion:
```
* Start [default] Application Installs
> [default] Application Installs: foobaz
? apt returned an error; review /etc/sdm/apt.log
? Plugin 'apps' exited with failure status '1'
umount: /mnt/sdm/boot/firmware unmounted
umount: /mnt/sdm unmounted
```
Thx

gitbls, 2024-03-06:
> > In the scenario I was testing, one of the apps specified no longer exists in a particular repository. You can reproduce this issue with something like `--plugin apps:"apps=fakeappdoesntexist"`
> 
> For this issue, the final return/exit code of SDM was not being set correctly. The error messages were printed as appropriate and the final return code was 0 despite the error. I'm not sure if that's been corrected in the console output you shared

Ah! Thanks for the clarification. Will address.

gitbls, 2024-03-22:
Fixed in V11.6. Please test/verify. Thx

gitbls, 2024-04-22:
Closing due to no activity. Please re-open or start a new issue if required.

closed 2024-04-22 by gitbls