gitbls/sdm #126: Copyfile fails to set ownership when group is given

stb68, 2023-11-11:
Hi

There seems to be an issue with the copyfile plugin when specifying owner and group.

If I specify both, e.g. 
`   --plugin copyfile:"from=/home/stb/Desktop/Pi/Shared_Scripts_etc/.config/nvim/settings.vim|to=/etc/skel/.config/nvim|chown=root:root|chmod=644|mkdirif"`
then the ownership / group don't get set.  If I only give the owner, then that bit works.

Seems that parsing of the options isn't working right and it's messing up on the colon (:)...
```
cp: cannot stat 'copyfile:from=/home/stb/Desktop/Pi/Shared_Scripts_etc/.config/nvim/settings.vim|to=/etc/skel/.config/nvim|chown=root': No such file or directory
% Update (or add) Plugin 
  'copyfile:from=/home/stb/Desktop/Pi/Shared_Scripts_etc/.config/nvim/settings.vim|to=/etc/skel/.c
  onfig/nvim|**chown=root'** to /usr/local/sdm/local-plugins
```
and later
```
> Run Plugin 'copyfile' (/mnt/sdm/usr/local/sdm/plugins/copyfile) Phase 0 with arguments: 
  'from=/home/stb/Desktop/Pi/Shared_Scripts_etc/.config/nvim/settings.vim|to=/etc/skel/.config/nvi
  m|chown=root:root|chmod=644|mkdirif'
* Plugin copyfile: Start Phase 0
> Plugin copyfile: Keys/values found:
   from: /home/stb/Desktop/Pi/Shared_Scripts_etc/.config/nvim/settings.vim
   to: /etc/skel/.config/nvim
   chown: **root=root**
   chmod: 644
   mkdirif: 
> Plugin copyfile: Create directory /mnt/sdm/etc/sdm/assets/copyfile
> Plugin copyfile: Add '/home/stb/Desktop/Pi/Shared_Scripts_etc/.config/nvim/settings.vim' to 
  files to copy
> Plugin copyfile: Directory '/etc/skel/.config/nvim' will be created
* Plugin copyfile: Complete Phase 0
```
and later
```
> Plugin copyfile: Copy 'settings.vim' from /etc/sdm/assets/copyfile to '/etc/skel/.config/nvim'
/usr/bin/chown: invalid user: '**root=root**'
```

Then the same for Phase 1.

gitbls, 2023-11-11:
I'll take a look at it when I get back to my desk in a day or two, but try user.group instead of using a colon. Crown may warn about it but it's ignorable

gitbls, 2023-11-13:
This will be fixed in the next release (V9.7).

gitbls, 2023-11-29:
Fixed in V10.0. Please verify. Thx

gitbls, 2023-12-05:
OK, with the additional information you provided offline, I found the problematic code path that's only executed when plugins are run during `--burn`. That will be fixed in V10.2.

gitbls, 2024-02-04:
Closing due to no activity

closed 2024-02-04 by gitbls