gitbls/sdm #127: Why entering the shell at the end of IMG customization

huaminghuangtw, 2023-11-11:
First of all, thanks for this handy tool!

I just wonder why it enters the shell at the end of `sudo sdm --customize myimg.img`? _Is it possible to avoid this behavior?_ It is because that I don't want any user input during the customization/burning process such that it's 100% automated. :-)
 
```
> IMG '2023-10-10-raspios-bookworm-arm64-lite.img' has 282980 1K-blocks (0.3GB, 0.3GiB) free at end of image customization
* Enter Shell Command Prompt
  'exit' to exit back to host system
root@sdm:/# \exit
```

gitbls, 2023-11-11:
The --batch switch will do what you want.

gitbls, 2023-11-17:
I'm assuming that the --batch switch gets you what you want, so closing this.