gitbls/sdm #128: Setting the LED flashing still possible

huaminghuangtw, 2023-11-12:


huaminghuangtw, 2023-11-12:
Is [setting the LED flashing](https://github.com/gitbls/sdm/blob/master/Docs/LED-flashing.md#led-flashing) still possible for the latest version of sdm? It seems it's [deprecated](https://github.com/gitbls/sdm/blob/07dde34f804251b1cc0098afac6e5032dd0f2135/sdm#L407), but not updated in the [documentation](https://github.com/gitbls/sdm/blob/master/Docs/LED-flashing.md#led-flashing).

gitbls, 2023-11-13:
The LED flashing is taking a break. When sdm was restructured into plugins, LED flashing did not get done. It's removed at the moment. You're the first person to ask about this, so there clearly isn't much demand for it.

Also, the LED flashing that was formerly implemented had nothing to do with OS customization. It was used at boot time to indicate progress/status while it was looking for configuration information on a USB disk.

If you're interested in this feature, please detail what you're looking for.

Thx

huaminghuangtw, 2023-11-13:
@gitbls I am just trying to make use of it when I saw this feature in the documentation. Now I understand your point. Thank you. :-)

stb68, 2023-11-13:
Yes, same here, it was in the documentation so I tried to use it in my initial attempts, but removed the option from my call when it didn't work. I had enough other bits to try at the time to worry about it.

I wasn't sure how useful it would be, just added it to see so I'm not missing it.

gitbls, 2023-12-05:
Closing this issue due to lack of activity.

closed 2023-12-05 by gitbls