gitbls/sdm #130: Best way to check if sdm is already installed

huaminghuangtw, 2023-11-13:
I would like to ask for advice on the best practice to check if sdm is already installed.

Right right, I am using the following code snippet:
```
### Install sdm (A Raspberry Pi SD Card Image Manager) ###
# Check if sdm is already installed
if [ ! -e "/usr/local/sdm" ]; then
    # Download official Raspberry Pi OS image file
    echo "========================================================"
    echo "Installing sdm - A Raspberry Pi SD Card Image Manager..."
    echo "========================================================"

    curl -L https://raw.githubusercontent.com/gitbls/sdm/master/EZsdmInstaller | bash
fi
```

But the problem is that it seems taking a considerably long time to finish the checking with `if [ ! -e "/usr/local/sdm" ]; then`, which I still had no clue why it is the case...

Do you have any suggestions how I can improve this to check if sdm is already installed?

Thank you in advance!

gitbls, 2023-11-13:
How did you determine that the test was taking a long time to complete the check?

1stcall, 2023-11-13:
Have you considered running the install script regardless of whether it is installed? This would have the benefit of ensuring you are using the latest version.

huaminghuangtw, 2023-11-14:
> How did you determine that the test was taking a long time to complete the check?

I figured it out... It was because of the line before this code snippet, which copies a large size of `*.img` file. Thanks for pointing me out!

huaminghuangtw, 2023-11-14:
> Have you considered running the install script regardless of whether it is installed? This would have the benefit of ensuring you are using the latest version.

Good point! Would it be better to only install sdm, only when:

- It is not installed yet by checking whether `/usr/local/sdm"` exists or not
- It is installed, but not the latest version

For the second scenario, I am not sure if there is an easy way to check it.

gitbls, 2023-11-14:
> > How did you determine that the test was taking a long time to complete the check?
> 
> I figured it out... It was because of the line before this code snippet, which copies a large size of `*.img` file. Thanks for pointing me out!

If you want to check for existence of a directory, you should use [ -d ... ] (is a directory) rather than [ -e ... ] (is a file).

You didn't really provide a whole scenario of what you're trying to do. If it's for a system that a) was not created using sdm and b) doesn't have sdm installed, you should, as @1stcall suggested, just install it.

If you want to spend time determining whether you have the latest version, you can easily get the current installed version with `sudo sdm --version`. 

I have no clue how to determine programmatically what the latest version in the sdm github is, but I will point out that I ensure that the CHANGELOG.md file comment is always the version of the "latest" version.

1stcall, 2023-11-14:
It's a bit hacky but the following seems to work.
`#/usr/bin/env bash

which sdm
if [ $? -ne 0 ]; then
    echo "sdm is not installed"
else
    sdmInstalledVer=$(cat $(which sdm) | bash -s -- --version 2>/dev/null)
    sdmLatestVer=$(curl -fsSL https://raw.githubusercontent.com/gitbls/sdm/master/sdm | bash -s -- --version 2>/dev/null)

    if [ "$sdmInstalledVer" \< "$sdmLatestVer" ]; then
        echo "sdm $sdmInstalledVer requires updating. $sdmLatestVer is available"
    else
        echo "sdm $sdmInstalledVer is up to date"
    fi
fi
`

1stcall, 2023-11-15:
`which sdm
if [ $? -ne 0 ]; then
    echo "sdm is not installed"
else
    sdmInstalledVer=$(cat $(which sdm) | bash -s -- --version 2>/dev/null)
    sdmLatestVer=$(curl -fsSL https://raw.githubusercontent.com/gitbls/sdm/master/sdm | bash -s -- --version 2>/dev/null)

    if [ "$sdmInstalledVer" \< "$sdmLatestVer" ]; then
        echo "sdm $sdmInstalledVer requires updating. $sdmLatestVer is available"
    else
        echo "sdm $sdmInstalledVer is up to date"
    fi
fi
`

1stcall, 2023-11-15:
It's a bit hacky but the following seems to work :-

```
        #/usr/bin/env bash

        which sdm
        if [ $? -ne 0 ]; then
            echo "sdm is not installed"
       else
            sdmInstalledVer=$(cat $(which sdm) | bash -s -- --version 2>/dev/null)
            sdmLatestVer=$(curl -fsSL https://raw.githubusercontent.com/gitbls/sdm/master/sdm | bash -s -- --version 2>/dev/null)

            if [ "$sdmInstalledVer" \< "$sdmLatestVer" ]; then
                echo "sdm $sdmInstalledVer requires updating. $sdmLatestVer is available"
            else
                echo "sdm $sdmInstalledVer is up to date"
            fi
        fi
```

gitbls, 2023-11-15:
> It's a bit hacky but the following seems to work :-

Understatement of the year 🤣...but ingenious! 👍

1stcall, 2023-11-15:
Sometimes you just need to use bash's particularities to your advantage LOL!

gitbls, 2024-02-08:
Closing due to lack of activity.

closed 2024-02-08 by gitbls