gitbls/sdm #131: Ownership of directories created via copyfile 'mkdirif' option

stb68, 2023-11-13:
When directories are created via the copyfile 'mkdirif' option, I think it would be useful if you could also specify that the directory ownerships should be changed to match the "chown=" option for the file.

I was initially thinking this should just be the way it works, but maybe there are circumstances where they should remain owned by root?

Maybe an alternative option to consider would be a 'create directory' plugin, so that it's possible to create a directory without putting anything in it.  If that also had the 'chown=' option that could be used before the copy and would provide additional functionality.

gitbls, 2023-11-13:
Ha! An exact note from my TODO list: "mkdir plugin'. Obviously I haven't done it yet but it's percolating up the list...

There is a fine line between too many plugins (separate mkdir plugin) vs fat plugins (too many options on a plugin). If you have any thoughts on this, I'm listening 👍

stb68, 2023-11-13:
> Ha! An exact note from my TODO list:

Well I'm glad it hasn't added yet another task for you :-)   I hesitate before adding these things and debated this one for a while!

> There is a fine line between too many plugins (separate mkdir plugin) vs fat plugins (too many options on a plugin).

Yes, it's a tricky balance.  Depends on the case and what direction things might go in the future.

In this case I think the separate 'mkdir' plugin would be good to have.  Though I also think having to create the directories separately to the copy command, just to get the ownership set, would seem a little bit unnecessary and long winded.  But not badly so.

Maybe you already have the idea of being able to copy a whole directory on your list?  I would go with that as part of 'copyfile' rather than a new plugin, and it wouldn't even need to have a separate option to specify - just have the "from" directory path end in a slash.  (Of course, it would need to work recursively!)

1stcall, 2023-11-13:
Just to add to the confusion. I use an "extract tar" custom plugin that creates directories and files with the correct owners and permissions from a tar file.

stb68, 2023-11-14:
That's a nice idea - and give the plugin an option of extracting as root (to preserve original ownership) or as a specific user (which I believe will change ownership to that user).

Would need to run after user creation though, so maybe that would be on firstboot?  (I'm not at all sure on the ordering of these things, when is user creation in regard to everything else - do plugins run in a given order or the order specified in arguments?  Maybe that isn't in the docs or maybe I've just missed it so far.)

1stcall, 2023-11-14:
Plugins are run in the order they are specified on the command line, so as long as it is specified after the user plugin, you can assume the user and home directory are created.

gitbls, 2023-11-14:
@1stcall is correct on both points. The run order is documented in the Plugins doc, as is the question of how to deal with user's home directories:
````

stb68, 2023-11-14:
Sounds good, many thanks both

gitbls, 2023-11-17:
> Maybe you already have the idea of being able to copy a whole directory on your list? I would go with that as part of 'copyfile' rather than a new plugin, and it wouldn't even need to have a [...]

I'm thinking a separate plugin, since the operations are fairly different. Here's my thinking on a copydir plugin. Appreciate your thoughts on it:

Copydir uses rsync, so standard rsync features, e.g., Trailing / at the end of the ‘from’ means copy the contents of the source, just like rsync.

* from: /path/to/source directory
* to: /path/to/dest directory
* nodirect: Don’t copy the stuff directly to the destination. Instead, “buffer” it in /etc/sdm/assets/copydir. Default is to copy direct to destination. nodirect is useful when the destination is created by something during Phase1
* rsyncopts: String of additional rsync options. Copydir uses “-a”, pretty much the bare minimum
* Tee: ? A separate file where the verbose log can be written. Not sure if this is useful or not

stb68, 2023-11-17:
rsync sounds good especially as it presumably will allow setting ownership however required (root who's running it, preserve from source or set via chown argument).

Not sure how much need there is for the Tee option, but it sounds like a simple add without any downside or extra complexity that's likely to cause issues, and somebody might find it useful.

gitbls, 2023-11-18:
The next release (coming AFTER US Thanksgiving holiday) will have a couple of changes relevant to this:
* `mkdir` plugin
* `--pluglist` command line switch

The `mkdir` plugin takes 3 arguments: `dir`, `chown`, `chmod`. The specified directory is created in Phase 0, so it's available early in customization. chown and chmod are applied in the post-install phase, since chown could rely on a user not created until Phase 1

The `--plugilist` switch is my favorite. It takes a list of files (e.g., `--pluglist file1,file2,file3`), and can also be invoked on the command line multiple times. Each file is a list of plugins. For example:
```
user:deluser=pi
user:adduser=bls
system:fstab=fstab.extension|eeprom=stable|sysctl=/path/to/some-sysctl-config.conf
network:nmconn=/path/to/eth0.nmconnection,/path/to/wlan0.nmconnection
trim-enable
quiet:consoleblank=300|noquiet|nosplash
vnc:tigervnc=2540x1350,1880x960,1700x1200,1880x1100
apps:name=myapps|apps=@myappslist
```

gitbls, 2023-12-05:
Closing this issue as resolved. Please reopen if further problems arise related to this. Thx

closed 2023-12-05 by gitbls