gitbls/sdm #132: Show stdout and stderr on the terminal with runatboot plugin

huaminghuangtw, 2023-11-15:
I am using the [runatboot](https://github.com/gitbls/sdm/blob/master/Docs/Plugins.md#runatboot) plugin to execute a custom script at First Boot.

Right now the stdout and stderr are both redirect to `/dev/null` to suppress any output or error on the terminal screen, with an option to specify a file path to log stdout and stderr to. For example:
```
# Run a custom script during the First Boot of the system 
# stdout and stderr will be written into a log file and to the console
--plugin runatboot:"script=./init_setup.sh|output=var/log/init_setup.log"
```

I was wondering if it is possible to _print messages to the terminal during the First Boot_ as well (while writing to a log file)?

I tried the following, but it still only wrote messages to `/var/log/init_setup.log`, but not printed messages to the console:
```
# Run a custom script during the First Boot of the system 
# stdout and stderr will be written into a log file and to the console
--plugin runatboot:"script=./init_setup.sh|output= >(tee "/var/log/init_setup.log")"
```

gitbls, 2023-11-15:
I don't know if this will work, but try redirecting stdout on your tee command? As in:
```
--plugin runatboot:"script=./init_setup.sh|output= >(tee "/var/log/init_setup.log" > /dev/console)"
```

gitbls, 2023-11-17:
Another thought. In your script, just write the stuff you want to write to /dev/console. That's what sdm-firstboot does. I can't imagine you'd want a huge amount of spew on the console.

huaminghuangtw, 2023-11-17:
@gitbls But what I want to achieve is to write `stdout` and `stderr` _to a log file_ as well as _to the console_.

gitbls, 2023-12-05:
Closing this issue since the OP hasn't provided any more concrete information or example about this.

OP, if you'd like to pursue further please provide sufficient additional information to help me understand what you're asking, how it would be used, and why the solution I offered was not acceptable for your use case.

closed 2023-12-05 by gitbls