gitbls/sdm #133: Version 9.6 issue/question

dwatrous, 2023-11-15:
I just updated to the latest sdm and tried to build an image using bookworm. It failed. Here's the output.

```
daniel@DESKTOP-LM9QIPJ:/mnt/c/Users/Daniel/Documents/mosquito-controller/MosquitoMax/device_controller$ sudo sdm --customize --plugin raspiconfig:"i2c=0" --plugin copyfile:"filelist=sdmfilelist" --plugin disables:piwiz --plugin L10n:host --plugin user:"adduser=mm|password=secret" --plugin user:"deluser=pi" --plugin network:"netman=nm|wifissid='Wishy Washy'|wifipassword='not to late to fate'|wificountry=US|noipv6" --plugin apps:"apps=python3-pip" --plugin mmsetup --extend --xmb 500 --restart --host mmdwatrous 2023-10-10-raspios-bookworm-armhf-lite.img
* Mount IMG '2023-10-10-raspios-bookworm-armhf-lite.img'
mount: /dev/loop0p2 mounted on /mnt/sdm.
mount: /dev/loop0p1 mounted on /mnt/sdm/boot.
* Unmount IMG for --extend
umount: /mnt/sdm/boot unmounted
umount: /mnt/sdm unmounted
* Extend IMG by 500MB (0.5GB, 0.5GiB)...
500+0 records in
500+0 records out
524288000 bytes (524 MB, 500 MiB) copied, 0.713693 s, 735 MB/s
* Resize partition 2 of '2023-10-10-raspios-bookworm-armhf-lite.img' to 2504MB
* Remount IMG to resize the file system
* Mount IMG '2023-10-10-raspios-bookworm-armhf-lite.img'
mount: /dev/loop0p2 mounted on /mnt/sdm.
mount: /dev/loop0p1 mounted on /mnt/sdm/boot.
* Resize the file system
% (Ignore 'on-line resizing required' message)
resize2fs 1.47.0 (5-Feb-2023)
Filesystem at /dev/loop0p2 is mounted on /mnt/sdm; on-line resizing required
old_desc_blocks = 1, new_desc_blocks = 1
The filesystem on /dev/loop0p2 is now 611312 (4k) blocks long.

* Start Configuration
> Command Line: /usr/local/bin/sdm --customize --plugin raspiconfig:i2c=0 --plugin
  copyfile:filelist=sdmfilelist --plugin disables:piwiz --plugin L10n:host --plugin
  user:adduser=mm|password=secret --plugin user:deluser=pi --plugin
  network:netman=nm|wifissid='Wishy Washy'|wifipassword='not to late to
  fate'|wificountry=US|noipv6 --plugin apps:apps=python3-pip --plugin mmsetup --extend --xmb 500
  --restart --host mmdwatrous 2023-10-10-raspios-bookworm-armhf-lite.img
* Host Information
   Hostname: DESKTOP-LM9QIPJ
   uname: Linux DESKTOP-LM9QIPJ 5.15.133.1-microsoft-standard-WSL2 #1 SMP Thu Oct 5 21:02:42
  UTC 2023 x86_64 GNU/Linux
   os-release Name: "Debian GNU/Linux"
   os-release Version: "12 (bookworm)"
   sdm Version: V9.6
* IMG Information
   Name: 2023-10-10-raspios-bookworm-armhf-lite.img
   Date: 2023-10-10
   RasPiOS Version: 12
   RasPiOS Architecture: 32-bit armhf
   os-release Version: "12 (bookworm)"
% sdm will use systemd-nspawn on this 'x86-64' host
  Retry the command with --chroot if this fails
  sdm will run more slowly due to qemu emulation
* Plugins included:
   * raspiconfig
       Args: i2c=0
   * copyfile
       Args: filelist=sdmfilelist
   * disables
       Args: piwiz
   * L10n
       Args: host
   * user
       Args: adduser=mm|password=secret
   * user
       Args: deluser=pi
   * network
       Args: netman=nm|wifissid='Wishy Washy'|wifipassword='not to late to fate'|wificountry=US|noipv6
   * apps
       Args: apps=python3-pip
   * mmsetup
> IMG '2023-10-10-raspios-bookworm-armhf-lite.img' has 787004 1K-blocks (0.8GB, 0.8GiB) free
> Copy sdm to /usr/local/sdm in the IMG
* Start Phase 0 image customization
> Run Plugins Phase '0'
> Run Plugin 'raspiconfig' (/mnt/sdm/usr/local/sdm/plugins/raspiconfig) Phase 0 with arguments:
  'i2c=0'
* Plugin raspiconfig: Start Phase 0
* Plugin raspiconfig: Complete Phase 0
> Run Plugin 'copyfile' (/mnt/sdm/usr/local/sdm/plugins/copyfile) Phase 0 with arguments:
  'filelist=sdmfilelist'
* Plugin copyfile: Start Phase 0
> Plugin copyfile: Keys/values found:
   filelist: sdmfilelist
> Plugin copyfile: Create directory /mnt/sdm/etc/sdm/assets/copyfile
> Plugin copyfile: Process filelist 'sdmfilelist'
> Plugin copyfile: Add 'calibrate_device.py' to files to copy
> Plugin copyfile: Directory '/home/mm/controller' will be created
> Plugin copyfile: Add 'cloud.py' to files to copy
> Plugin copyfile: Directory '/home/mm/controller' will be created
> Plugin copyfile: Add 'constants.py' to files to copy
> Plugin copyfile: Directory '/home/mm/controller' will be created
> Plugin copyfile: Add 'controller.py' to files to copy
> Plugin copyfile: Directory '/home/mm/controller' will be created
> Plugin copyfile: Add 'device_sensors.py' to files to copy
> Plugin copyfile: Directory '/home/mm/controller' will be created
> Plugin copyfile: Add 'device.py' to files to copy
> Plugin copyfile: Directory '/home/mm/controller' will be created
> Plugin copyfile: Add 'environment.py' to files to copy
> Plugin copyfile: Directory '/home/mm/controller' will be created
> Plugin copyfile: Add 'hx711.py' to files to copy
> Plugin copyfile: Directory '/home/mm/controller' will be created
> Plugin copyfile: Add 'register_device.txt' to files to copy
> Plugin copyfile: Directory '/home/mm/controller' will be created
cp: cannot stat 'register_device.txt': No such file or directory
> Plugin copyfile: Add 'requirements.txt' to files to copy
> Plugin copyfile: Directory '/home/mm/controller' will be created
> Plugin copyfile: Add 'utils.py' to files to copy
> Plugin copyfile: Directory '/home/mm/controller' will be created
> Plugin copyfile: Add 'visualcrossing.py' to files to copy
> Plugin copyfile: Directory '/home/mm/controller' will be created
> Plugin copyfile: Add 'zone.py' to files to copy
> Plugin copyfile: Directory '/home/mm/controller' will be created
* Plugin copyfile: Complete Phase 0
> Run Plugin 'disables' (/mnt/sdm/usr/local/sdm/plugins/disables) Phase 0 with arguments:
  'piwiz'
* Plugin disables: Start Phase 0
> Plugin disables: Keys/values found:
   piwiz:
* Plugin disables: Complete Phase 0
> Run Plugin 'L10n' (/mnt/sdm/usr/local/sdm/plugins/L10n) Phase 0 with arguments: 'host'
* Plugin L10n: Start Phase 0
> Plugin L10n: Keys/values found:
   host:
> Plugin L10n: Read L10n configuration from host system
% /etc/default/keyboard missing; Keyboard set to 'us'
> Plugin L10n: Load Localization (L10N) settings from running system
> Plugin L10n:   Keymap:       us
> Plugin L10n:   Locale:       en_US.UTF-8
> Plugin L10n:   Timezone:     America/Chicago
> Plugin L10n:   WiFi Country:
> Plugin L10n: Save L10n configuration
? Prefix L10n: Unrecognized keymap 'us'
* Plugin L10n: Complete Phase 0
> Run Plugin 'user' (/mnt/sdm/usr/local/sdm/plugins/user) Phase 0 with arguments:
  'adduser=mm|password=secret'
* Plugin user: Start Phase 0
> Plugin user: Keys/values found:
   adduser: mm
   password: secret
> Plugin user: Create user 'mm' home directory '/home/mm' so available for other plugins
* Plugin user: Complete Phase 0
> Run Plugin 'user' (/mnt/sdm/usr/local/sdm/plugins/user) Phase 0 with arguments: 'deluser=pi'
* Plugin user: Start Phase 0
> Plugin user: Keys/values found:
   deluser: pi
* Plugin user: Complete Phase 0
> Run Plugin 'network' (/mnt/sdm/usr/local/sdm/plugins/network) Phase 0 with arguments:
  'netman=nm|wifissid='Wishy Washy'|wifipassword='not to late to fate'|wificountry=US|noipv6'
> Plugin network: Configuring 'network-manager' for network configuration management
> Run Plugin 'apps' (/mnt/sdm/usr/local/sdm/plugins/apps) Phase 0 with arguments:
  'apps=python3-pip'
* Plugin apps: Start Phase 0
> Plugin apps: Keys/values found:
   apps: python3-pip
> Plugin apps [default]: Saved apps list: python3-pip
* Plugin apps: Complete Phase 0
> Run Plugin 'mmsetup' (/mnt/sdm/usr/local/sdm/local-plugins/mmsetup) Phase 0 with no arguments
* Plugin mmsetup: Start Phase 0
* Plugin mmsetup: Complete Phase 0
* Phase 0 Completed
* Enter image '2023-10-10-raspios-bookworm-armhf-lite.img' for Phase 1
Failed to open system bus: No such file or directory
umount: /mnt/sdm/boot unmounted
umount: /mnt/sdm unmounted
daniel@DESKTOP-LM9QIPJ:/mnt/c/Users/Daniel/Documents/mosquito-controller/MosquitoMax/device_controller$ sudo sdm -V
sdm: invalid option -- 'V'
? /usr/local/bin/sdm: Unable to parse command
daniel@DESKTOP-LM9QIPJ:/mnt/c/Users/Daniel/Documents/mosquito-controller/MosquitoMax/device_controller$ sudo sdm --versi
on
sdm V9.6
```

I looked at https://github.com/gitbls/sdm/blob/1beea52cb27dcb95dab8d6f6a50bf519bef9c2ac/CHANGELOG.md, but didn't see it updated for the latest version yet.

dwatrous, 2023-11-15:
I just downgraded to 9.1.1 and I'm getting the same issue. Maybe it's bookworm related. I'm still investigating.

dwatrous, 2023-11-15:
I'm running on WSL 2 (Windows 10 Pro). I found some posts about others having issues with dbus, and I tried enabling dbus, but I'm still getting this.

`Failed to register machine: Launch helper exited with unknown return code 1`

This appears to be WSL 2 related. I'm not sure why it used to work and stopped working today. Maybe I'll try deleting my QSL instance and installing fresh. Let me know if you have seen this or have any ideas to fix it.

gitbls, 2023-11-15:
Hmm...interesting. I'm not seeing this in my Debian 11 WSL2. What exact distro are you running? (note to self: add this to the info output to /etc/sdm/history)

Have you added systemd=true to /etc/wsl.conf as noted here and elsewhere (according to search for:wsl2 enable systemd): https://superuser.com/questions/1785697/systemd-in-wsl-on-windows-11-is-not-working  If systemd isn't running, try enabling that (edit wsl.conf, shutdown WSL completely, restart WSL) and rerun your test.

And, pls point me at the others having issues with dbus that you found.

dwatrous, 2023-11-16:
Lots of https://github.com/microsoft/WSL/issues/10350, https://github.com/microsoft/WSL/issues/2941, and [other sites ](https://superuser.com/questions/1561076/systemctl-user-failed-to-connect-to-bus-no-such-file-or-directory-debian-9)...

For my work yesterday I spun up a GCE instance, which worked great.

I did just add /etc/wsl.conf to my WSL disto with the directive to start systemd at boot. I then restarted the instance and I can now build the image again. Thank you.

closed 2023-11-16 by dwatrous

gitbls, 2023-11-16:
> Lots of [microsoft/WSL#10350](https://github.com/microsoft/WSL/issues/10350), [microsoft/WSL#2941](https://github.com/microsoft/WSL/issues/2941), and [other sites ](https://superuser.com/questions/1561076/systemctl-user-failed-to-connect-to-bus-no-such-file-or-directory-debian-9)...
> 
> For my work yesterday I spun up a GCE instance, which worked great.
> 
> I did just add /etc/wsl.conf to my WSL disto with the directive to start systemd at boot. I then restarted the instance and I can now build the image again. Thank you.

Two of those links are 3+ year old issues, before wsl supported systemd. I'm surprised, however, that the most recent issue (https://github.com/microsoft/WSL/issues/10350) was having problems, since the OP mentioned he had done the wsl.conf "fix".

Great that the fix worked for you. On to the next challenge!