gitbls/sdm #134: Set Password for PI User

dbeckwith01, 2023-11-23:
I upgraded to v9.6 and I'm trying to recreate to recreate this `sdm` image configuration but I can't get the password to work for a raspberry pi.  How do I set the password for the pi user?

**Old config:**
```
filename="2023-05-03-raspios-bullseye-arm64.img" 

sudo sdm --customize --batch $filename \
	--extend --xmb 3096 \
	--l10n --restart --disable piwiz \
	--regen-ssh-host-keys --nowait-timesync \
	--autologin --bootset boot_behaviour:B4 \
	--password-pi brnkl \
	--plugin apps:"apps=iperf3 cmake python3-pip libudev-dev i2c-tools can-utils lshw pciutils usbutils lsscsi bc curl hwinfo hdparm nvme-cli dmidecode smartmontools fio sdparm xxd libxml-dumper-perl git neovim libusb-1.0-0-dev pkg-config build-essential cmake" \
	--cscript sdm_commands_testjig.sh

sudo sdm --shrink $filename
```

**New Config:**

```

filename="2023-05-03-raspios-bullseye-arm64-lite.img" 

sudo sdm --customize --batch $filename \
	--extend --xmb 3096 \
	--plugin L10n:host --restart --plugin disables:piwiz \
	--regen-ssh-host-keys --nowait-timesync \
	--autologin --plugin raspiconfig:boot_behaviour:B2 \
	--plugin user:`setpassword=pi|password=brnkl` \
	--plugin apps:"apps=iperf3 cmake python3-pip libudev-dev i2c-tools can-utils lshw pciutils usbutils lsscsi bc curl hwinfo hdparm nvme-cli dmidecode smartmontools fio sdparm xxd libxml-dumper-perl git neovim libusb-1.0-0-dev pkg-config build-essential cmake" \
	# --cscript sdm_commands_testjig.sh

sudo sdm --shrink $filename
```

1stcall, 2023-11-23:
> --plugin user:`setpassword=pi|password=brnkl` \
Looks like you are using backticks instead of quotes. ` vs " or '

gitbls, 2023-11-23:
@1stcall is correct. Double quotes, not backticks.

dbeckwith01, 2023-11-23:
Thank you for answering, I will try that tomorrow.  I did use the '--explore' option to configure the password as I wasn't sure that backticks would work but after booting it said the user wasn't defined.

gitbls, 2023-11-23:
JFYI if you want/need to create a user when you're `--explore` ing an IMG or SD card, you can use the `useradd` command to create one. See `man useradd` for details.

dbeckwith01, 2023-11-23:
@gitbls That was the issue, thank you for the help.

closed 2023-11-23 by dbeckwith01