gitbls/sdm #135: Custom Script And Coping Files

dbeckwith01, 2023-11-24:
After upgrading from v8.x to v9.6 I am having trouble copying files from the host to the IMG using my `--cscript`. 

I have the following script which should copy files form my host to the IMG in Phase 0 but it doesn't seem to be working after the upgrade.

Here is my command to `sdm` and the output is at the end.  It seems like the Phase 0 copy is happening in Phase 1.  I tried putting an `ls -al /home/pi` in the script and it only showed the contents of the PI directory on the image not my host.

```
filename="2023-05-03-raspios-bullseye-arm64-lite.img" 

sudo sdm --customize --batch $filename \
	--extend --xmb 3096 \
	--cscript sdm_commands_testjig.sh 

sudo sdm --shrink $filename
```

```
#!/bin/bash
# This runs inside the cm4 image during creation

function loadparams() {
    source $SDMPT/etc/sdm/sdm-readparams
}

# $1 is the phase: "0", "1", or "post-install"
phase=$1
pfx="$(basename $0)"
if [ "$phase" == "0" ]
loadparams
then
  logtoboth "* $pfx Phase 0"
  # install custom config.txt
  cp config.txt $SDMPT/boot/config-new.txt
  cp mcp2515-can1-0.dtbo $SDMPT/boot/overlays
  
  # install custom udev rule for rpi probe
  cp 50-rpiprobe.rules $SDMPT/etc/udev/rules.d
  
  source /home/pi/Test-Jig/.env
  echo $GITHUB_AUTH_TOKEN > $SDMPT/home/pi/.github_token
  
  logtoboth "* $pfx Phase 0 Completed"
 elif [ "$phase" == "post-install" ]
 then
  # finally, force config.txt
  sudo mv /boot/config-new.txt /boot/config.txt
fi
```

Here is the output of SDM:

```
* Mount IMG '2023-05-03-raspios-bullseye-arm64-lite.img'
mount: /dev/loop0p2 mounted on /mnt/sdm.
mount: /dev/loop0p1 mounted on /mnt/sdm/boot.
** IMG '2023-05-03-raspios-bullseye-arm64-lite.img' has already been customized. Redo? [y/n]? y
* Unmount IMG for --extend
umount: /mnt/sdm/boot unmounted
umount: /mnt/sdm unmounted
* Extend IMG by 3096MB (3.2GB, 3.0GiB)...

3243245568 bytes (3.2 GB, 3.0 GiB) copied, 39 s, 83.1 MB/s
3096+0 records in
3096+0 records out
3246391296 bytes (3.2 GB, 3.0 GiB) copied, 39.0613 s, 83.1 MB/s
* Resize partition 2 of '2023-05-03-raspios-bullseye-arm64-lite.img' to 5539MB
* Remount IMG to resize the file system
* Mount IMG '2023-05-03-raspios-bullseye-arm64-lite.img'
mount: /dev/loop0p2 mounted on /mnt/sdm.
mount: /dev/loop0p1 mounted on /mnt/sdm/boot.
* Resize the file system
% (Ignore 'on-line resizing required' message)
resize2fs 1.46.2 (28-Feb-2021)
Filesystem at /dev/loop0p2 is mounted on /mnt/sdm; on-line resizing required
old_desc_blocks = 1, new_desc_blocks = 1
The filesystem on /dev/loop0p2 is now 1352385 (4k) blocks long.

* Start Configuration
> Command Line: /usr/local/bin/sdm --customize --batch 
  2023-05-03-raspios-bullseye-arm64-lite.img --extend --xmb 3096 --cscript sdm_commands_testjig.sh
* Host Information
   Hostname: black-test-testjig
   uname: Linux black-test-testjig 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 
  aarch64 GNU/Linux
   os-release Name: "Debian GNU/Linux"
   os-release Version: "11 (bullseye)"
   sdm Version: V9.6
* IMG Information
   Name: 2023-05-03-raspios-bullseye-arm64-lite.img
   Date: 2023-05-03
   RasPiOS Version: 11
   RasPiOS Architecture: 64-bit aarch64
   os-release Version: "11 (bullseye)"
% No plugins are selected
> IMG '2023-05-03-raspios-bullseye-arm64-lite.img' has 3766448 1K-blocks (3.9GB, 3.6GiB) free 
> Copy sdm to /usr/local/sdm in the IMG
> Copy Custom Phase Script '/home/pi/BRNKL-Black-Test/raspberrypi/sdm_commands_testjig.sh' to 
  /usr/local/sdm in the IMG
* Start Phase 0 image customization
> No Plugins specified with --plugin
* Phase 0 Completed
* Enter image '2023-05-03-raspios-bullseye-arm64-lite.img' for Phase 1

* Start Phase 1 image customization
> IMG '2023-05-03-raspios-bullseye-arm64-lite.img' has 3766448 1K-blocks (3.9GB, 3.6GiB) free at start of Phase 1 image customization
> Configure and enable sdm FirstBoot service (sdm-firstboot)
> Start 'apt update'
> IMG '2023-05-03-raspios-bullseye-arm64-lite.img' has 3766448 1K-blocks (3.9GB, 3.6GiB) free at start of 'apt update'
> IMG '2023-05-03-raspios-bullseye-arm64-lite.img' has 3766456 1K-blocks (3.9GB, 3.6GiB) free at end of 'apt update'
> Run Custom Phase Script '/usr/local/sdm/sdm_commands_testjig.sh' Phase 1
* sdm_commands_testjig.sh Phase 0
cp: cannot stat 'config.txt': No such file or directory
cp: cannot stat 'mcp2515-can1-0.dtbo': No such file or directory
cp: cannot stat '50-rpiprobe.rules': No such file or directory
/usr/local/sdm/sdm_commands_testjig.sh: line 22: /home/pi/BRNKL-Black-Test/.env: No such file or directory
* sdm_commands_testjig.sh Phase 0 Completed
sudo: unable to resolve host sdm: No address associated with hostname
mv: cannot stat '/boot/config-new.txt': No such file or directory
> No Plugins specified with --plugin
* Phase 1 post-app installation/configuration
> Start 'apt upgrade'
> IMG '2023-05-03-raspios-bullseye-arm64-lite.img' has 3766456 1K-blocks (3.9GB, 3.6GiB) free at start of 'apt upgrade'
> IMG '2023-05-03-raspios-bullseye-arm64-lite.img' has 3766456 1K-blocks (3.9GB, 3.6GiB) free at end of 'apt upgrade'
> Start 'apt autoremove'
> IMG '2023-05-03-raspios-bullseye-arm64-lite.img' has 3766456 1K-blocks (3.9GB, 3.6GiB) free at start of 'apt autoremove'
> IMG '2023-05-03-raspios-bullseye-arm64-lite.img' has 3766456 1K-blocks (3.9GB, 3.6GiB) free at end of 'apt autoremove'
> IMG '2023-05-03-raspios-bullseye-arm64-lite.img' has 3766456 1K-blocks (3.9GB, 3.6GiB) free at end of Phase 1 image customization
* Phase 1 Completed
> Run Custom Phase Script '/usr/local/sdm/sdm_commands_testjig.sh' post-install
* sdm_commands_testjig.sh Phase 0
cp: cannot stat 'config.txt': No such file or directory
cp: cannot stat 'mcp2515-can1-0.dtbo': No such file or directory
cp: cannot stat '50-rpiprobe.rules': No such file or directory
/usr/local/sdm/sdm_commands_testjig.sh: line 22: /home/pi/BRNKL-Black-Test/.env: No such file or directory
* sdm_commands_testjig.sh Phase 0 Completed
sudo: unable to resolve host sdm: No address associated with hostname
mv: cannot stat '/boot/config-new.txt': No such file or directory
> No Plugins specified with --plugin
> Run graphics post-install configuration
> Enable SSH
> Plugin phase1: Enable SSH service via /boot/ssh
> IMG '2023-05-03-raspios-bullseye-arm64-lite.img' has 3766456 1K-blocks (3.9GB, 3.6GiB) free at end of image customization
* Customization complete
* Batch Mode exit
umount: /mnt/sdm/boot unmounted
umount: /mnt/sdm unmounted
* Start shrink on IMG '2023-05-03-raspios-bullseye-arm64-lite.img'
> Check the file system
rootfs: 42144/335328 files (0.4% non-contiguous), 404277/1352385 blocks
resize2fs 1.46.2 (28-Feb-2021)
> Shrink the file system to 559913
resize2fs 1.46.2 (28-Feb-2021)
Resizing the filesystem on /dev/loop0 to 559913 (4k) blocks.
Begin pass 2 (max = 22659)
Relocating blocks             XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
Begin pass 3 (max = 42)
Scanning inode table          XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
The filesystem on /dev/loop0 is now 559913 (4k) blocks long.

> Remove old partition
> Make new partition
> Shrink the image by truncating to 2566033408
* Shrink complete
  IMG '2023-05-03-raspios-bullseye-arm64-lite.img' was (5.8GB, 5.4GiB) now (2.6GB, 2.4GiB)
  ```

gitbls, 2023-11-24:
Sigh...when I did the extreme surgery to sdm to fully implement plugins, I removed a wee bit too much code from phase 0...specifically the code that runs the cscript Phase 0. Your cscript Phae0 didn't run, so files didn't get copied into the IMG, so obviously not in the IMG container for Phase 1 or post-install.

The expedient path forward for you is to convert your cscript to a Plugin. The changes are extremely minimal. I'm not in a place where I can fully check this out, but it could be as simple as removing `--cscript yourscript` and adding `--plugin yourscript`.  Since cscripts don't have any arguments you don't need to add calls to `plugin_getargs` or any other plugin-specific functions unless and until you decide to fully embrace plugin features.

I'd also encourage you to use /full/path/to/config.txt and other "un-pathed" references (I noticed 3 of them) in your Phase 0 code, because sdm does not guarantee that it won't change directories on you.

See https://github.com/gitbls/sdm/blob/master/Docs/Programming-Plugins-and-Custom-Phase-Scripts.md for further details.

Apologies for this. Will correct cscript for the next release.

dbeckwith01, 2023-11-24:
@gitbls Thanks for talking the time to answer.  I will try converting this to a plugin and let you know the result.

dbeckwith01, 2023-11-25:
Hi @gitbls, unfortunately that didn't seem to work as the files didn't copy. For now I have downgraded back to v8.6. As I need to move forward with some testing.

gitbls, 2023-11-25:
Sorry to hear that, but I can't help you if you don't provide any details of the errors as you did in the base post...

gitbls, 2023-11-29:
sdm V10.0 corrects the missing call to custom phase scripts in phase 0. Please test and verify. Thx

dbeckwith01, 2023-11-30:
Thank you very much, I will give it a test soon.

gitbls, 2023-12-05:
Closing this issue as resolved. If your testing uncovers additional issues, please feel free to reopen. Thx