gitbls/sdm #136: IMG already attached to a loop device

dwatrous, 2023-11-25:
I'm trying to burn an image to disk so I can write it using raspi imager. I have done this many times with bullseye, but this is my first time with bookworm (I'm not sure it's related).

Here's the output I get.

```
$ sudo sdm --burnfile mm.img --expand-root 2023-10-10-raspios-bookworm-armhf-lite.img
% hostname not specified with --host; will not be written
* Burn '2023-10-10-raspios-bookworm-armhf-lite.img' (3.0GB, 2.8GiB) to Output IMG 'mm.img'...
dd if=2023-10-10-raspios-bookworm-armhf-lite.img of=mm.img status=progress bs=16M iflag=direct
2701131776 bytes (2.7 GB, 2.5 GiB) copied, 6 s, 448 MB/s
181+1 records in
181+1 records out
3045064704 bytes (3.0 GB, 2.8 GiB) copied, 6.79882 s, 448 MB/s
Disk identifier changed from 0x7550d926 to 0x0f2622c4.

The partition table has been altered.
Calling ioctl() to re-read partition table.
Syncing disks.
> Disk ID 0f2622c4 set successfully with sfdisk
? IMG 'mm.img' already attached to a loop device
```

For reference, here's the full sdm command used before calling `--burnfile`.

`sudo sdm --customize --plugin raspiconfig:"i2c=0" --plugin copyfile:"filelist=sdmfilelist" --plugin disables:piwiz --plugin L10n:host --plugin user:"adduser=user|password=passwd" --plugin user:"deluser=pi" --plugin network:"netman=nm|wifissid='MyWiFi'|wifipassword='mywifipwd'|wificountry=US|noipv6" --plugin apps:"apps=python3-pip,python3-venv,python3-dev" --plugin mmsetup --extend --xmb 500 --restart --host mmdwatrous 2023-10-10-raspios-bookworm-armhf-lite.img`

gitbls, 2023-11-25:
I won't be able to look at this for a couple of days. In the meantime, could you please post the output from the following commands?
```
sudo losetup -a
df
sudo sdm --version
```
Thx. I also see that you specify the `--host` switch on customize but the burn command seems to be oblivious to that. Will correct that as well.

dwatrous, 2023-11-25:
Where do you want me to run those? On the host where I run sdm? in the image using --explore? At the initramfs prompt?

```
$ sudo sdm --version
sdm V9.6
```

dwatrous, 2023-11-25:
On the WSL host where I run sdm

```
$ df
Filesystem      1K-blocks      Used  Available Use% Mounted on
none              8174580         4    8174576   1% /mnt/wsl
none            497987224 375215380  122771844  76% /usr/lib/wsl/drivers
/dev/sdc       1055762868    800764 1001258632   1% /
none              8174580       504    8174076   1% /mnt/wslg
none              8174580         0    8174580   0% /usr/lib/wsl/lib
rootfs            8171284      2056    8169228   1% /init
none              8171284         0    8171284   0% /dev
none              8174580       484    8174096   1% /run
none              8174580         0    8174580   0% /run/lock
none              8174580         0    8174580   0% /run/shm
tmpfs                4096         0       4096   0% /sys/fs/cgroup
none              8174580        76    8174504   1% /mnt/wslg/versions.txt
none              8174580        76    8174504   1% /mnt/wslg/doc
drvfs           497987224 375215380  122771844  76% /mnt/c
drvfs          1953512032 841129844 1112382188  44% /mnt/e
drvfs           976760828 704676092  272084736  73% /mnt/f
```

I don't get any output for the `losetup -a` command.

gitbls, 2023-11-25:
Thanks for the additional input. Will look into it when I'm back at my desktop.

gitbls, 2023-11-26:
You're running on WSL, so you may not have the system journal enabled. But if you do, could you check the log (with journalctl) for 
```
sudo journalctl | grep 'I/O error'
```
And see if there's an error logged on loop0 at about the time you were doing the above burnfile?

dwatrous, 2023-11-26:
The command you give doesn't produce any output. When I look for "loop0", I do get this:

```
Nov 26 14:31:42 DESKTOP-LM9QIPJ kernel: loop0: detected capacity change from 0 to 5947392
Nov 26 14:31:42 DESKTOP-LM9QIPJ kernel:  loop0: p1 p2
Nov 26 14:31:42 DESKTOP-LM9QIPJ kernel:  loop0: p1 p2
```

gitbls, 2023-11-27:
OK, thx. That error was a red herring (for me). I'm now able to repro it on a Pi5 _sometimes_ ... Working on understanding the problem in order to properly fix it.

gitbls, 2023-11-29:
The repro case takes somewhere between 30-50 runs. The updated version did not exhibit the error in over 200 runs, so I think it's nailed. Please LMK if this issue reoccurs on V10.0. Thx

dwatrous, 2023-12-03:
This is now working with V10.0. Thank you.

closed 2023-12-03 by dwatrous