gitbls/sdm #137: Extend an image with "qemu-img" instantly rather than using "dd" copying process

carriba, 2023-11-25:
More of an enhancement request for the "[extendimage](https://github.com/gitbls/sdm/blob/master/sdm) function currently in place with **SDM**:

Instead of extending an image file using the file copying process of [dd](https://en.wikipedia.org/wiki/Dd_(Unix)), why not using the [qemu](https://en.wikipedia.org/wiki/qemu) image file size extension utility?

To backup my question, here my test and my findings:

Using the command `ls -l *.img` I am getting the following output pertaining to the original image file:

> -rw-r--r--  1 pi pi  1877954560 Nov 25 06:28  raspbian.img

Firing the command `qemu-img info -f raw raspbian.img`, outputs:

> image: raspbian.img
> file format: raw
> virtual size: 1.75 GiB (1877954560 bytes)
> disk size: 1.48 GiB

Firing the command `sudo sdm --extend raspbian.img`, I am getting after some patience the image file extended. I can confirm this by firing the command  `qemu-img info -f raw raspbian.img` that displays:

> image: raspbian.img
> file format: raw
> virtual size: 3.75 GiB (4025438208 bytes)
> disk size: 3.48 GiB

Using the same original image file as above, and firing the command `qemu-img resize -f raw raspbian.img +2G`, I am getting instantly the image file resized (no waiting as above).

Firing the command `qemu-img info -f raw raspbian.img`, outputs:

> image: raspbian.img
> file format: raw
> virtual size: 3.75 GiB (4021174272 bytes)
> disk size: 1.48 GiB

Firing the command `sudo sdm --shrink raspbian.img`, I am getting after some short while the image file shrunk.

I can verify this with the command `qemu-img info -f raw raspbian.img` that outputs:

> image: raspbian.img
> file format: raw
> virtual size: 1.73 GiB (1861214208 bytes)
> disk size: 1.48 GiB

Using the command `ls -l *.img` I am getting the following output pertaining to the shrunk image file:

> -rw-r--r-- 1 pi pi  1861214208 Nov 25 06:44 raspbian.img

Until now, all my initiatives to extend an image file with the utility `qemu-img` and mount the image file for further processing worked fine.

gitbls, 2023-11-25:
Thanks for this. I'll have a look at it to see if it works in all scenarios required by sdm.

1stcall, 2023-11-26:
truncate also works much faster than dd.  Also consider using copy rather than dd when using the --burnimage option as this works much faster on COW filesystems such as BTRFS, which is getting more popular nowadays.

gitbls, 2023-11-27:
Thanks. Will review it carefully when I start looking at this in detail.

1stcall, 2023-11-27:
just use `cp --reflink=auto`  every time would be simplest solution.  No need to dd at all with burnfile.

gitbls, 2023-11-27:
> just use `cp --reflink=auto` every time would be simplest solution. No need to dd at all with burnfile.

Yes, but cp doesn't provide feedback, while dd does (status=progress). While it doesn't matter for COW, since it's fast, progress feedback is important for non-COW systems.

My current inclination is to go the (icky) switch route. Generally less muss and fuss, and only required if you're on a COW file system.

1stcall, 2023-11-27:
I did have an iscow function some time ago but I can't find it or remember how I did it.  I was surprised but it worked on nfs mounts also.  I'll have another look in the morning. I lost a drive earlier this year and I suspect it was on there.

gitbls, 2023-11-27:
> I did have an iscow function some time ago but I can't find it or remember how I did it. I was surprised but it worked on nfs mounts also. I'll have another look in the morning. I lost a drive earlier this year and I suspect it was on there.

Backup ooopsie? If you can dig it out I'd like to see it, but if not, hiho a switch-ing we will go

1stcall, 2023-11-28:
You'll not like it but `cp --reflink=always <source> <destination> 2>/dev/null || dd if=<source> of=<destination> status=progress` would work.

Or replace the ddsw switch with ddcmd switch.

1stcall, 2023-11-28:
This works on nfs shares also

1stcall, 2023-11-28:
cp --progress would be nice.

gitbls, 2023-11-28:
> cp --reflink=always <source> <destination> 2>/dev/null || dd if=<source> of=<destination> status=progress

Indeed. Too bad nobody has added it.

gitbls, 2023-11-28:
> cp --progress would be nice.
BTW How long does it take to copy a RasPiOS IMG using cp with COW enabled?

gitbls, 2023-11-28:
I was afraid you were going to say that 🤣 Looks like I'm going to have to sort that out soon. But next sdm release first!

gitbls, 2023-11-29:
sdm V10.0 uses qemu-img to extend an IMG. I'm off to rebuild my work disk as btrfs 👍

1stcall, 2023-11-30:
btrfs-convert works well.

Maybe for version 11 or 12 :-
```#!/usr/bin/env bash

#set -x
imgfile=${1:-unset}
[ ! -f $imgfile ] && echo "$imgefile: File not found." && exit 1
loopdev=$(sudo losetup --show -P -f $imgfile)
sudo btrfs-convert ${loopdev}p2
sudo partprobe -s $loopdev
sudo mount ${loopdev}p1 /mnt/tempmnt
mv -v /mnt/tempmnt/cmdlint.txt /mnt/tempmnt/cmdline.txt.prev
cat /mnt/tempmnt/cmdline.txt.prev | sed 's/root=LABEL=RASPIROOT/root=LABEL=RASPIROOT ***@***.*** rootfstype=btrfs/g' | sed 's/fsck.repair=yes/fsck.repair=no/g' | sed 's/  / /g' | tee /mnt/tempmnt/cmdlint.txt
copy -v /mnt/tempmnt/cmdline.txt /mnt/tempmnt/cmdline.txt
sudo umount /mnt/tempmnt
sudo mount ${loopdev}p2 /mnt/tempmnt
cat /mnt/tempmnt/etc/fstab.orig | sed 's$/ ext4 rw 0 1$/ btrfs ***@***.*** 0 0$g' | sudo tee /mnt/tempmnt/etc/fstab
cp -v /mnt/tempmnt/etc/fstab /mnt/tempmnt/etc/fstab.new
sudo btrfs subvolume create ***@***.***
sudo mv -v /mnt/tempmnt/* ***@***.***/
sudo umount /mnt/tempmnt
#set +x
exit 0
``
________________________________
From: Benn ***@***.***>
Sent: 29 November 2023 7:41 PM
To: gitbls/sdm ***@***.***>
Cc: Carl McAlwane ***@***.***>; Mention ***@***.***>
Subject: Re: [gitbls/sdm] Extend an image with "qemu-img" instantly rather than using "dd" copying process (Issue #137)

sdm V10.0 uses qemu-img to extend an IMG. I'm off to rebuild my work disk as btrfs 👍

—
Reply to this email directly, view it on GitHub<https://github.com/gitbls/sdm/issues/137#issuecomment-1832591059>, or unsubscribe<https://github.com/notifications/unsubscribe-auth/ADCHHLNCIYJXJO2HXS7KXGLYG6FW7AVCNFSM6AAAAAA72HWUDCVHI2DSMVQWIX3LMV43OSLTON2WKQ3PNVWWK3TUHMYTQMZSGU4TCMBVHE>.
You are receiving this because you were mentioned.Message ID: ***@***.***>

gitbls, 2023-11-30:
Thanks. Filed for future reference. I did the conversion by copying everything off, fdisk/delpart,mkfs btrfs, and copy everything back. Not as efficient as btrfs-convert, but did the job.

1stcall, 2023-11-30:
Probably safer.

I don't trust BTRFS with raid yet.  That is how I lost my data earlier in the year.

gitbls, 2023-11-30:
> Probably safer. I don't trust BTRFS with raid yet. That is how I lost my data earlier in the year.
I'm in the "trying it out" stage with BTRFS. It's on a 1TB SSD that would be a pain if I lost it, but there's nothing irreplacable on it. Those super-fast copies are def 👍❣

1stcall, 2023-11-30:
snapshots are quite cool too

gitbls, 2023-12-05:
Closing this issue as resolved. Thanks for the suggestion....it's wicked fast!

closed 2023-12-05 by gitbls

carriba, 2023-12-06:
Here my elaborations to extend the image file with the **qemu-img** tool and extend the file system of the 2nd partition. I used this during the past past days with some extra checks to make sure that the script is running fine.

You have to install the following packages with the command **sudo apt -y install qemu** before running the script below.

Then you can run the following script to increase the image file:

````
#/bin/bash

do_expand_fs_img() {
if [ "$IMG_FILE" -a -f "$IMG_FILE" ]; then
  qemu-img resize -f raw $IMG_FILE $IMG_EXP
  INIT=`ps --no-headers -o comm 1`
  IMG_DEV=`losetup --show -P -f $IMG_FILE`
  if [ "$INIT" = "systemd" -a -n "$IMG_DEV" -a -e "$IMG_DEV" ]; then
    PART_NUM=`fdisk -lo device $IMG_DEV | grep -c "^${IMG_DEV}p.$"`
    if [ $PART_NUM -eq `partprobe -s $IMG_DEV | grep -o '.$'` ]; then
      IMG_PART=`fdisk -lo device $IMG_DEV | grep "^${IMG_DEV}p.$" | tail -n 1`
      if [ "${IMG_DEV}p${PART_NUM}" = "$IMG_PART" -a -e "$IMG_PART" ]; then
        PART_START=`fdisk -lo start $IMG_DEV | tail -n 1`
        if [ $PART_START -gt `fdisk -lo end $IMG_DEV | tail -n 2 | head -n 1` ]; then
          tune2fs -O ^has_journal $IMG_PART
          fdisk $IMG_DEV <<EOF
d
$PART_NUM
n
p
$PART_NUM
$PART_START

w
EOF
        e2fsck -pf $IMG_PART
        resize2fs $IMG_PART
        tune2fs -j $IMG_PART
        fi
      fi
      losetup -d $IMG_DEV
      sync
      sleep 2
      sync
    fi
  fi
fi
}

IMG_EXP="+2G"
IMG_FILE=/home/pi/raspbian.img
do_expand_fs_img
````

Looking forward to your feedback on this. I may need to add an extra check to verify that **qemu-img** did indeed run without any errors as I did run out of disk space on the host machine during one of my tests.

If required I can rework above towards a plug-in.

gitbls, 2023-12-07:
> Here my elaborations to extend the image file with the **qemu-img** tool and extend the file system of the 2nd partition. I used this during the past past days with some extra checks to make sure that the script is running fine.
> 
> You have to install the following packages with the command **sudo apt -y install qemu-utils** before running the script below.
> 
> Then you can run the following script to increase the image file and the embedded file system:
> 
> ```
> #/bin/bash
> 
> do_expand_fs_img() {
> if [ "$IMG_FILE" -a -f "$IMG_FILE" ]; then
>   qemu-img resize -f raw $IMG_FILE $IMG_EXP
>   INIT=`ps --no-headers -o comm 1`
>   IMG_DEV=`losetup --show -P -f $IMG_FILE`
>   if [ "$INIT" = "systemd" -a -n "$IMG_DEV" -a -e "$IMG_DEV" ]; then
>     PART_NUM=`fdisk -lo device $IMG_DEV | grep -c "^${IMG_DEV}p.$"`
>     if [ $PART_NUM -eq `partprobe -s $IMG_DEV | grep -o '.$'` ]; then
>       IMG_PART=`fdisk -lo device $IMG_DEV | grep "^${IMG_DEV}p.$" | tail -n 1`
>       if [ "${IMG_DEV}p${PART_NUM}" = "$IMG_PART" -a -e "$IMG_PART" ]; then
>         PART_START=`fdisk -lo start $IMG_DEV | tail -n 1`
>         if [ $PART_START -gt `fdisk -lo end $IMG_DEV | tail -n 2 | head -n 1` ]; then
>           tune2fs -O ^has_journal $IMG_PART
>           fdisk $IMG_DEV <<EOF
> d
> $PART_NUM
> n
> p
> $PART_NUM
> $PART_START
> 
> w
> EOF
>         e2fsck -pf $IMG_PART
>         resize2fs $IMG_PART
>         tune2fs -j $IMG_PART
>         fi
>       fi
>       losetup -d $IMG_DEV
>       sync
>       sleep 2
>       sync
>     fi
>   fi
> fi
> }
> 
> IMG_EXP="+2G"
> IMG_FILE=/home/pi/raspbian.img
> do_expand_fs_img
> ```
> 
> Looking forward to your feedback on this. I may need to add an extra check to verify that **qemu-img** did indeed run without any errors as I did run out of disk space on the host machine during one of my tests.
> 
> If required I can rework above towards a plug-in.

Please explain how this is different than the sdm command: `sdm --extend --xmb NNNN img.img`, which extends the 2nd partition in the IMG by NNNN MB.

carriba, 2023-12-07:
I had a look again into the [SDM](https://github.com/gitbls/sdm/blob/master/sdm) script and compared it against [my script](https://github.com/gitbls/sdm/issues/137#issuecomment-1843820953). 

I just created my script to review and finetune extending the FS after expanding its image file. I am using fdisk, however parted also does the job, despite the small differences both tools bring with; see article "[What’s the Difference Between fdisk and parted](https://www.baeldung.com/linux/fdisk-vs-parted)".

I see that the **[qemu-img](https://www.qemu.org/docs/master/tools/qemu-img.html)** tool is now reflected within the function "_extendimage_" as an optional raw image resizing option, this is great.

I see that the function "_extendandresize_" does fire the "resize2fs" command. Though, in my script I remove the journal before extending the FS in the image file, and recreate the journal afterwards according to the guidelines given with the article “[Getting Started with QEMU and KVM: Resizing an image](https://subscription.packtpub.com/book/cloud-and-networking/9781788294676/1/ch01lvl1sec14/resizing-an-image)”.

If recreating the journal is just optional, I am fine to live with this.

Thank you for having considered my little submission. I will update my local installation of the SDM tool, and take it from there... (And great work and initiative you are taking here by automating the customisation efforts of RPi OS image files!)

gitbls, 2023-12-07:
Thanks for the reference to QEMU/KVM. I'll look into removing the journal, it seems like a good thing to do.

Appreciate the kind words about sdm. If you really like it, please ⭐the repo. Thanks!

1stcall, 2023-12-08:
I'm still not sure what qem-img brings to the table?  Why not use `truncate <imgfile> +X` ? Much faster than dd, but no dependencies.  Still need to `resize2fs`.

gitbls, 2023-12-08:
> I'm still not sure what qem-img brings to the table? Why not use `truncate <imgfile> +X` ? Much faster than dd, but no dependencies. Still need to `resize2fs`.

I managed to either miss or forget about your previous comment re truncate, but not this time. I'm having a look at it.

The great thing about linux is the rich set of tools. Of course, not knowing every one them is one of the more challenging aspects of linux. 😵

1stcall, 2023-12-08:
The name don't help!

gitbls, 2023-12-08:
True that. Just did a quick test and indeed it seems to produce identical results. A one-line change that removes an installation prereq while retaining performance is a win in my book. Thanks for the re-ping!

carriba, 2023-12-09:
> I'm still not sure what qemu-img brings to the table? Why not use `truncate <imgfile> +X` ? Much faster than dd, but no dependencies. Still need to `resize2fs`.

I have been using "**truncate**" as well, and am doing some tests pertaining to image integrity. Though, seems that functionality is identical.

carriba, 2023-12-10:
Have done several tests, and can confirm that "**truncate**" accomplishes the same functionality as "**qemu-img**" to expand an image file. Thus, feel free to opt for this approach, or even offer both.

Regarding the question about the "journal" of an EXT4 FS in an image file, from my understanding, it can be deleted without any consequences after having associated the corresponding partition of the image file to a block device and before mounting the block device to a directory ; command: `sudo tune2fs -O ^has_journal /dev/loop0p2`.

The "journal" can be recreated empty after having unmounted the block device and before disassociating it; command: command: `sudo tune2fs -j /dev/loop0p2`.