gitbls/sdm #139: nfs boot option

1stcall, 2023-11-28:
I boot several of my pi's via nfs/pxeboot and currently extract the filesystem from an image file produced by sdm using this script https://github.com/1stcall/sdm/blob/mydev/extfilesys.sh . I don't  think this is possible with custom plugin. Could u add an option to produce just a boot and root directories?  

I currently use this script to extract the filesystem from an image. https://github.com/1stcall/sdm/blob/mydev/extfilesys.sh

gitbls, 2023-11-28:
Just to be clear, you're looking for a way to extract the boot and root directories (presumably customized) from an IMG and store them at a given location in the file system?

1stcall, 2023-11-28:
Yes. The boot dir is then shared via tftp and the root via nfs.  This is to facilitate network booting of the pi.  The extract script works fine so there is no rush for this.  It would just be nice to not require the extra step.  You have already done the disabling of expanding root on 1st boot etc. which is obviously not required.  Bonus would be to delete the image file after extraction.

gitbls, 2023-11-28:
Got it. Seems fairly simple. Finishing up this release first, then will toss this in.

What does lifecycle maintenance look like? Do you update the nfs-booted systems (and hence the extracted file systems), or do you rebuild the IMG, update, and recreate the nfs-shared directories? Or not update at all? 😵

1stcall, 2023-11-29:
Just a standard apt update && apt upgrade-y along with a few git pulls.

gitbls, 2023-11-29:
So, to be clear, you simply run on that NFS tree, updating it in place with apt. Then, when it's time to rebuild the tree for whatever reason, you shutdown the network-booted pi, wipe and replace the tree on the nfs server, and boot the network-booted pi on the new tree.

Correct?

1stcall, 2023-11-30:
Yes.  When expeiementing and something works well, I write a custom sdm-plugin so that it is there on the next build.  I must experiment with running plugin on a customised system.

1stcall, 2023-11-30:
The nfs host is a Raspberry cm4, built with sdm.

1stcall, 2023-11-30:
My fork of sdm is a thttps://github.com/1stcall/sdm/tree/mydev
[https://opengraph.githubassets.com/08aeda9a64ed5a3ba99cbf7e0d0e00b27a88c8b84dd28b3ff62c2029659d29f5/1stcall/sdm]<https://github.com/1stcall/sdm/tree/mydev>
GitHub - 1stcall/sdm: Raspberry Pi SD Card Image Manager<https://github.com/1stcall/sdm/tree/mydev>
Raspberry Pi SD Card Image Manager. Contribute to 1stcall/sdm development by creating an account on GitHub.
github.com
I haven't done much with it for a while.  I'm currently teaching myself sveltekit with a veiw to build a sdm as a web server that returns a customised image, build using a cluster of rasberry pis.

1stcall, 2023-11-30:
https://github.com/1stcall/sdmui/tree/development
https://sdmui.1stcall.uk

gitbls, 2023-12-05:
I just pushed plugins/extractfs with V10.2. It's not included in EZsdmInstaller, you'll need to pull it down some other way.
Usage:
```sh
sdm --runonly plugins --directory / --oklive --plugin extractfs:"verbose|img=/path/to/img.img|bootfs=/path/to/bootfs|rootfs=/path/to/rootfs"
```
It's a bit clunky as a plugin, but I wasn't prepared to add it into the core sdm.

Let me know your thoughts.

1stcall, 2023-12-14:
> Why did you suggest changing the declare -x? Was there an issue with it as it was?
Yes.  SDMPT is already exported.  Therefore, in bash, if you re export it within a function, it will only export it to sub processes or functions called from within that function.

gitbls, 2024-01-05:
`extractfs` is in sdm V11 as a burn plugin. Much simpler! Please take it for a spin.

closed 2024-02-04 by gitbls