gitbls/sdm #140: Overwrite an already-customized IMG file

huaminghuangtw, 2023-11-28:
Will `sdm --customize <img_file>` completely overwrite the already-customized `<img_file>` if it's run for the second time?

From my observation, it seems not the case. - I tried to run the following code snippet twice, trying to overwrite the existing customized IMG file `<img_file>` from the first call of `sdm --customize <img_file>` . Surprisingly, the custom First Boot script `init_setup.sh` was executed TWICE with the customized IMG file from the second call of `sdm --customize <img_file>` 

```
args=(
        --customize "${img_file}"
		
        # Add user and password
        --plugin user:"adduser=${USERNAME}|password=${PASSWORD}"

        # Run a cutom script during the First Boot of the system 
        # stdout and stderr will be written into a log file and to the console
        --plugin runatboot:"script=./init_setup.sh|args=${SYSTEM_NAME}"

    )
    sudo sdm "${args[@]}"
```

gitbls, 2023-11-28:
I'm afraid that I don't understand your question. sdm NEVER overwrites an IMG, even on the first customization.

Rather, it ALWAYS customizes/updates the IMG _in place_.

sdm looks to see if an IMG has already been customized by checking for the existence of  /etc/sdm/custom.ized in the IMG. This file is touched by sdm at the end of Phase 0. If the file exists, sdm will notify that the IMG has already been customized and ask if you want to redo. You can avoid the check and always redo the customization by using the `--redo-customize` switch.

In your example, if you rerun the same customization on the IMG, the `runatboot` plugin will indeed cause init_setup.sh to be run twice.

The `runatboot` plugin could certainly check for a script already run, but since you can easily work around this by simply redoing the customization on an un-customized IMG, I don't really see a compelling argument to change this behavior.

huaminghuangtw, 2023-11-28:
> I'm afraid that I don't understand your question. sdm NEVER overwrites an IMG, even on the first customization.
>
> Rather, it ALWAYS customizes/updates the IMG _in place_.
>
> sdm looks to see if an IMG [...]

@gitbls Now I understand. Thank you very much for the detailed explanation! In my case, I think I would always copy an unmodified, official IMG file to a separate directory and customize with this IMG file, to avoid `init_setup.sh` being run twice.

gitbls, 2023-11-28:
Yes, that's what I do. I keep an unmodified copy handy, and copy it whenever I need to do a customization. 

It's MUCH cleaner this way.