gitbls/sdm #144: sdm 10.0 and earlier incorrect /boot/firmware handling

gitbls, 2023-12-02:
A fairly serious bug has been uncovered in sdm with respect to Bookworm. Bookworm IMGs customized with 'apt upgrade' and kernel 6.1.58 gets installed, that is, were updated to 6.1.58-1+rpt2 kernel during an IMG customize, are not really running the 6.1.58 kernel.

If the system was booted and runnng when the upgrade to 6.1.58 was done, there is no problem.

**IDENTFICATION**

You can identify problematic systems that are booted and running by:
* Determine the latest kernel version that is installed: `sudo apt search linux-image | grep installed`
* Determine what kernel is running: `uname -r`

If the version of the kernel running does NOT match the latest installed kernel, the system needs to be repaired.

**REPAIRING THE RUNNING SYSTEM**

* Determine which kernels are installed: `sudo apt search linux-image | grep 6.1.58 | grep installed`
* Update each installed kernel: `sudo apt install --reinstall linux-image 6.1.0-rpi6-rpi-v8 linux-headers-6.1.0-rpi6-rpi-v8`

Note that on 64-bit systems you should update both the -v8 and -2712 kernels.

**REPAIRING SDM**

* Install the latest sdm (V10.1) on your host: `curl -L https://raw.githubusercontent.com/gitbls/sdm/master/EZsdmInstaller | bash`
* Delete and recreate any affected customized IMGs, or alternatively use the above REPAIRING THE RUNNING SYSTEM while doing a `--explore` on the IMG with sdm 10.1

**REPAIRING CUSTOMIZED IMGs**

First update sdm on the host per above, then:
* `sdm --explore /path/to/img.img
* Determine which kernels are installed: `sudo apt search linux-image | grep 6.1.58 | grep installed`
* `apt install --reinstall linux-image 6.1.0-rpi6-rpi-v8 linux-headers-6.1.0-rpi6-rpi-v8`

**WHAT WAS BROKEN**

Long story short: sdm was not updated correctly for /boot/firmware in Bookworm

gitbls, 2023-12-04:
This is very strange: Note, selecting 'raspberrypi-kernel' instead of 'linux-image'
I've never seen that. I'd need to see /etc/sdm/history and /etc/sdm/apt.log from the IMG to have any hope of sorting out what's going on.

I'm also quite puzzled by the 2nd part, there hasn't been any kernel with 6.2.0-37 that I'm aware of. Where did that come from?

If possible, I'd encourage you to rebuild the IMG using sdm V10.1. If that's not possible, this is going to require much more investigation and would be much easier if you could email me the above-mentioned files at the email address in the sdm README. Thx.

Jackbennett, 2023-12-04:
This is a neat project but I had the same issue initially v10 and then from v10.1 on a fresh download, had to fix it in a pinch so 
 going back to the bullseye images worked.

This was being customized from a container on the RPI and mounting an SD card via USB. So if there's any scripts assuming the running linux is like the SD card that's not going to line up.

gitbls, 2023-12-04:
Inspired by the feedback here, I took yet another look into this issue. It appears that for some reason, when doing a customize, the linux-image and linux-headers packages are not updated UNLESS you add `--apt-dist-upgrade` to your `--customize` command line.

Without `--apt-dist-upgrade` the system boots using the original bookworm kernel, 6.1.54, and that can be updated using standard update/upgrade once the system is up and running.

I didn't notice this before because my sdm customization script always uses `--apt-dist-upgrade`. My recommendation is that you do so as well.

gitbls, 2023-12-04:
@rosesyrett Can you please test the following on that IMG with V10.1?
* `sdm --explore` into the IMG
* `apt update`
* `apt dist-upgrade`
* `apt search linux-image | grep 6.1.58 | grep installed

Presumably the `apt search` will show what you just upgraded, but I'm not 100% sure without testing/no time for that today. 😳

Jackbennett, 2023-12-06:
> here, I took yet another look into this issue. It appears that for some reason, when doing a customize, the linux-image and linux-headers packages are not updated UNLES

Thanks for the reply, please disregard because I don't think it adds anything. but from my console `uname -r` is a talos container kernel not physical system. That may not matter anyway.

gitbls, 2023-12-06:
> Thanks for the reply, please disregard because I don't think it adds anything. but from my console `uname -r` is a talos container kernel not physical system. That may not matter anyway.

I might be misunderstanding what you said, but in the sdm container `-uname` shows the host system, not the OS in the container that we're customizing. I could tell by looking at the files in /boot and /boot/firmware.

gitbls, 2024-02-04:
Closing this issue as it's ancient history now

closed 2024-02-04 by gitbls