gitbls/sdm #145: Manually enabling SSH

huaminghuangtw, 2023-12-04:
I know SSH is by default enabled automatically by sdm. But somehow it turned out SSH was not enabled in one of the SD cards I burned into using sdm.

My question is: is there a way to manually enable SSH in sdm?

Thanks!

gitbls, 2023-12-04:
I would like to understand how *somehow* SSH was not enabled.

Please post or email me the file /etc/sdm/history from the system in question.

gitbls, 2023-12-04:
Ah, just discovered an issue with setting up ssh on bookworm. The location for the 'ssh' file has moved from /boot/ssh to /boot/firmware/ssh

This will be fixed in sdm V10.2. In the meantime, you can easily `--explore` the IMG or the burned disk and touch the file yourself.

Apologies for the duff.

huaminghuangtw, 2023-12-07:
> /etc/sdm/history

Hi, I couldn't find the file `/etc/sdm/history`. Could you please point me out what I missed?
```
root@my-rpi:~# cat /etc/sdm/history
cat: /etc/sdm/history: No such file or directory
```

stb68, 2023-12-07:
> root@my-rpi:~# cat /etc/sdm/history
> cat: /etc/sdm/history: No such file or directory

Hi, I think you need to use sudo
   `sudo cat /etc/sdm/history`

huaminghuangtw, 2023-12-07:
> > root@my-rpi:~# cat /etc/sdm/history
> > cat: /etc/sdm/history: No such file or directory
> 
> Hi, I think you need to use sudo `sudo cat /etc/sdm/history`

Hi, still the same unfortunately...
```
root@my-rpi:~# sudo cat /etc/sdm/history
cat: /etc/sdm/history: No such file or directory
```

gitbls, 2024-01-05:
Closing due to no further activity. OP, please feel free to re-open this or a new issue as needed.

closed 2024-01-05 by gitbls