gitbls/sdm #152: sdm --burn fails when using /dev/disk/by-uuid rather than /dev/sd?

1stcall, 2023-12-18:
+ sdm --burn /dev/disk/by-uuid/7909438b-2884-423e-aa05-577fdea75c97 --expand-root --hostname rescue.1stcall.uk --plugin trim-enable --reboot 5 ./images/2023-12-05-raspios-bookworm-arm64-cm4.img
* Burn './images/2023-12-05-raspios-bookworm-arm64-cm4.img' (4.8GB, 4.5GiB) to '/dev/disk/by-uuid/7909438b-2884-423e-aa05-577fdea75c97'...
dd if=./images/2023-12-05-raspios-bookworm-arm64-cm4.img of=/dev/disk/by-uuid/7909438b-2884-423e-aa05-577fdea75c97 status=progress bs=16M iflag=direct
4764729344 bytes (4.8 GB, 4.4 GiB) copied, 118 s, 40.4 MB/s
285+1 records in
285+1 records out
4790960128 bytes (4.8 GB, 4.5 GiB) copied, 121.937 s, 39.3 MB/s
cat: /sys/block/disk/by-uuid/7909438b-2884-423e-aa05-577fdea75c97/size: No such file or directory
cat: /sys/block/disk/by-uuid/7909438b-2884-423e-aa05-577fdea75c97/disk/by-uuid/7909438b-2884-423e-aa05-577fdea75c972/size: No such file or directory
> Expand Root: Expand partition 'disk/by-uuid/7909438b-2884-423e-aa05-577fdea75c972' on device '/dev/disk/by-uuid/7909438b-2884-423e-aa05-577fdea75c97' from (0.0GB, 0.0GiB) to (-30.6GB, -28.5GiB)
parted: invalid option -- '1'
Usage: parted [-hlmsfv] [-a<align>] [DEVICE [COMMAND [PARAMETERS]]...]
? parted resizepart failed with status 1

gitbls, 2023-12-18:
While I knew that /dev/disk/by-uuid existed, I've never looked into it.

It seems a bit flakey to me, in that not all disks are there all the time. For instance, it shows 
```
pw/dev/disk/by-uuid# l
total 0
drwxr-xr-x 2 root root 100 Dec 18 12:29 .
drwxr-xr-x 8 root root 160 Dec 16 11:48 ..
lrwxrwxrwx 1 root root  10 Dec 18 12:29 7909438b-2884-423e-aa05-577fdea75c97 -> ../../sda2
lrwxrwxrwx 1 root root  10 Dec 16 11:48 bfcd917c-874a-404b-ae7b-613e22232a34 -> ../../sdb1
lrwxrwxrwx 1 root root  10 Dec 18 12:29 CFDB-695B -> ../../sda1
```
but when I plug in an SD card, here's what I get:
```
pw/dev/disk/by-uuid# l
total 0
drwxr-xr-x 2 root root 100 Dec 18 12:31 .
drwxr-xr-x 8 root root 160 Dec 16 11:48 ..
lrwxrwxrwx 1 root root  10 Dec 18 12:31 7909438b-2884-423e-aa05-577fdea75c97 -> ../../sdd2
lrwxrwxrwx 1 root root  10 Dec 16 11:48 bfcd917c-874a-404b-ae7b-613e22232a34 -> ../../sdb1
lrwxrwxrwx 1 root root  10 Dec 18 12:31 CFDB-695B -> ../../sdd1
```
IOW the sda1, sda2 entries disappear and are replaced by sdd1 and sdd2, even though sda1, and sda2 remain mounted in the system.

Secondly, all the entries in `by-uuid` are partitions (sdd1, sdd2) rather than disks (sdd). sdm needs a disk for burning, not a partition. This would seem to make `by-uuid` less than useful for sdm.

gitbls, 2024-01-05:
Closing due to no further activity.

closed 2024-01-05 by gitbls