gitbls/sdm #153: raspiconfig serial option prompts for user input

echo-bravo-yahoo, 2023-12-20:
Setting the serial interface to enabled (`--plugin raspiconfig:"serial=1"`) causes sdm to display the raspiconfig text UI, and the user has to interactively answer two questions:
- Would you like a login shell to be accessible over serial? _(this does not seem implied by asking sdm to turn on the serial interface)_
- Would you like the serial port hardware to be enabled? _(this seems implied by asking sdm to turn on the serial interface)_

It seems to me like this might be worth exposing in sdm as two options. I haven't tested against any images before Bookworm to see if this behavior changed in raspi-config.

Would you be interested in a PR to address this? I'm getting so much use out of SDM that I would like to contribute back.

gitbls, 2023-12-20:
I took a quick look at this, and it appears that the issue is raspi-config itself. The questions are asked regardless of the `nonint` setting on the raspi-config command line. 😵

Off to file a bug on raspi-config and (for now) handle serial-setting in sdm. Sigh.

echo-bravo-yahoo, 2023-12-20:
> As far as contributing, I'd be much more interested in contributions of new plugins that add value to sdm and users.

Do you have any suggestions? So far, all of my use cases are handled by sdm well with the exception of:
- downloading a raspi base image before using customize
- installing nodeJS (the version in apt is very, very old, so I'm fetching/installing a newer version)
- provisioning AWS IoT device certs

gitbls, 2023-12-20:
The first one (download base img) would be difficult to do as a plugin, since plugins aren't called until the IMG is already verified as existing. I've thought about adding this to sdm (without using a plugin), but thus far I've landed on the "it's easier for people to download their own, especially since sufficiently big disks are quite cheap these days.

I haven't looked into nodeJS, so don't understand the issues there, or the complexity in installing a newer version. Is this a point-in-time problem, or something that's likely to continue forever? What would the complexity be in doing it as a plugin?

The AWS iOT device certs sounds interesting. Can you point me at some info on it?

At the moment I'm a bit sad about the raspi-config `serial` situation, as a proper fix appears to be a bit complex, because the settings are different for a Pi5 vs all other Pis. Since the target Pi isn't known until boot time, that makes setting the serial config properly for the first boot extremely problematic in the general case. Thinking about how to best handle this...😕

echo-bravo-yahoo, 2023-12-20:
> I haven't looked into nodeJS, so don't understand the issues there, or the complexity in installing a newer version. Is this a point-in-time problem, or something that's likely to continue forever? What would the complexity be in doing it as a plugin?

For most raspis, I suspect it's as easy as using apt to pull down a fresh build. Unfortunately for the raspi 0, the open js foundation discontinued support for armv6 quite a while ago, and so there aren't supported first-party builds. They do still build the binaries, but they don't list them in apt. This could be solved either with a plugin that allows you to add new apt sources (since they do list the builds in an unofficial apt source) or by downloading the binary directly, untarring it, and making a few symlinks. I'm currently doing the latter because it was easier to bootstrap, but I think the former is more useful as a plugin.

> The AWS iOT device certs sounds interesting. Can you point me at some info on it?

Sure! You can find [all the docs here](https://docs.aws.amazon.com/iot/). The developer guide is probably the best starting place, but the very high level overview of the service is:
- AWS IoT is [an MQTT broker](https://en.wikipedia.org/wiki/MQTT#Overview); MQTT is a very light-weight pub/sub protocol.
- AWS IoT's implementation is stateful, and each Thing (device) has a Shadow (state). The Thing's Shadow contains the desired state of the device and the last reported state of the device. This allows parts of the distributed system to request state changes even when a device is offline, and allows offline devices to set themselves to the correct state when they're reconnected to the system.
- Provisioning a Thing can happen in a staggering number of ways, but the end result is typically that the device has a private key and a certificate which are associated in AWS IoT to a policy. The Thing can then connect with its credentials and take the actions allowed in its policy; usually subscribe and publish to various MQTT channels. This is the part that I've written some glue for. It may or may not be a good fit for a plugin because it has a dependency on the AWS CLI or SDK, and I don't see how you'd sidestep that.

> At the moment I'm a bit sad about the raspi-config `serial` situation, as a proper fix appears to be a bit complex, because the settings are different for a Pi5 vs all other Pis. Since the target Pi [...]

Can the first boot script interrogate the image to determine how to configure? E.g., if this is on the PATH, configure this way?

gitbls, 2023-12-20:
> Can the first boot script interrogate the image to determine how to configure? E.g., if this is on the PATH, configure this way?

Yes, of course, BUT: This means that the serial port can't be enabled during sdm FirstBoot, since enabling the serial port changes one or both of cmdline.txt and config.txt, which are read LONG before sdm FirstBoot runs. Having a working serial port in the system First Boot seems like a really good idea for debugging, etc.

Thus far my best idea is to configure the serial port for a Pi5 or not-Pi5. Still has a few problems, but it can be configured at Burn time, when theoretically the target system is known. If this pans out in my thought process, I'll remove `serial` from the raspiconfig plugin, and add a `serial` plugin that does this, and only this. (As i'm typing this, the more I type, the more I like it).

As far as the AWS config, thanks for the explanation. I think this would be interesting. Since there are a "staggering number of ways" 😲, it seems that a best-choice prescriptive approach would be good. Since there are AWS CLI/SDK dependencies, I'd expect the plugin to install those required bits as well. There are a couple of plugins that install bits not available in apt, such as btwifiset, imon, knockd, ndm, and pistrong.

gitbls, 2024-01-05:
sdm V11 has the `serial` plugin, which addresses the question in the original post (providing more granular control over the serial configuration) and the secondary issue of getting the serial port configured for the wrong Pi5 vs not-Pi5 target system.

gitbls, 2024-02-03:
Closing due to lack of activity. OP, please re-open if futher issues noted. Thx!

closed 2024-02-03 by gitbls