gitbls/sdm #154: named system plugin fails on re-customization

echo-bravo-yahoo, 2023-12-20:
When invoking customize for a second time on a given image, named system plugins fail the customization attempt.

E.g., `--plugin system:"name=largeSwap|swap=4096` will succeed on the first customization and fail on all subsequent customizations until either the plugin is renamed (which presumably creates duplicate scripts?) or the image is deleted and downloaded fresh.

The error looks like:
```
* Plugin system: Start Phase 0
> Plugin system: Keys/values found:
   name: largeSwap
   swap: 4096
? Plugin system: '/mnt/sdm/etc/sdm/assets/system-largeSwap' already exists; Add argument 
  'name=somename' to this system plugin invocation
? Plugin 'system' exited with failure
```

gitbls, 2023-12-20:
That's a feature at the moment. If it were to allow it to be rerun, it would either have to a) simply ignore all the settings, or b) redo them all.

a) is relatively easy. But b) would require testing/validating/fixing a lot of code to ensure that it's handled properly, both in this plugin and every other plugin.

Personally, I never rerun a customize on an already-customized IMG. It's much simpler to rerun the customize and guarantee that everything is correct.

echo-bravo-yahoo, 2023-12-20:
I see. Re-downloading the image is time-consuming, so I've been iteratively mutating it in place; I'll change my workflow to making a copy of the base image, customizing that, then trashing and re-creating the customized copy next time I need to mutate.

Do you think it's worth changing the plugin error message to make it clearer it's not intended to be idempotent? I believe sdm already detects re-customization (it provides a y/n prompt for it), but the current error message is about providing a name.

gitbls, 2023-12-20:
> I see. Re-downloading the image is time-consuming, so I've been iteratively mutating it in place; I'll change my workflow to making a copy of the base image, customizing that, then trashing and re-creating the customized copy next time I need to mutate.
> 
Yea, I keep a "library" of "virgin" IMGs, and always copy the one I want at the start of customization. Normally not a big deal except when debugging a new feature or problem and I have to redo it far too many times.

> Do you think it's worth changing the plugin error message to make it clearer it's not intended to be idempotent? I believe sdm already detects re-customization (it provides a y/n prompt for it), but the current error message is about providing a name.
>
Yes, this is a good idea. Added it to my todo list. Thx

gitbls, 2023-12-24:
Can you please provide a simple repro for this? I've checked the code, and it will only fail on an un-named system plugin use if the file $SDMPT/etc/sdm/assets/system already exists, and indeed, my simple test worked as expected:
```
sdm --customize $1 --csrc /rpi --poptions noupdate,noupgrade,noautoremove --plugin system:"fstab=/rpi/etc/fstab.lan
...
> Run Plugins Phase '0'
> Run Plugin 'system' (/mnt/sdm/usr/local/sdm/plugins/system) Phase 0 with arguments: 
  'fstab=/rpi/etc/fstab.lan'
* Plugin system: Start Phase 0
> Plugin system: Keys/values found:
   fstab: /rpi/etc/fstab.lan
> Plugin system: Create directory /mnt/sdm/etc/sdm/assets/system
> Plugin system: Create directory /mnt/sdm/etc/sdm/assets/system/fstab
> Plugin system: Copy '/rpi/etc/fstab.lan' to /mnt/sdm/etc/sdm/assets/system/fstab
* Plugin system: Complete Phase 0
* Phase 0 Completed
...
```

timrowledge, 2023-12-25:
Ah - a thought; this might be a case where I ran a customize on an IMG that already had been whacked. Obvioulsy I don't have the original error message handy, because that would be far too helpful. I will attempt to try from a completely clean file sometime soon.

gitbls, 2024-01-05:
sdm V11 will produce a more helpful message in this scenario.

gitbls, 2024-02-03:
Closing due to lack of activity.

closed 2024-02-03 by gitbls