gitbls/sdm #156: Minimal steps to create a bootable Debian (bookworm) RPi image?

adrelanos, 2023-12-22:
Hi,

could you share please the minimal shell commands required to create a bootable Debian (bookworm) RPi image?

This might benefit various system OS image builds tools:

* https://github.com/grml/grml-debootstrap/issues/114
* https://github.com/systemd/mkosi/discussions/2103

gitbls, 2023-12-22:
Hello. The [README](https://github.com/gitbls/sdm) does exactly that. If you feel that something is missing there, please explain.

adrelanos, 2023-12-22:
It uses sdm. The projects I've linked couldn't (or won't) really use the full sdm. That really doesn't fit. (grml-debootstrap maybe because that's bash but mkosi highly unlikely since implemented in python and sdm would be a huge dependency just for RPi support which is just another architecture.)

This is kinda the pseudo commands I am looking for however actually working to produce a bootable RPi image.

```
parted -s "${TARGET}" 'mklabel msdos'
parted -s "${TARGET}" 'mkpart primary ext4 4MiB 100%'
parted -s "${TARGET}" 'set 1 boot on'

chroot "${MNTPOINT}" grub-install --target=arm64-efi --efi-directory=/boot/efi --bootloader-id=debian --recheck --no-nvram --removable
```

Or...

> 1. create partition table this way
> 2. install bootloader
> 3. install these packages
> 4. copy these files into /boot

Sorry for the confusion. This isn't so much a question about sdm but about the RPi bootable image creation process generally.

gitbls, 2023-12-22:
sdm does none of that. It's strictly about customizing an already-existing IMG (ie the ones provided by RaspberryPI Labs). Seems like you want something to actually create the IMG.

closed 2023-12-23 by gitbls