gitbls/sdm #159: plugin user gid

jkt628, 2023-12-26:
in my plugin configuration file i have this line above others

```text
user:addgroup=foo|gid=4321
```

but sdm does not honor the configured gid, instead it grabs the next free gid above 1000.

gitbls, 2023-12-27:
According to the Plugin doc at https://github.com/gitbls/sdm/blob/master/Docs/Plugins.md#user, the syntax is
```
addgroup=groupname,gid
```
So, in your example, you should use
```
user:addgroup=foo,4321
```
The `gid` argument is used to put the user into the specified group. (It appears that there may be a buglet with this argument, will have a look at it.)

jkt628, 2023-12-27:
yep, missed that, sorry.  it works.

closed 2023-12-27 by jkt628

gitbls, 2023-12-27:
> yep, missed that, sorry. it works.

Excellent. There's a lot of documentation, and easy to miss things. I do that myself sometimes! 😵