gitbls/sdm #160: Desktop setting problem and enhancement suggestion

timrowledge, 2024-01-13:
(For future readers: derived from the discussion at https://forums.raspberrypi.com/viewtopic.php?t=363246)

I noticed that when building IMGs for the Bookworm OS release, the lxde plugin puts the pcmanfm.conf file in ~/.config/pcmanfm, whereas it needs to go in ~/.config/pcmanfm/LXDE-pi

I'd also suggest that the filer bookmarks file ought to be included in the list of options; it defines the places to be shown in the places list of the filer window. It lives in ~/.config/gtk-3.0

With both these files in the appropriate place I was able to configure my system to not-simplified-ui, show places, have the list of bookmarked places, and be happy.

gitbls, 2024-01-13:
Will look into it. Thx for the details.

gitbls, 2024-01-13:
Fixing pcmanfm.conf will be done. 

Question about bookmarks for you. I added a couple, and they did not show up in ~/config/gtk-3.0. Instead they were in the file ~/.gtk-bookmarks.

Can you re-check your bookmarks and where they end up? If it's not in the same place as mine, we need to sort out what's different between our two systems. 

Thx!

timrowledge, 2024-01-13:
Hi-

> On 2024-01-13, at 10:39 AM, Benn ***@***.***> wrote:
> 
> Question about bookmarks for you. I added a couple, and they did not show up in ~/config/gtk-3.0. Instead they were in the file ~/.gtk-bookmarks.
> 
> Can you re-check your bookmarks and where they end up? If it's not in the same place as mine, we need to sort out what's different between our two systems.

Well, that's odd. I have identical bookmarks file in both places. Gronk? So to see which one appears to be actually used I just added another bookmark using the file UI; the .config/gtk-3.0/bookmarks file got updated, but the .gtk-bookmarks didn't. So I guess that tells us ..... something? Don't you just love how convoluted this stuff has become?

The only info I've found online suggests that they changed from the .gtk-bookmarks a few years ago. How that explains the file appearing in Bookmark OS release system I don't know; tools not all being updated maybe. The pcmanfm man page mentions support for bookmarks but nothing about how, or which file is used or indeed anything useful.

tim
--
tim Rowledge; ***@***.***; http://www.rowledge.org/tim
"How many Teela Browns does it take to change a lightbulb?” 
"Stupid question."

timrowledge, 2024-01-13:
Just spotted - 
https://www.remembertheusers.com/2016/05/0161-gtk-bookmarks.html

"The general problem with GTK bookmarks is the base file that is used. The file is stored in the user’s $HOME as .gtk-bookmarks. In GTK 3, because well, you know, GTK 3 has to be different, that location is moved to $HOME/.config/gtk-3.0/bookmarks. In both GTK 2 and GTk 3 the file is exactly the same."

tim
--
tim Rowledge; ***@***.***; http://www.rowledge.org/tim
Oxymorons: Almost exactly

gitbls, 2024-01-13:
Thx. for now will stuff the file into both places, because.

gitbls, 2024-01-13:
Huh. Just updated the code to copy the bookmarks to both locations. Booted the system, added a new bookmark interactively, and it only got added to ~/.config/gtk-3.0/bookmarks

Nonetheless, I'm going to leave it copying the file to both locations for now. Weird.

timrowledge, 2024-01-13:
> On 2024-01-13, at 12:33 PM, Benn ***@***.***> wrote:
> 
> 
> Huh. Just updated the code to copy the bookmarks to both locations. Booted the system, added a new bookmark interactively, and it only got added to ~/.config/gtk-3.0/bookmarks
> 
> Nonetheless, I'm going to leave it copying the file to both locations for now. Weird.

Yeah. Life - loathe it or ignore it, you can't love it (attribt. Marvin the paranoid android)

tim
--
tim Rowledge; ***@***.***; http://www.rowledge.org/tim
Strange OpCodes: RLBM: Ruin Logic Board Multiple

gitbls, 2024-02-03:
Closing due to lack of activity. OP, please re-open if needed. Thx!

closed 2024-02-03 by gitbls