gitbls/sdm #161: Desktop wallpaper setting issue

timrowledge, 2024-01-13:
This is simply a record of a problem noticed when changing settings; it almost certainly isn't an sdm problem but it is faintly possible that some interaction with the sdm process and Wayland is involved.

When using Wayland and wayfire to have a VNC connection, if the desktop wallpaper has been set to 'none (so that a plain background colour is used), then it appears that vnccannot connect properly. An error message is displayed - 
[image]

Using ssh to log in and either delete the ~/.config/pcmanfm/desktop-items-0.conf files, or edit it to set 'wallpaper_mode=crop', followed by rebooting, seems to work. Changing the wallpaper mode setting from the running desktop works *until you reboot*.

Very strange, and TBH, somebody else's problem.

gitbls, 2024-01-13:
Yes, saw that here as well. Was not able to fix it with wallpaper_mode=crop. VNC with wayland still has some newbie pains, apparently.

timrowledge, 2024-01-13:
Just for completeness, tried on aPi 4, no different  

> 
> 
> Yes, saw that here as well. Was not able to fix it with wallpaper_mode=crop. VNC with wayland still has some newbie pains, apparently.

tim
--
tim Rowledge; ***@***.***; http://www.rowledge.org/tim
There *are* worse crimes than burning books.  One of them is not reading them.