gitbls/sdm #162: Setting Wayland/X11 choice

timrowledge, 2024-01-16:
I suspect some addition to the raspiconfig plugin is needed to add the `sudo raspi-config nonint do_wayland <W1/W2>` command.

It *might* be practical to hook this up to the choice in the nvc plugin such that using `--plugin vnc:"wayvnc=1600x1200" ` works out that Wayland may need to be chosen, whereas `--plugin vnc:"realvnc=1600x1200" ` means choose X11.

timrowledge, 2024-01-16:
D'Oh; just discovered the 'graphics' plugin, though my point about connecting the X11/Wayland option to use of the vnc choice may still be interesting

gitbls, 2024-01-16:
I'm not a big fan of having functionality in two places, and I like having all the VNC configuration in one place, and that's in the vnc plugin.

As far as enabling wayland, yes I punted on that because it's not clear to me that it's of use.

If you want wayland you can get it by default using the "with desktop" version.

For X11, you can either start with Lite and install X11, or you can start with a "with desktop" version, which defaults to wayland, and switch it to X11.

What do you see as the need for a way to enable wayland?

timrowledge, 2024-01-16:
> On 2024-01-15, at 7:51 PM, Benn ***@***.***> wrote:
> 
> 
> I'm not a big fan of having functionality in two places, and I like having all the VNC configuration in one place, and that's in the vnc plugin.

I'd generally agree with that preference. I see a potential for a conflict that makes for a nasty surprise if somebody used graphics=X11 and vnc"wayvnc=yyyyxzzzz". Perhaps simply checking that there is not a clash would be more sensible?

> As far as enabling wayland, yes I punted on that because it's not clear to me that it's of use.
>
> If you want wayland you can get it by default using the "with desktop" version.
>
> For X11, you can [...]

I guess since RP have decided that defaulting to Wayland is The Thing for Pi4/5, there isn't a need to set it, for those models. I'm not certain if it is an option for older models like 3's & 3+ and maybe it shouldn't be. (I'm not actually a huge fan of it with my current experience.)

If one started from a Lite and used graphics=X11, would LXDE be installed, or would it need to be added to the apps plugin list? I'm a bit twitchy about that stuff ever since trying pretty much that process with an x64 xubuntu-box and ending up with an annoying system where vnc regularly just locks up.

tim
--
tim Rowledge; ***@***.***; http://www.rowledge.org/tim
Borrow money from pessimists--they don't expect it back.

gitbls, 2024-01-16:
sdm assumes that the person invoking sdm (you) knows exactly what settings they're using. 🤔

In other words, sdm currently does literally NO checking of cross-plugin dependencies, I'll think about whether this is a) worthwhile, and b) doable.

as far as `graphics=x11`, no this does not cause ANYTHING to be installed. This is by design, since there are so many different X11 desktops that could be installed. For instance, I use xdm for my display manager (login screen) and icewm for a window manager. Simple and small.

So, yes, you would need to add `task-lxde-desktop`  or `lxde` (I think that should do it) to the apps list. Not sure which will do what you want, but you can easily try each and see what works for you.

The cool thing about sdm is that it makes it SUPER-simple to try as many different X11 configurations as you have time to play with. 😂

timrowledge, 2024-01-17:
> On 2024-01-16, at 7:09 AM, Benn ***@***.***> wrote:
> 
> 
> sdm assumes that the person invoking sdm (you) knows exactly what settings they're using. 🤔

Oh my, that's dangerous. I'd never be so brave as to assume users have a clue :-)

Right, well I've been experimenting to see what is needed to get an X11 setup on a Pi 5. It works fine but it (unsurprisingly I guess) does rely on the order of the commands. Having the realvnc thing at the end doesn't work because it overwrites the desktop config files one had so carefully copied previously. Whoops.

The bookmarks file has to be name .gtk-bookmarks and lives in the home directory instead of the .config tree. The panel settings stuff is in .config/lxpanel/LXDE-pi/panels/panel The desktop-items-0.conf file work fine even with wallpaper_mode=none.

tim
--
tim Rowledge; ***@***.***; http://www.rowledge.org/tim
Oxymorons: Found missing

timrowledge, 2024-01-17:
Hmph, an experiment with using the same script with the smaller non-full IMG & X11 failed; it tried to open a vnc connection but failed, and I couldn't ssh into it. I'll check a wayland setup as well for completeness.

tim
--
tim Rowledge; ***@***.***; http://www.rowledge.org/tim
The next generation of computers will have a "Warranty Expired" interrupt.

gitbls, 2024-01-17:
Happy to have a look at what you're doing with X11 if you'd like. It's not difficult or complex unless you're trying to get tricky in some weird way. 🤣

timrowledge, 2024-01-17:
> On 2024-01-16, at 5:37 PM, Benn ***@***.***> wrote:
> 
> 
> Happy to have a look at what you're doing with X11 if you'd like. It's not difficult or complex unless you're trying to get tricky in some weird way. 🤣

Does ignoring an error message count as being tricky?

This is trying out with the not-full IMG 023-12-05-raspios-bookworm-arm64.img

=======
2024-01-17 12:33:09 apt-get -qq -y upgrade

E: You don't have enough free space in /var/cache/apt/archives/.
======

Do I need to extend something? Let's try adding a -extend -xmb clause ... yes, that seems to solve that issue. Now burn it and try out...

Hurr. No sign of it booting at all. Try a burn again and --extend-root? Nope.

The customise script -
========
#!/bin/bash
# Working on X11 config stuff
# This one intended for the not-full IMG

# bookmarks file has to be a specific name, differnet to that for Wayland. Of course.
cp /home/tim/Documents/SDM-files/bookmarks /home/tim/Documents/SDM-files/.gtk-bookmarks

sudo sdm --customize \
--extend --xmb 2048 \
--plugin user:"deluser=pi" \
--plugin user:"adduser=tim|password=XXXXXX" \
--plugin system:"name=fs1|fstab=/home/tim/Documents/SDM-files/fstab-additional" \
--plugin L10n:host \
--plugin disables:"bluetooth|piwiz|triggerhappy" \
--plugin raspiconfig:"boot_behaviour=B4" \
--plugin graphics:"graphics=X11" \
--plugin vnc:"realvnc=1600x1200" \
--plugin mkdir:"dir=/home/tim/Documents/squeak|chown=tim:tim" \
--plugin mkdir:"dir=/home/tim/.local|chown=tim:tim" \
--plugin mkdir:"dir=/home/tim/.local/share|chown=tim:tim" \
--plugin mkdir:"dir=/home/tim/.local/share/applications|chown=tim:tim" \
--plugin copydir:"from=/home/tim/Documents/CleanSqueakInstall/|to=/home/tim/Documents/squeak" \
--plugin copyfile:"from=/home/tim/Documents/SDM-files/squeak.conf|to=/etc/security/limits/d/squeak.conf|mkdirif|chown=tim:tim" \
--plugin copyfile:"from=/home/tim/Documents/SDM-files/userapp-squeak.desktop|to=/home/tim/.local/share/applications|mkdirif|chmod=644|chown=tim:tim" \
--plugin mkdir:"dir=/home/tim/.config|chown=tim:tim" \
--plugin mkdir:"dir=/home/tim/.config/pcmanfm|chown=tim:tim" \
--plugin mkdir:"dir=/home/tim/.config/pcmanfm/LXDE-pi|chown=tim:tim" \
--plugin mkdir:"dir=/home/tim/.config/lxpanel|chown=tim:tim" \
--plugin mkdir:"dir=/home/tim/.config/lxpanel/LXDE-pi|chown=tim:tim" \
--plugin mkdir:"dir=/home/tim/.config/lxpanel/LXDE-pi/panels|chown=tim:tim" \
--plugin mkdir:"dir=/home/tim/.config/libfm|chown=tim:tim" \
--plugin mkdir:"dir=/home/tim/.config/gtk-3.0|chown=tim:tim" \
--plugin copyfile:"from=/home/tim/Documents/SDM-files/pcmanfm.conf|to=/home/tim/.config/pcmanfm/LXDE-pi|chmod=644|chown=tim:tim" \
--plugin copyfile:"from=/home/tim/Documents/SDM-files/desktop-items-0.conf|to=/home/tim/.config/pcmanfm/LXDE-pi|chmod=644|chown=tim:tim" \
--plugin copyfile:"from=/home/tim/Documents/SDM-files/libfm.conf|to=/home/tim/.config/libfm|chmod=644|chown=tim:tim" \
--plugin copyfile:"from=/home/tim/Documents/SDM-files/.gtk-bookmarks|to=/home/tim|chmod=644|chown=tim:tim" \
--plugin copyfile:"from=/home/tim/Documents/SDM-files/panel|to=/home/tim/.config/lxpanel/LXDE-pi/panels|chmod=644|chown=tim:tim|mkdirif" \
--aptcache 192.168.51.237 \
--reboot 20 \
/home/tim/Documents/SDM-files/2023-12-05-raspios-bookworm-arm64.img
===========

and burn -
=====
sudo sdm --burn /dev/sdb --host ucalegon --expand-root /home/tim/Documents/SDM-files/2023-12-05-raspios-bookworm-arm64.img
======

This isn't hugely important since it works for the full IMG, but knowing what makes the difference might be useful sometime. The only differences between the script I used to successfully make a -full system and this  are -
a) the --extend --xmb 2048; I didn't use that 
b) the IMG filename (duh)

tim
--
tim Rowledge; ***@***.***; http://www.rowledge.org/tim
You never finish a program, you just stop working on it.

gitbls, 2024-01-17:
Thx. I'm gonna need a bit more informaiton...

What signs of it booting would there be? How did you determine that it didn't boot?

timrowledge, 2024-01-17:
> On 2024-01-17, at 2:52 PM, Benn ***@***.***> wrote:
> 
> 
> Thx. I'm gonna need a bit more informaiton...
> 
> What signs of it booting would there be? How did you determine that it didn't boot?

Initial green light, a flash or two, then nothing. Left it for, oh five minutes? Couldn't connect via ssh or vnc, apparently hadn't connected to the router

tim
--
tim Rowledge; ***@***.***; http://www.rowledge.org/tim
If it was easy, the hardware people would take care of it.

gitbls, 2024-01-17:
OK. Can you put the disk on a system that has sdm installed? Then do
```
bash$ sudo sdm --explore /dev/sdX
sdm# cd /var/log
sdm# journalctl -D /var/log/journal --list-boots
sdm# journalctl -D /var/log/journal -b
```

If the `--list-boots` shows only one boot, then the 2nd journalctl cmd should be as above.

If there is more than one boot, then pls show me the output so I can direct you to further detective-ing.

gitbls, 2024-01-17:
Can you connect a monitor? 

You said this exact script works with the "-full" IMG, and the resulting IMG boots? There's literally no difference between them WRT anything sdm is doing. 

Do you want to pursue this to completion? If so,I will need to see the sdm logs (/etc/sdm/history) for both the one that works and the one that doesn't work. You can email them to me if you'd prefer to not post them here (a totally rational preference, BTW)

timrowledge, 2024-01-18:
This may be a dumb question (so I should probably post it on Quora :-) but how do I copy those files?

I just did a completely fresh build and burn and boot; same problem (so at least it's consistent)  with no boot related led flashing after an initial flicker, no network connection. When mounted back on the SDM host machine the journalctl output is the same lack of files error

> Can you connect a monitor?
>
> You said this exact script works with the "-full" IMG, and the resulting IMG boots? There's literally no difference between them WRT anything sdm is doing.
>
> Do you want [...]

tim
--
tim Rowledge; ***@***.***; http://www.rowledge.org/tim
atheism: not a religion, more a personal relationship with reality.

gitbls, 2024-01-18:
You can copy the /etc/sdm/history file from an SD card this way:
* `sdm --customize /dev/sdX`
* `cd /etc/sdm`
* `cp history user@host:` for some `user` on Linux system `host`

timrowledge, 2024-01-18:
> On 2024-01-17, at 5:38 PM, Benn ***@***.***> wrote:
> 
> 
> You can copy the /etc/sdm/history file from an SD card this way:
> 
> 	• sdm --customize /dev/sdX
> 	• cd /etc/sdm
> 	• cp history ***@***.***: for some user on Linux system host

I'm not doing this right, evidently. I'm not conversant with container stuff at all.

Opening the sd card FS with sdm --customize works fine. 
Trying the copy above simply copies the file into ***@***.***:' which doesn't seem what we want.
"cp history ***@***.***://home/tim"
returns error 
"cp: cannot stat ***@***.***:/home/tim: Not a directory"
I've tried assorted combinations based on wild guesses and faint recollections of some faffing with docker years ago. I don't even know good tokens to feed google to find answers for this bit...

Obviously *something* must work since you have to be using it to make sdm work! Isn't this all fun!

tim
--
tim Rowledge; ***@***.***; http://www.rowledge.org/tim
Useful Latin Phrases:- Nullo metro compositum est = It doesn't rhyme.

gitbls, 2024-01-18:
Whoops! In my guide I dropped an "s". it should be `scp` not `cp`:
```
scp history user@host:   #for some user on Linux system host
```

timrowledge, 2024-01-18:
> On 2024-01-17, at 8:47 PM, Benn ***@***.***> wrote:
> 
> 
> Whoops! In my guide I dropped an "s". it should be scp not cp:
> 
> scp history ***@***.***:   #for some user on Linux system host

OK, that seems to do more; what it told me is that sometihng is broken with resolving hostnames. I had to use the actual IP to get the file copied, but that did actually then copy the file.

What looks weird in the file is the dates of things being processed; I did a full customise build yesterday but the dates are from five days ago. That makes me wonder if the sd  card has locked itself and I'm not getting my new data on it. I'll try a different card.

tim
--
tim Rowledge; ***@***.***; http://www.rowledge.org/tim
Useful Latin Phrases:- Vidistine nuper imagines moventes bonas? = Seen any good movies lately?

timrowledge, 2024-01-18:
> On 2024-01-18, at 11:05 AM, Tim Rowledge ***@***.***> wrote:
> 
> I'll try a different card.

Grrr. {insert many eloquent swearwords here}

Yes, a dead card. With  A N Other card it worked perfectly. Grrr. Damn things should melt when they die, or sometihng actually indicative of the problem.

tim
--
tim Rowledge; ***@***.***; http://www.rowledge.org/tim
Strange OpCodes: FSE: Fake Serious Error

gitbls, 2024-01-19:
There are a lot of failure mode weaknesses in this space, although it's a lot better than it used to be.

gitbls, 2024-01-19:
PS Thanks for the kind words about sdm in the Pi forums. The more people that use it, the better, as far as I'm concerned.

And, I certainly don't consider you to be "pestering" me. I hope you don't think that either!

gitbls, 2024-01-23:
Not like the Monty Python parrot, and only "mostly closeable"?

closed 2024-01-23 by gitbls

timrowledge, 2024-01-23:
> On 2024-01-23, at 6:28 AM, Benn ***@***.***> wrote:
> 
> 
> Not like the Monty Python parrot, and only "mostly closeable"?

It's just a Flash wound!

tim
--
tim Rowledge; ***@***.***; http://www.rowledge.org/tim
<-------- The information went data way -------->