gitbls/sdm #163: Extend 'apps' plugin to allow removal of packages

timrowledge, 2024-01-16:
For example, removing the wolfram-engine seems to save about 4Gb of space on my Pi4/SSD system. Maybe allow *!packagename* as a way to specify?

timrowledge, 2024-01-16:
Double checked that ridiculous sounding number on a freshly built IMG and it's actually true. Went from 1.7Gb free to 5.5Gb on a 16Gb SD card.

gitbls, 2024-01-16:
Interesting. Nobody's ever asked for that before. If you're going to remove wolfram engine, why don't you just use the desktop version that doesn't contain it?

timrowledge, 2024-01-16:
> On 2024-01-15, at 7:47 PM, Benn ***@***.***> wrote:
> 
> 
> Interesting. Nobody's ever asked for that before. If you're going to remove wolfram engine, why don't you just use the desktop version that doesn't contain it?

I think the 'next one down' has quite a few other things - some of which are useful to me - missing. It's always an interesting question whether it is better to start full and remove, or start hardly there and add.

But generically I'd say that having the capability is more useful than not having it.

tim
--
tim Rowledge; ***@***.***; http://www.rowledge.org/tim
boutique - a startling hardwood

gitbls, 2024-01-16:
Currently the apps plugin does NO parsing of the apps list, and this doesn't strike me as a feature worthy of churning up the app for.

That said, it could be added with the argument `remove=list,of,packages` to the apps plugin.

gitbls, 2024-01-17:
It will be in the next release:
```
> Run Plugins Phase '1'
> Run Plugin 'apps' (/usr/local/sdm/plugins/apps) Phase 1 with arguments: 
  'remove=wolfram-engine|apps=emacs'
* Plugin apps: Start Phase 1
> IMG '2023-12-05-raspios-bookworm-arm64-full.img' has 1134868 1K-blocks (1.2GB, 1.1GiB) free at start of [default] Application Removes
> Plugin apps: Remove application(s) 'wolfram-engine'
> IMG '2023-12-05-raspios-bookworm-arm64-full.img' has 5130596 1K-blocks (5.3GB, 4.9GiB) free at start of [default] Application Installs
* Start [default] Application Installs
> [default] Application Installs: emacs
* [default] Application Installs Completed
```

timrowledge, 2024-01-17:
> On 2024-01-17, at 3:23 PM, Benn ***@***.***> wrote:
> 
> 
> It will be in the next release:
> 
> > Run Plugins Phase '1'
> > Run Plugin 'apps' (/usr/local/sdm/plugins/apps) Phase 1 with arguments: 
>   'remove=wolfram-engine|apps=emacs'

Excellent!

tim
--
tim Rowledge; ***@***.***; http://www.rowledge.org/tim
Useful Latin Phrases:-
Catapultam habeo. Nisi pecuniam omnem mihi dabis, ad caput tuum saxum immane mittam 
= I have a catapult. Give me all the money, or I will fling an enormous rock at your head.

gitbls, 2024-01-18:
sdm V11.2 released with this capability.

gitbls, 2024-02-03:
Closing due to lack of activity. OP, please re-open if needed. Thx!

closed 2024-02-03 by gitbls