gitbls/sdm #165: satrim trim-enable: Cannot find /sys/block/nvme0n1

dantchapman, 2024-01-28:
Thank you for providing the satrim script - I have used in in my pi4 machines.

I was trying to repeat the same on my pi5 with an nvme disk attached. The following error resulted:

There is an SSD attached to USB (sda) and also an nvme attached via pci. I'm in the process of migrating from the external ssd to the 'internal'  nvme.

* Plugin trim-enable: found disk /dev/sda
* Plugin trim-enable: Device /dev/sda provisioning mode already set to unmap
* Plugin trim-enable: Renaming existing udev rule /etc/udev/rules.d/10-sda-trim.rules to /etc/udev/rules.d/.10-sda-trim-pre-2024-01-28-17-59-54.rules
* Plugin trim-enable: Write udev rule /etc/udev/rules.d/10-sda-trim.rules
* Plugin trim-enable:  with contents: ACTION=="add|change", ATTRS{idVendor}=="04e8", ATTRS{idProduct}=="61f5", SUBSYSTEM=="scsi_disk", ATTR{provisioning_mode}="unmap"
ACTION=="add|change", KERNEL=="sda", SUBSYSTEM=="block", RUN+="/bin/sh -c 'echo 2147450880 > /sys/block/sda/queue/discard_max_bytes'"
* Plugin trim-enable: found disk /dev/nvme0n1
? Plugin trim-enable: Cannot find /sys/block/nvme0n1/device/scsi_disk/N:0:4:1/provisioning_mode

Is this because the nvme drive is being misinterpreted in some way? 

note that fstrim -v reports no errors when trimming a mounted directory.

gitbls, 2024-01-31:
OK, I have my nvme up and running. Indeed, it doesn't look like a scsi disk, so the method that trim-enable implements won't work for nvme drives, at least for those connected using the geekworm x1001 that I'm using.

This is going to take some research. 😢

gitbls, 2024-01-31:
Hold the phone. Or whatever. I just did `sudo fstrim -v /` and it trimmed the disk:
```
pt$ sudo fstrim -v /
/: 453.7 GiB (487199145984 bytes) trimmed
```
Not sure what this means just yet.

gitbls, 2024-02-01:
As far as I can tell, trim is enabled on NVME disks by default (at least on my Kingston drive). I suppose there may be NVME drives that don't support trim, but there's no way using the technique implemented by sdm to enable them, either.

I've updated the trim-enable plugin (and satrim) to "do the right thing" by checking whether trim is enabled via `lsblk --discard`. If the drive doesn't have that `provisioning_mode` directory and trim is enabled already, it will report that. If trim isn't enabled, it will report that it can't be enabled.

This will be in sdm V11.3, coming soon (I hope 🤞)

gitbls, 2024-02-03:
satrim in V11.3 should correct this. Please test. Thx!

dantchapman, 2024-02-04:
Tested and no more errors. Thanks again for all your efforts!

closed 2024-02-04 by dantchapman