gitbls/sdm #168: Error mounting after successful burn rpi5 NVMe

paulromeijn, 2024-01-30:
Disclaimer: I am not an expert at linux, but I think this might be interesting to let you know.
I am using an RPi 5 with a Pimoroni NVMe bottom HAT and a Patriot P310 240Gb NVMe SSD. The burn seems to be almost successful. After burning there is an error mounting the image, if I understand correctly.  The (headless) pi CAN now boot from the NVMe drive and the requested user is created with the set password and I can login via the enabled SSH. However, the hostname is not set to what I specified and still at raspberrypi. 

Here's what I did:

I customised an image with some basic settings:
` sudo sdm --customize --plugin user:"adduser=me|prompt" --plugin L10n:host --plugin disables:piwiz --regen-ssh-host-keys --restart 2023-12-11-bookworm-arm64-lite-my-custom-image.img`

Then burned the image: 
`sudo sdm --burn /dev/nvme0n1 --hostname pi --expand-root --regen-ssh-host-keys 2023-12-11-bookworm-arm64-lite-my-custom-image.img`

The following output showed: 
``` 
* Burn '2023-12-11-bookworm-arm64-lite-my-custom-image.img' (2.7GB, 2.6GiB) to '/dev/nvme0n1'...
dd if=2023-12-11-bookworm-arm64-lite-my-custom-image.img of=/dev/nvme0n1 status=progress bs=16M iflag=direct
2734686208 bytes (2.7 GB, 2.5 GiB) copied, 26 s, 105 MB/s
163+1 records in
163+1 records out
2738880512 bytes (2.7 GB, 2.6 GiB) copied, 26.817 s, 102 MB/s
> Expand Root: Expand partition 'nvme0n1p2' on device '/dev/nvme0n1' from (2.2GB, 2.0GiB) to (239.5GB, 223.1GiB)
* Mount /dev/nvme0n1 to resize the root file system
* Mount device '/dev/nvme0n1'
mount: /mnt/sdm: special device /dev/nvme0n12 does not exist.
       dmesg(1) may have more information after failed mount system call.
? Error mounting IMG '/dev/nvme0n1'
```

I check the devices: 
``` 
$ lsblk
NAME        MAJ:MIN RM   SIZE RO TYPE MOUNTPOINTS
mmcblk0     179:0    0  58.2G  0 disk
├─mmcblk0p1 179:1    0   512M  0 part /boot/firmware
└─mmcblk0p2 179:2    0  57.7G  0 part /
nvme0n1     259:0    0 223.6G  0 disk
├─nvme0n1p1 259:3    0   512M  0 part
└─nvme0n1p2 259:4    0 223.1G  0 part
```

dmesg output shown below due to length.

Is the device name maybe incorrect somehow? `device /dev/nvme0n12`

dmesg output (last parts only): 
```
[12522.986819] loop0: detected capacity change from 0 to 4292608
[12522.990990] EXT4-fs (loop0): mounted filesystem with ordered data mode. Quota mode: none.
[12522.994237] loop1: detected capacity change from 0 to 1048576
[12649.728228] EXT4-fs (loop0): unmounting filesystem.
[12745.919759] pcieport 0000:00:00.0: AER: Corrected error received: 0000:00:00.0
[12745.919805] pcieport 0000:00:00.0: PCIe Bus Error: severity=Corrected, type=Data Link Layer, (Transmitter ID)
[12745.919810] pcieport 0000:00:00.0:   device [14e4:2712] error status/mask=00001000/00002000
[12745.919817] pcieport 0000:00:00.0:    [12] Timeout
[12760.016392] pcieport 0000:00:00.0: AER: Corrected error received: 0000:00:00.0
[12760.016431] pcieport 0000:00:00.0: PCIe Bus Error: severity=Corrected, type=Data Link Layer, (Transmitter ID)
[12760.016435] pcieport 0000:00:00.0:   device [14e4:2712] error status/mask=00001000/00002000
[12760.016439] pcieport 0000:00:00.0:    [12] Timeout
[12760.018105] pcieport 0000:00:00.0: AER: Corrected error received: 0000:00:00.0
[12760.018112] pcieport 0000:00:00.0: PCIe Bus Error: severity=Corrected, type=Data Link Layer, (Transmitter ID)
[12760.018113] pcieport 0000:00:00.0:   device [14e4:2712] error status/mask=00001000/00002000
[12760.018115] pcieport 0000:00:00.0:    [12] Timeout
[12768.270624] pcieport 0000:00:00.0: AER: Corrected error received: 0000:00:00.0
[12768.270636] pcieport 0000:00:00.0: PCIe Bus Error: severity=Corrected, type=Data Link Layer, (Transmitter ID)
[12768.270639] pcieport 0000:00:00.0:   device [14e4:2712] error status/mask=00001000/00002000
[12768.270642] pcieport 0000:00:00.0:    [12] Timeout
[12768.274330] pcieport 0000:00:00.0: AER: Corrected error received: 0000:00:00.0
[12768.274342] pcieport 0000:00:00.0: PCIe Bus Error: severity=Corrected, type=Data Link Layer, (Transmitter ID)
[12768.274344] pcieport 0000:00:00.0:   device [14e4:2712] error status/mask=00001000/00002000
[12768.274347] pcieport 0000:00:00.0:    [12] Timeout
[12768.276632] pcieport 0000:00:00.0: AER: Corrected error received: 0000:00:00.0
[12768.276636] pcieport 0000:00:00.0: PCIe Bus Error: severity=Corrected, type=Data Link Layer, (Transmitter ID)
[12768.276638] pcieport 0000:00:00.0:   device [14e4:2712] error status/mask=00001000/00002000
[12768.276641] pcieport 0000:00:00.0:    [12] Timeout
[12768.277541] pcieport 0000:00:00.0: AER: Corrected error received: 0000:00:00.0
[12768.277547] pcieport 0000:00:00.0: PCIe Bus Error: severity=Corrected, type=Data Link Layer, (Transmitter ID)
[12768.277549] pcieport 0000:00:00.0:   device [14e4:2712] error status/mask=00001000/00002000
[12768.277552] pcieport 0000:00:00.0:    [12] Timeout
[12768.929238]  nvme0n1: p1 p2
[12770.080733]  nvme0n1: p1 p2
[12770.099507] /dev/nvme0n12: Can't open blockdev
```

paulromeijn, 2024-01-30:
I will check the version asap, but I installed it yesterday following the quick start `curl -L https://raw.githubusercontent.com/gitbls/sdm/master/EZsdmInstaller | bash ` command from the pi. So it should be a very recent version. Will update when I check it.

gitbls, 2024-01-30:
If you installed yesterday you've got the latest. Hopefully my nvme adapter and disk arrive...and work...so I can nail whatever this is. Will update this later today.

paulromeijn, 2024-01-31:
Great, that did the trick! This now seems. I do have version 11.2, btw, now confirmed.
```
> Set hostname 'myhostname'
> First System Boot Custom Boot Scripts disabled
> First System Boot automatic restart enabled
* Burn Completed
umount: /mnt/sdm/boot/firmware unmounted
umount: /mnt/sdm unmounted
* Storage '/dev/nvme0n1' is ready

$ lsblk
NAME        MAJ:MIN RM   SIZE RO TYPE MOUNTPOINTS
mmcblk0     179:0    0  58.2G  0 disk
├─mmcblk0p1 179:1    0   512M  0 part /boot/firmware
└─mmcblk0p2 179:2    0  57.7G  0 part /
nvme0n1     259:0    0 223.6G  0 disk
├─nvme0n1p1 259:5    0   512M  0 part
└─nvme0n1p2 259:6    0 223.1G  0 part
```

One other thing I tried did not go as expected, but that might be me using it wrong. I tried again using parted as a burn-plugin to add a partition and resizing the root and the added one: 
`sudo sdm --burn /dev/nvme0n1 --hostname myhostname--burn-plugin parted:"rootexpand=122880|addpartition=102400,ext4" --regen-ssh-host-keys 2023-12-11-bookworm-arm64-lite-my-custom-image.img` 

But this ends in:
```
* Run Post-Burn plugins
> Run Plugins Phase 'burn-complete'
> Run Plugin 'parted' (/usr/local/sdm/plugins/parted) Phase burn-complete with arguments: 'rootexpand=122880|addpartition=102400,ext4|burndev=/dev/nvme0n1|burnfilefile=|imgtype=Device'
/usr/local/sdm/plugins/parted: line 19: /etc/sdm/sdm-readparams: No such file or directory
/usr/local/sdm/plugins/parted: line 221: plugin_getargs: command not found
? Plugin 'parted' exited with failure

$ lsblk
NAME        MAJ:MIN RM   SIZE RO TYPE MOUNTPOINTS
mmcblk0     179:0    0  58.2G  0 disk
├─mmcblk0p1 179:1    0   512M  0 part /boot/firmware
└─mmcblk0p2 179:2    0  57.7G  0 part /
nvme0n1     259:0    0 223.6G  0 disk
├─nvme0n1p1 259:3    0   512M  0 part
└─nvme0n1p2 259:4    0     2G  0 part
```

Root is not expanded because parted is called and new partition not created, as expected from the specification. But am I using it wrong or could this be related?

gitbls, 2024-01-31:
I just found and fixed a few bugs in the emerging 11.3 version, but I've never seen these errors:
```
/usr/local/sdm/plugins/parted: line 19: /etc/sdm/sdm-readparams: No such file or directory
/usr/local/sdm/plugins/parted: line 221: plugin_getargs: command not found
```
Only the first error is interesting, since the first error results in the 2nd error.

A couple of questions so I can repro it here:
* What is the host hardware (where sdm is running)?
* What is the host OS?

And a note for the future: The first several lines of /etc/sdm/history in the customized IMG or burned SSD/SD Card are super-helpful, especially the first lines to "Plugins selected:" for these types of issues. For plugin problems, helpful to include up to the first report of free disk space.

paulromeijn, 2024-02-01:
A bit of delay because of our timezones. 

The host hardware is a Raspberry Pi 5 Model B Rev 1.0 (revision d04170) 8Gb with a pimoroni bottom nvme hat. The Nvme drive is a Patriot P310 240gb. It's running `Linux version 6.1.0-rpi8-rpi-2712 (debian-kernel@lists.debian.org) (gcc-12 (Debian 12.2.0-14) 12.2.0, GNU ld (GNU Binutils for Debian) 2.40) #1 SMP PREEMPT Debian 1:6.1.73-1+rpt1 (2024-01-25)`. 

I am redoing a test from scratch: a customized latest version of rapberry os lite. Two tests: 1) set burn plugin at burn command and 2) set it at customization command:

**_Test 1_**
Customization without parted plugin parameters, I will set that at burn time:
`$ sudo sdm --customize --plugin user:"adduser=testuser|prompt" --plugin L10n:host --plugin disables:piwiz --regen-ssh-host-keys --restart 2023-12-11-bookworm-arm64-lite-test-custom.img`

And then burn with parted parameters:
`$ sudo sdm --burn /dev/nvme0n1 --hostname testpi --burn-plugin parted:"rootexpand=122880|addpartition=102400,ext4" --regen-ssh-host-keys 2023-12-11-bookworm-arm64-lite-test-custom.img`

Which does not set partitions as I expect and the following output:
```
* Burn '2023-12-11-bookworm-arm64-lite-test-custom.img' (2.7GB, 2.6GiB) to '/dev/nvme0n1'...
dd if=2023-12-11-bookworm-arm64-lite-test-custom.img of=/dev/nvme0n1 status=progress bs=16M iflag=direct
2634022912 bytes (2.6 GB, 2.5 GiB) copied, 25 s, 105 MB/s
163+1 records in
163+1 records out
2738880512 bytes (2.7 GB, 2.6 GiB) copied, 26.6617 s, 103 MB/s
> Set new disk ID '8ec64463' on '/dev/nvme0n1'
Disk identifier changed from 0x4e639091 to 0x8ec64463.

The partition table has been altered.
Calling ioctl() to re-read partition table.
Syncing disks.
> Disk ID 8ec64463 set successfully with sfdisk
* Mount device '/dev/nvme0n1'
mount: /dev/nvme0n1p2 mounted on /mnt/sdm.
mount: /dev/nvme0n1p1 mounted on /mnt/sdm/boot/firmware.
> Set new Disk ID '8ec64463' in /etc/fstab and /boot/firmware/cmdline.txt
> Flush accumulated burn log messages
> Force --no-expand-root for parted burn-plugin
> Force --regen-ssh-host-keys for --no-expand-root
> Disable unneeded RasPiOS firstboot service and /etc/init.d/resize2fs_once
> Disable regenerate_ssh_host_keys service; First Boot will run it instead
> Set hostname 'testpi'
> First System Boot Custom Boot Scripts disabled
> First System Boot automatic restart enabled
* Burn Completed
umount: /mnt/sdm/boot/firmware unmounted
umount: /mnt/sdm unmounted
* Run Post-Burn plugins
> Run Plugins Phase 'burn-complete'
> Run Plugin 'parted' (/usr/local/sdm/plugins/parted) Phase burn-complete with arguments: 'rootexpand=122880|addpartition=102400,ext4|burndev=/dev/nvme0n1|burnfilefile=|imgtype=Device'
/usr/local/sdm/plugins/parted: line 19: /etc/sdm/sdm-readparams: No such file or directory
/usr/local/sdm/plugins/parted: line 221: plugin_getargs: command not found
? Plugin 'parted' exited with failure

$  lsblk
NAME        MAJ:MIN RM   SIZE RO TYPE MOUNTPOINTS
mmcblk0     179:0    0  58.2G  0 disk
├─mmcblk0p1 179:1    0   512M  0 part /boot/firmware
└─mmcblk0p2 179:2    0  57.7G  0 part /
nvme0n1     259:0    0 223.6G  0 disk
├─nvme0n1p1 259:1    0   512M  0 part
└─nvme0n1p2 259:2    0     2G  0 part
```

**_Test 2: set burn plugin at customization:_** 
Customize image and set burn plugin parted:
`$ sudo sdm --customize --plugin user:"adduser=testuser|prompt" --plugin L10n:host --plugin disables:piwiz --burn-plugin parted:"rootexpand=122880|addpartition=102400,ext4" --regen-ssh-host-keys --restart 2023-12-11-bookworm-arm64-lite-test-custom.img`

Gives output:
```
** IMG '2023-12-11-bookworm-arm64-lite-test-custom.img' has already been customized. Redo? [y/n]? y
* Start Configuration
> Command Line: /usr/local/bin/sdm --customize --plugin user:adduser=testuser|prompt --plugin
  L10n:host --plugin disables:piwiz --burn-plugin
[... 131 lines cut ...]
> Force --regen-ssh-host-keys for --no-expand-root
> Disable unneeded RasPiOS firstboot service and /etc/init.d/resize2fs_once
> Disable regenerate_ssh_host_keys service; First Boot will run it instead
> Set hostname 'testpi'
> First System Boot Custom Boot Scripts disabled
> First System Boot automatic restart enabled
* Burn Completed
umount: /mnt/sdm/boot/firmware unmounted
umount: /mnt/sdm unmounted
* Run Post-Burn plugins
> Run Plugins Phase 'burn-complete'
> Run Plugin 'parted' (/usr/local/sdm/plugins/parted) Phase burn-complete with arguments: 'rootexpand=122880|addpartition=102400,ext4|burndev=/dev/nvme0n1|burnfilefile=|imgtype=Device'
/usr/local/sdm/plugins/parted: line 19: /etc/sdm/sdm-readparams: No such file or directory
/usr/local/sdm/plugins/parted: line 221: plugin_getargs: command not found
? Plugin 'parted' exited with failure

$ lsblk
NAME        MAJ:MIN RM   SIZE RO TYPE MOUNTPOINTS
mmcblk0     179:0    0  58.2G  0 disk
├─mmcblk0p1 179:1    0   512M  0 part /boot/firmware
└─mmcblk0p2 179:2    0  57.7G  0 part /
nvme0n1     259:0    0 223.6G  0 disk
├─nvme0n1p1 259:5    0   512M  0 part
└─nvme0n1p2 259:6    0     2G  0 part
```

paulromeijn, 2024-02-01:
I have also rerun the test to set burnplugin at customization with a fresh copy of clean raspberry os image (uncustomized, just in case). I set the burnplugin `sudo sdm --customize --plugin user:"adduser=testuser|prompt" --plugin L10n:host --plugin disables:piwiz --burn-plugin parted:"rootexpand=122880|addpartition=102400,ext4" --regen-ssh-host-keys --restart 2023-12-11-bookworm-arm64-lite-test.img` and then I check the contents of the files in the errors that appeared in previous output.
Output:
```
* Start Configuration
> Command Line: /usr/local/bin/sdm --customize --plugin user:adduser=testuser|prompt --plugin
  L10n:host --plugin disables:piwiz --burn-plugin
  parted:rootexpand=122880|addpartition=102400,ext4 --regen-ssh-host-keys --restart
  2023-12-11-bookworm-arm64-lite-test.img
* Host Information
   Hostname: unipi-sd
   uname: Linux unipi-sd 6.1.0-rpi8-rpi-2712 #1 SMP PREEMPT Debian 1:6.1.73-1+rpt1 (2024-01-25)
  aarch64 GNU/Linux
   os-release Name: Debian GNU/Linux 12 (bookworm)
   Version: 12 (bookworm)
   sdm Version: V11.2
* IMG Information
   Name: 2023-12-11-bookworm-arm64-lite-test.img
   Date: 2023-12-11
   RasPiOS Version: 12
   RasPiOS Architecture: 64-bit aarch64
   os-release Version: 12 (bookworm)
* Plugins selected:
   * user
       Args: adduser=testuser|prompt
   * L10n
       Args: host
   * disables
       Args: piwiz
> IMG '2023-12-11-bookworm-arm64-lite-test.img' has 337076 1K-blocks (0.3GB, 0.3GiB) free
> Copy sdm to /usr/local/sdm in the IMG
* Start Phase 0 image customization
> Run Plugins Phase '0'
> Run Plugin 'user' (/mnt/sdm/usr/local/sdm/plugins/user) Phase 0 with arguments:
  'adduser=testuser|prompt'
* Plugin user: Start Phase 0
> Plugin user: Keys/values found:
   adduser: testuser
   prompt:
> Plugin user: Create user 'testuser' home directory '/home/testuser' so available for other
  plugins
* Plugin user: Complete Phase 0
> Run Plugin 'L10n' (/mnt/sdm/usr/local/sdm/plugins/L10n) Phase 0 with arguments: 'host'
* Plugin L10n: Start Phase 0
> Plugin L10n: Keys/values found:
   host:
> Plugin L10n: Read L10n configuration from host system
> Plugin L10n: Load Localization (L10N) settings from running system
> Plugin L10n:   Keymap:       us
> Plugin L10n:   Locale:       en_GB.UTF-8
> Plugin L10n:   Timezone:     Europe/Madrid
> Plugin L10n:   WiFi Country:
> Plugin L10n: Save L10n configuration
* Plugin L10n: Complete Phase 0
> Run Plugin 'disables' (/mnt/sdm/usr/local/sdm/plugins/disables) Phase 0 with arguments:
  'piwiz'
* Plugin disables: Start Phase 0
> Plugin disables: Keys/values found:
   piwiz:
* Plugin disables: Complete Phase 0
* Phase 0 Completed
* Enter image '2023-12-11-bookworm-arm64-lite-test.img' for Phase 1
* Start Phase 1 image customization
> IMG '2023-12-11-bookworm-arm64-lite-test.img' has 337032 1K-blocks (0.3GB, 0.3GiB) free at start of Phase 1 image customization
> Configure and enable sdm FirstBoot service (sdm-firstboot)
> Start 'apt update'
> IMG '2023-12-11-bookworm-arm64-lite-test.img' has 337028 1K-blocks (0.3GB, 0.3GiB) free at start of 'apt update'
> IMG '2023-12-11-bookworm-arm64-lite-test.img' has 221928 1K-blocks (0.2GB, 0.2GiB) free at end of 'apt update'
> Run Plugins Phase '1'
> Run Plugin 'user' (/usr/local/sdm/plugins/user) Phase 1 with arguments:
  'adduser=testuser|prompt'
[... 81 lines cut ...]
                snolq="${value#\"}"        # Get string without open quote
                if [ "$snolq" != "$value" ] # if lq was there, then update string and also del close quote
                then
                    value=$snolq
                    value="${value%\"}"
                fi
                snolq="${value#\'}"        # Get string without open quote
                if [ "$snolq" != "$value" ] # Ditto double quote comment
                then
                    value=$snolq
                    value="${value%\'}"
                fi
                printf -v "$rpifun" "%s" "$value"     #eval "${rpifun}=\"$value\""

            fi
        done < $cf
    fi
    [ -f $SDMPT/$sdmdir/sdm-cparse ] && source $SDMPT/$sdmdir/sdm-cparse || echo "? Internal error: File '$SDMPT/$sdmdir/sd>    return
}
# Call the function after we've been sourced
readparams
```
So they are present in the image. But somehow code doesn't find it? I don't know how to debug this so not sure how for example `$SDMPT/etc/sdm/sdm-readparams` will evaluate.

Hope this and previous helps?

1stcall, 2024-02-01:
@gitbls In line 41 of the parted plugin, you are exporting SDMPT (declare -x) within a function whilst SDMPT is already exported within the main sdm script.  This does not work as you expect.  I have not tested this, but if you replace this line with SDMPT=$(makemtpt), I thik it will work.

paulromeijn, 2024-02-01:
Haha yeah I figured you know better than me what to filter out

gitbls, 2024-02-01:
Just did a test with 11.2 (which you're using). Wasn't able to replicate the problem with "/etc/sdm/sdm-readparams: No such file or directory", but did run into another issue I fixed yesterday. Reran the same test with 11.3 and all good. 

Now I just need to finish up 11.3 for you 🥴Still chasing a couple of other features/bugs, but should have it done in the next day or two, worst case.

paulromeijn, 2024-02-02:
Too bad that it doesn't reproduce, but fortunately I can work around it. I burned the image without the `--expand-root` flag and manually expanded the primary partition+fs and created the extra partition+fs. Seems to work.

Could it be due to us using different brand NVMe hats? Also tried setting/unsetting `dtparam=pciex1_gen=3` in boot config.txt just to rule it out, but no difference.

Nonetheless, I am grateful for such a great tool being here in the first place.

gitbls, 2024-02-03:
I believe these are all fixed in 11.3, per testing on my NVME drive. Please test with your configuration. Thx!

paulromeijn, 2024-02-05:
I just tested on a new Rpi5 4Gb, now with a kingston NV2 1TB drive and another pimoroni base hat using SDM 11.3 and still have the issue: 
```
Run Plugin 'parted' (/usr/local/sdm/plugins/parted) Phase burn-complete with arguments: 'rootexpand=122880|addpartition=102400,ext4|burndev=/dev/nvme0n1|burnfilefile=|imgtype=Device'
/usr/local/sdm/plugins/parted: line 19: /etc/sdm/sdm-readparams: No such file or directory
/usr/local/sdm/plugins/parted: line 221: plugin_getargs: command not found
? Plugin 'parted' exited with failure```

But it is another issue than the one I started with, so I could close this one because now that does work. As I'm still with the pimoroni, maybe that has something to do with it?

gitbls, 2024-02-05:
I'm not worried about whether it's this issue or another issue. This one is open, so let's just continue here.

I need to see the /etc/sdm/history file from in the IMG that you're burning. You can either post it here or email it to me (email on https://github.com/gitbls/sdm/README.md). My suspicion is that there's something in the way your IMG has been customized, but in addition to the history file, please also shoot me the commands you're using to customize and burn. Please feel free to redact passwords as necessary, but leave everything else intact.

Thx!

gitbls, 2024-02-05:
Ah! Never mind...I was in my best thinking place (shower 😂) and realized exactly what the problem is.  🥴

gitbls, 2024-02-05:
On host systems running sdm where sdm was NOT used to burn the host system disk, there is an error in the installer. This will be corrected shortly in the default download. The issue is that the installer doesn't do these steps, which burn-plugins rely on. 

You can correct your system so that the parted plugin works with the following. It assumes that you put sdm in the default location (/usr/local/sdm). If you've changed that, modify these commands appropriately:
```
sudo mkdir -p /etc/sdm
sudo cp /usr/local/sdm/sdm-readparams /etc/sdm/readparams
sudo cat >| /etc/sdm/cparse <<EOF
sdmdir:/usr/local/sdm
src:/usr/local/sdm
plugindebug:0
EOF
```
Please LMK if this fixes it.

Apologies for the delay in nailing this  for you 😳

paulromeijn, 2024-02-07:
The first two lines ok. But then I'm unfamiliar with the syntax. I execute it line by line and there is an error `-bash: /etc/sdm/cparse: Permission denied` 
I do use `sudo cat ` + the rest

gitbls, 2024-02-07:
> The first two lines ok. But then I'm unfamiliar with the syntax. I execute it line by line and there is an error `-bash: /etc/sdm/cparse: Permission denied` I do use `sudo cat ` + the rest

Sorry, was apparently in too much of a hurry. Try this.

After you type the cat command, you should be prompted with `> `. Type each of the 3 lines in turn, then type EOF (on a line by itself), and the cat command will complete.
```
sudo rm -rf /etc/sdm
sudo mkdir -p /etc/sdm
sudo cp -a /usr/local/sdm/sdm-readparams /etc/sdm
sudo cat > /etc/sdm/cparams <<EOF
sdmdir:/usr/local/sdm
src:/usr/local/sdm
plugindebug:0
EOF
```

paulromeijn, 2024-02-08:
Still get permission denied:
```
~ $ sudo rm -rf /etc/sdm
~ $ sudo mkdir -p /etc/sdm
~ $ sudo cp -a /usr/local/sdm/sdm-readparams /etc/sdm
~ $ sudo cat > /etc/sdm/cparams <<EOF
> sdmdir:/usr/local/sdm
> src:/usr/local/sdm
> plugindebug:0
> EOF
-bash: /etc/sdm/cparams: Permission denied
```
So I outputted the file first to home dir and then copied it to `/etc/sdm` manually. Now the cparams is in that folder.
The image was customized with the burn parameter `--burn-plugin parted:"rootexpand=307200"` and then burned without --root-expand:
`$ sudo sdm --burn /dev/nvme0n1 --hostname testpi --regen-ssh-host-keys 2023-12-11-bookworm-arm64-lite-custom.img`

And success! 😎 Nice 
```
umount: /mnt/sdm/boot/firmware unmounted
umount: /mnt/sdm unmounted
* Storage '/dev/nvme0n1' is ready
~ $ lsblk
NAME        MAJ:MIN RM   SIZE RO TYPE MOUNTPOINTS
mmcblk0     179:0    0    58G  0 disk
├─mmcblk0p1 179:1    0   512M  0 part /boot/firmware
└─mmcblk0p2 179:2    0  57.4G  0 part /
nvme0n1     259:0    0 931.5G  0 disk
├─nvme0n1p1 259:3    0   512M  0 part
└─nvme0n1p2 259:4    0   302G  0 part
```
Not sure why cat has no permissions while using sudo?

But anyway now it is all working as expected. Any suggestion how I can set another partition on that disk using the burn plugin that expands to fill the whole free space? It's not fully clear from the docs to me, only how to expand root. I have about 630GB left that I want to configure to the last byte, but would like to be able to do it without calculating all clusters or guessing etc.

gitbls, 2024-02-08:
> But anyway now it is all working as expected. Any suggestion how I can set another partition on that disk using the burn plugin that expands to fill the whole free space? It's not fully clear from [...]

Glad that you got the cparams file created. Not sure what's going on there.

Re creating a "last partition" that fills the disk: There's no way to do that at the moment, but there should be. I've got the parted plugin open to add GPT disk support, so will look into adding this "makes perfect sense" feature.

gitbls, 2024-02-08:
Confirmed. Adding a last plugin with a size of 0 will fill the whole free space in the next release.

Of course, if there's another partition specified after a 0-length partition, it will fail miserably, since there will be no room for it.

UnconnectedBedna, 2024-02-09:
Hi, me again. xD

It was good I read this because I was thinking of forking and sending 1 or 2 requests.
There are things I could provide knowledge wise that could speed up the "burn" process by hundreds of % if you are interested?

I also saw you are doing some unnecessary operations that I could help out with if you'd like.

But you should def make the changes you have in mind first because I might need to move a few things around to do what I have in mind.

gitbls, 2024-02-09:
> Hi, me again. xD
> 
> It was good I read this because I was thinking of forking and sending 1 or 2 requests. There are things I could provide knowledge wise that could speed up the "burn" process by hundreds of % if you are interested? (rsync instead of dd, like I do in my script)
> 
> I also saw you are doing some unnecessary operations that I could help out with if you'd like. (partition resizing and stuff)
> 
> But you should def make the changes you have in mind first because I might need to move a few things around to do what I have in mind.

Please don't hijack other threads...please start a new Issue (or Discussion if you'd prefer) where we can chat about this.

I'm interested in your code to understand the code complexity, tradeoffs, possible issues, etc. as well as proven performance data.

Thx.

gitbls, 2024-02-11:
> But anyway now it is all working as expected. Any suggestion how I can set another partition on that disk using the burn plugin that expands to fill the whole free space? It's not fully clear from the docs to me, only how to expand root. I have about 630GB left that I want to configure to the last byte, but would like to be able to do it without calculating all clusters or guessing etc.
> 
> --edit: (re-reading the docs) Or would the following command do the above and expand the extra partition by setting size as 0? `--burn-plugin parted:"rootexpand=307200|addpartition=0,ext4"`
> 
This is fixed in the just-released V11.4. Please test. Thx!

paulromeijn, 2024-02-15:
Ok interesting, I tried installing latest version using

Resulted in this:
```
(--snip---)
Downloading 'plugins/wsdd' to '/usr/local/sdm/plugins/wsdd'
Downloading '1piboot/1piboot.conf' to '/usr/local/sdm/1piboot/1piboot.conf'
* Create link for sdm: /usr/local/bin/sdm
**bash: line 177: /etc/sdm/cparse: Permission denied**
* Ensure required packages are installed: systemd-container qemu-user-static binfmt-support parted
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
systemd-container is already the newest version (252.19-1~deb12u1).
qemu-user-static is already the newest version (1:7.2+dfsg-7+deb12u3).
binfmt-support is already the newest version (2.2.2-2).
parted is already the newest version (3.5-3).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.

* Invoke sdm with 'sudo sdm'

  Complete sdm documentation: https://github.com/gitbls/sdm/tree/master/Docs

$ sudo sdm --version
sdm V11.4
```
So here also no access to the cparams file. Not sure why though, my installation here is pretty clean and basic afaik.

gitbls, 2024-02-15:
> Ok interesting, I tried installing latest version using `curl -L https://raw.githubusercontent.com/gitbls/sdm/master/EZsdmInstaller | bash`
> 
> Resulted in this:
...
> Note the line `bash: line 177: /etc/sdm/cparse: Permission denied` So here also no access to the cparams file. Not sure why though, my installation here is pretty clean and basic afaik.
> 
> --edit: using `sudo curl ~~ ` to install same result. Not sure if sudo is necessary there though

Pretty sure I understand what's happening, but what is the host OS and version that you're installing on?
Thx

paulromeijn, 2024-02-15:
Hardware is Raspberry Pi 5 Model B Rev 1.0 (4GB)
```
$ cat /proc/version
Linux version 6.1.0-rpi8-rpi-2712 (debian-kernel@lists.debian.org) (gcc-12 (Debian 12.2.0-14) 12.2.0, GNU ld (GNU Binutils for Debian) 2.40) #1 SMP PREEMPT Debian 1:6.1.73-1+rpt1 (2024-01-25)
```

gitbls, 2024-02-15:
OK, thx. Please download the latest EZsdmInstaller and rerun it. It will properly create cparse, and I'm off to read about "here documents" and sudo 😕

paulromeijn, 2024-02-15:
ok that went well now, great!
```
(---)
Downloading '1piboot/1piboot.conf' to '/usr/local/sdm/1piboot/1piboot.conf'
* Create link for sdm: /usr/local/bin/sdm
* Ensure required packages are installed: systemd-container qemu-user-static binfmt-support parted
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
systemd-container is already the newest version (252.19-1~deb12u1).
qemu-user-static is already the newest version (1:7.2+dfsg-7+deb12u3).
binfmt-support is already the newest version (2.2.2-2).
parted is already the newest version (3.5-3).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.

* Invoke sdm with 'sudo sdm'

  Complete sdm documentation: https://github.com/gitbls/sdm/tree/master/Docs
```

Will now try the image burning again

paulromeijn, 2024-02-15:
ok that's looking good, great work! 
```
(----)
Allocating group tables: done
Writing inode tables: done
Creating journal (262144 blocks): done
Writing superblocks and filesystem accounting information: done

* Storage '/dev/nvme0n1' is ready

$ lsblk
NAME        MAJ:MIN RM   SIZE RO TYPE MOUNTPOINTS
mmcblk0     179:0    0    58G  0 disk
├─mmcblk0p1 179:1    0   512M  0 part /boot/firmware
└─mmcblk0p2 179:2    0  57.4G  0 part /
nvme0n1     259:0    0 931.5G  0 disk
├─nvme0n1p1 259:4    0   512M  0 part
├─nvme0n1p2 259:5    0   302G  0 part
└─nvme0n1p3 259:6    0   629G  0 part
```

closed 2024-02-25 by gitbls