gitbls/sdm #170: making installing apps faster

atifghaffar, 2024-02-08:
Hi, thanks for the fantastic software.

How can I make installing the apps faster?
I want to run my whole process multiple times and when installing software it takes a while.

Can I do some kind of proxying/caching?

thanks and kind regards

atifghaffar, 2024-02-08:
my build script is the following

#!/bin/bash

sdm --customize --extend --autologin \
--plugin copyfile:"from=./files/root/.ssh/id_rsa|to=/root/.ssh|chown=root.root|chmod=0600|mkdirif"  \
--plugin copyfile:"from=./files/root/.ssh/id_rsa.pub|to=/root/.ssh|chown=root.root|chmod=0644|mkdirif"  \
--plugin copyfile:"from=./files/root/.ssh/authorized_keys|to=/root/.ssh|chown=root.root|chmod=0644|mkdirif"  \
--plugin copyfile:"from=./files/root/.ssh/config|to=/root/.ssh|chown=root.root|chmod=0644|mkdirif"  \
--plugin copyfile:"from=./files/home/pi/.profile|to=/home/pi/|chown=pi.pi|chmod=0644|mkdirif"  \
--plugin copyfile:"from=./files/home/pi/.xinitrc|to=/home/pi/|chown=pi.pi|chmod=0644|mkdirif"  \
--plugin runatboot:"script=./files/firstrun.sh" \
--plugin user:"setpassword=pi|password=xxx"  \
--plugin disables:piwiz --plugin L10n:host \
--plugin copydir:"from=./projects/opt/|to=/opt/"  \
--plugin apps:"apps=@core-apps|name=coreapps"  \
--plugin apps:"apps=@x-apps|name=xapps"  \
--plugin system:"rclocal=/bin/bash /opt/scripts/startup.sh" \
--plugin quietness:"consoleblank=600|noquiet=keep|nosplash=keep|noplymouth" \
--regen-ssh-host-keys \
--restart build/2023-12-11-raspios-bookworm-arm64-lite.img

atifghaffar, 2024-02-08:
and here are the contents of core-apps and x-apps

cryptsetup 
cryptsetup-bin 
overlayroot
wireguard
git
tmux
php-cli
php-fpm
php-redis
php-mysql
php-yaml
nginx-full
imagemagick
dnsmasq
mesa-vdpau-drivers
vdpau-driver-all
xorg
x11-xserver-utils
xserver-xorg-video-fbdev
xserver-xorg-video-fbturbo
chromium-browser
pulseaudio
pulseaudio-module-bluetooth

atifghaffar, 2024-02-08:
thanks. I will look into it immediately

atifghaffar, 2024-02-08:
So the goal is to build the image faster on the build server. 

On the build server, I install apt-cacher-ng

Then I should use in in the build command to make the build faster?

So like this 

sdm --aptcache localhost  --customize --extend --autologin .... other options ... /path/of/the/base.distro.img

correct?

thanks

atifghaffar, 2024-02-08:
My workflow is the following.

1. Copy the base-image from dnld/bookworm.img  to build/bookworm.img 
2. Run the sdm command on build/bookworm.img 
3. check everything is chroot
4. burn the card 
5. test on the pi
6. redo if even a smallest of the change is needed

So I hope ( and it looks like )  the apt-cacher-ng will help speed up the step 2.

I will do a couple of runs and will report

atifghaffar, 2024-02-08:
PS: I am moving from pi-gen to sdm hoping to reduce the time needed.

gitbls, 2024-02-08:
That's pretty much how I do things. Lots of burns/reburns.

I do download and save an unmodified IMG and then just copy that locally. Save a few more seconds!..time is of the essence!

atifghaffar, 2024-02-08:
ok, so I will time the following commands ( removing all my copying etc ) 

sdm  --customize  --plugin apps:"apps=tmux"  build/2023-12-11-raspios-bookworm-arm64-lite.img

vs

sdm   --aptcache localhost   --customize  --plugin apps:"apps=tmux"  build/2023-12-11-raspios-bookworm-arm64-lite.img

and then take it from there. 

Just to see where my bottlenecks are ..

thanks

gitbls, 2024-02-08:
You should pick something bigger than tmux, you'll probably not even notice the difference unless your internet connection is
s-l-o-w.

Don't forget that the first time is throwaway, since that loads the cache. Subsequent runs will be faster.

atifghaffar, 2024-02-08:
ok thanks. then I will try to load chromium instead.

atifghaffar, 2024-02-08:
the apt upgrade is taking a lot of time too.
is there a way to avoid doing that?

atifghaffar, 2024-02-08:
* Phase 1 post-app installation/configuration
> Start 'apt upgrade'
> IMG 'build/2023-12-11-raspios-bookworm-arm64-lite.img' has 219500 1K-blocks (0.2GB, 0.2GiB) free at start of 'apt upgrade'

gitbls, 2024-02-08:
`--poptions noupgrade`

If that's all the free space you have, add `--extend --xmb N` to add more space. N=1024 or 2048 add 1 or 2GB. I make mine big and then figure out what the "correct" number should be.

It's worth doing the apt update (which you can also avoid with `--poptions`), but sometimes you get caught because newer versions of stuff you actually want to install aren't known until you do an update.

atifghaffar, 2024-02-08:
Yes, I prefer to not do the apt. upgrade as I may burn the same cards over time and they should be exactly the same.
I do use the --extend but for this test, i am ommitting it.

atifghaffar, 2024-02-08:
so this would be the baseline command to use. correct? ( I will replace tmux with chromium when doing the benchmark ) 
Any other optimzations?

sdm  --customize --poptions noupgrade --poptions noautoremove  --poptions noupdate  --batch   --plugin apps:"apps=tmux"   build/2023-12-11-raspios-bookworm-arm64-lite.img

thanks

gitbls, 2024-02-08:
Use --poptions noupdate,noupgrade,noautoremove

atifghaffar, 2024-02-08:
So I will run the following

time sdm  --extend  --customize --poptions noupdate,noupgrade,noautoremove  --batch   --plugin apps:"apps=chromium-browser"   build/2023-12-11-raspios-bookworm-arm64-lite.img

and 

time sdm --aptcache localhost --extend  --customize --poptions noupdate,noupgrade,noautoremove  --batch   --plugin apps:"apps=chromium-browser"   build/2023-12-11-raspios-bookworm-arm64-lite.img

to compare the results

atifghaffar, 2024-02-08:
time without the cache: real    3m8.553s

atifghaffar, 2024-02-08:
Will do two runs now with the cached

atifghaffar, 2024-02-08:
first run with cache: real    3m7.563s

atifghaffar, 2024-02-08:
second run with cache: real    3m6.467s

no big change

atifghaffar, 2024-02-08:
is there something not correct with my command?

gitbls, 2024-02-08:
1. Did you install apt-cacher-ng on the system that is to be the server? If so, how did you install it? NVM I see that you did
2. Is /etc/apt/apt.conf.d/02proxy set up correctly?
3. Did you `sudo systemctl start apt-cacher-ng`
4. What is the output from `netstat -ltpn | grep cache`

atifghaffar, 2024-02-08:
I installed on the host  with apt install apt-cache-ng

content of /etc/apt/apt.conf.d/02proxy
Acquire::http::proxy "http://:3142";

output of netstat -ltpn | grep cache
tcp        0      0 0.0.0.0:3142            0.0.0.0:*               LISTEN      1032725/apt-cacher-
tcp6       0      0 :::3142                 :::*                    LISTEN      1032725/apt-cacher-

I can see the chromium is in the cache

root@hp-music-box:~# du -sh /var/cache/apt-cacher-ng/archive.raspberrypi.com/debian/pool/main/c/chromium-browser/
212M	/var/cache/apt-cacher-ng/archive.raspberrypi.com/debian/pool/main/c/chromium-browser/

atifghaffar, 2024-02-08:
I did the same tests on my local lab

On my other build machine ( local ADSL network ) 
without  cache: 		real 6m20.656s
with cache: first-run: 	real	6m14.969s
with cache: second-run: real	5m22.214s

atifghaffar, 2024-02-08:
I will now do the most basics of the test without installing any packages to get the baseline.

gitbls, 2024-02-08:
Run a big test, installing all your stuff. You won't see as much of a difference with such a small operation. 

Also see the video that shows the improvement. There's a pointer to it on the README and the time to watch at is mentioned in the doc link I posted above, I believe.

atifghaffar, 2024-02-08:
time sdm  --customize --poptions noupdate,noupgrade,noautoremove  --batch --regen-ssh-host-keys --restart  --plugin user:"setpassword=pi|password=xxx"    build/2023-12-11-raspios-bookworm-arm64-lite.img

real	0m19.916s
user	0m3.052s
sys	0m0.726s

atifghaffar, 2024-02-08:
I will run the full build

atifghaffar, 2024-02-08:
and I will watch/listen to  the video in full.

gitbls, 2024-02-08:
The video is old and many things have changed since it was made but the apt-cacher-ng part is valid.

atifghaffar, 2024-02-08:
Thanks. Its very useful. 

In anycase, it take less time than pi-gen.

atifghaffar, 2024-02-08:
Well now I have these errors when installing the apps

W: Failed to fetch http://deb.debian.org/debian/dists/bookworm/InRelease  Could not resolve '127.0.0.0.1'

which makes sense as I am passing --aptcache 127.0.0.1

how to solve this?

Must I set the local ip of the host?

atifghaffar, 2024-02-08:
oops. i see the extra zero..

atifghaffar, 2024-02-08:
finaly, way too much time. 

I will offload the installation to the firstboot.

gitbls, 2024-02-08:
> finaly, way too much time.
> 
> I will offload the installation to the firstboot.

I have no clue what you're referring to, but from my perspective, I try to avoid doing anything that makes the first boot longer.

atifghaffar, 2024-02-08:
I meant, I could not figure out what was taking so much time, I will do some more tests now.
Yes I agree, pre-installing the packages is much better to avoid issues when the packages change.

atifghaffar, 2024-02-09:
Latest update. 

I installed ubuntu 22.x on my Mac M2 using multipass
installed sdm, apt-cacher, etc
ran the sdm build

It took only 1m and a few seconds
with apt-cache only 47 seconds. 
WOW.

How is that possible?

The virtual machine is running with a single processor and 1g of ram

while the build machine has 4 procs and 8g of ram.

Any idea of what is making it so fast on my mac which is using qemu as the backend for multipass?

atifghaffar, 2024-02-09:
I reinstalled my local build server but It seems like the disk is slow. 
The Mac M2 ssd speed is stupid fast or it is the way multipass mounts it.

gitbls, 2024-02-10:
If you want educated answers/guesses to your huge # of questions, you'll need to post each of them as a separate issue and provide sufficient detail for me to reasonably comment on them.

This thread has way too much going on, and there's insufficient detail on any of nearly all of the various questions you've raised, so I'm closing it. 

Happy to provide you with assistance, provided there are actual questions with details. 

Thx

closed 2024-02-10 by gitbls

atifghaffar, 2024-02-10:
ok. thanks. For me its working, my last comments were to only describe what I found out so to benefit perhaps other users.