gitbls/sdm #171: Changing the splash screen

atifghaffar, 2024-02-08:
hi, 

is it possible with sdm to change the splash screen from the rainbow image to a custom image?

Could not find the setting for that.

thanks.

gitbls, 2024-02-08:
ISTR that there's no way to change it. It's only on the screen for a bit, and to me, never seemed worth spending any time on it.

atifghaffar, 2024-02-08:
I will have a dozen screens running in a gym and they maybe rebooted ( power-cycled ) when stuck/mafunctioning or they will be shutdown in the night and started in the morning.

I wanted to make the startup screen as clean as possible, no startup message, no kernel messages, etc, just a screen with the logo while the system boots. 

Any ideas on to achieve that using sdm?

perhaps something with quietness plugin.

I have currently

quietness:"consoleblank=600|noquiet=keep|nosplash=keep|noplymouth"
but I guess I have to play with different variations of it, 

I am not worried about the first-boot screen, that is fine, it just the recurring boots should be as less verbose as possible.

thanks

gitbls, 2024-02-08:
Why are you using `noquiet`, or is that just for testing? By default quiet is set. When you use `noquiet` you get a much more chatty boot.

The other 2 make it quieter, which is what you want.

atifghaffar, 2024-02-09:
So, I wanted that on firstboot, there are kernel messages, so : noquiet
Then next boots quiet. 

So I guess it should remove the=keep from the noquiet
something like this

quietness:"consoleblank=600|noquiet|nosplash=keep|noplymouth"

rebuilding with this now

gitbls, 2024-02-10:
How's this for a deal?

If you can figure out how to reliably replace the rainbow image with a custom image, I'll gladly add support for it to sdm.

gitbls, 2024-02-11:
I don't need a patch. Just tell me what you did, and I'll see if it's something that can be incorporated into sdm.

atifghaffar, 2024-02-11:
sure. 

For changing the splash image to I used the default raspberry plymouth theme called pix and changed its splash image.

with the  following options

--plugin quietness:"quiet|nosplash|plymouth" \
--plugin apps:"apps=rpd-plym-splash"  \
--plugin copyfile:"from=./files/mysplash.png|to=/usr/share/plymouth/themes/pix/splash.png|chown=root.root|chmod=0644|mkdirif"  \

atifghaffar, 2024-02-11:
you can find some more optimizations here 
https://retropie.org.uk/docs/FAQ/#how-do-i-hide-the-boot-text

for hiding the rainbow image:
disable_splash=1 in /boot/config.txt ( I do it in my first-run script )

gitbls, 2024-02-11:
Thx will have a look at it.

EricAndrechek, 2025-04-06:
> For changing the splash image to I used the default raspberry plymouth theme called pix and changed its splash image.
>
> with the following options
>
> --plugin quietness:"quiet|nosplash|plymouth" [...]

> for hiding the rainbow image: disable_splash=1 in /boot/config.txt ( I do it in my first-run script )

Can confirm this worked out-of-the-box on a new install for me on my second try – I just needed to remove the `splash.png` from the `to=` string to make it:

```bash
# sudo sdm --customize and various flags here
# flags added to get splash working:
--plugin apps:"apps=rpd-plym-splash" \
--plugin bootconfig:"disable_splash=1" \
--plugin quietness:"consoleblank=0|quiet|nosplash|plymouth" \
--plugin copyfile:"from=./splash.png|to=/usr/share/plymouth/themes/pix/|chown=root.root|chmod=0644|mkdirif"
```

Thanks @atifghaffar for documenting this and @gitbls for the awesome program!

atifghaffar, 2025-04-06:
you are welcome @EricAndrechek