gitbls/sdm #176: Error when mounting img

pierreCAMANALYTICS, 2024-02-17:
Hello Benn,

First of all, thank you very much for your work, it's just awesome ! 

I just got a new laptop and when I tried my usual routine on sdm I get this error:

sudo sdm --mount 2023-05-03-raspios-bullseye-arm64-lite.img and ->

**mount: /mnt/sdm: wrong fs type, bad option, bad superblock on /dev/loop13, missing codepage or helper program, or other error.**

Could you help me on this topic please ?

gitbls, 2024-02-17:
That's a curious error, for sure. A couple of questions:
* What OS and version running on your laptop?
* Was the IMG a fresh download from the RasPiOS site, or modified in some way? If modified, how?
* Please provide the output from the following commands:
  * `sudo sfdisk -l 2023-05-03-raspios-bullseye-arm64-lite.img `
  * `sudo losetup -a`
  *  `df -h`

pierreCAMANALYTICS, 2024-02-17:
What OS and version running on your laptop?
**Ubuntu 20.04  and  sdm V11.4**

Was the IMG a fresh download from the RasPiOS site, or modified in some way? If modified, how?
**The image was a fresh download, not modified, I tested with 2022-09-22-raspios-bullseye-arm64-lite.img and 2023-05-03-raspios-bullseye-arm64-lite.img **

Please provide the output from the following commands

- sudo sfdisk -l 2023-05-03-raspios-bullseye-arm64-lite.img ->
`Disk 2023-05-03-raspios-bullseye-arm64-lite.img: 1,98 GiB, 2101346304 bytes, 4104192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0x544c6228

Device                                      Boot  Start     End Sectors  Size Id Type
2023-05-03-raspios-bullseye-arm64-lite.img1        8192  532479  524288  256M  c W95 
2023-05-03-raspios-bullseye-arm64-lite.img2      532480 4104191 3571712  1,7G 83 Linu
`

- sudo losetup -a
`/dev/loop1: [66306]:30960129 (/var/lib/snapd/snaps/core20_2182.snap)
/dev/loop8: [66306]:30959444 (/var/lib/snapd/snaps/gnome-3-38-2004_143.snap)
/dev/loop6: [66306]:30941870 (/var/lib/snapd/snaps/snapd_20671.snap)
/dev/loop13: [66306]:30941502 (/var/lib/snapd/snaps/snapd_18357.snap)
/dev/loop4: [66306]:30941499 (/var/lib/snapd/snaps/gnome-3-38-2004_119.snap)
/dev/loop11: [66306]:30935667 (/var/lib/snapd/snaps/core22_1122.snap)
/dev/loop2: [66306]:30960122 (/var/lib/snapd/snaps/gnome-42-2204_141.snap)
/dev/loop0: [66306]:30941497 (/var/lib/snapd/snaps/bare_5.snap)
/dev/loop9: [66306]:30960027 (/var/lib/snapd/snaps/core22_1033.snap)
/dev/loop7: [66306]:30933433 (/var/lib/snapd/snaps/snap-store_959.snap)
/dev/loop5: [66306]:30938251 (/var/lib/snapd/snaps/rpi-imager_606.snap)
/dev/loop12: [66306]:30960026 (/var/lib/snapd/snaps/core20_2105.snap)
/dev/loop3: [66306]:30941500 (/var/lib/snapd/snaps/gtk-common-themes_1535.snap)
/dev/loop10: [66306]:30941501 (/var/lib/snapd/snaps/snap-store_638.snap)`

- df -h
`Filesystem      Size  Used Avail Use% Mounted on
udev            3,8G     0  3,8G   0% /dev
tmpfs           767M  2,5M  764M   1% /run
/dev/nvme0n1p2  468G   23G  422G   6% /
tmpfs           3,8G   23M  3,8G   1% /dev/shm
tmpfs           5,0M  4,0K  5,0M   1% /run/lock
tmpfs           3,8G     0  3,8G   0% /sys/fs/cgroup
/dev/loop0      128K  128K     0 100% /snap/bare/5
/dev/loop2      497M  497M     0 100% /snap/gnome-42-2204/141
/dev/loop3       92M   92M     0 100% /snap/gtk-common-themes/1535
/dev/loop1       64M   64M     0 100% /snap/core20/2182
/dev/loop4      347M  347M     0 100% /snap/gnome-3-38-2004/119
/dev/loop6       41M   41M     0 100% /snap/snapd/20671
/dev/loop5      207M  207M     0 100% /snap/rpi-imager/606
/dev/loop7       13M   13M     0 100% /snap/snap-store/959
/dev/loop10      46M   46M     0 100% /snap/snap-store/638
/dev/loop8      350M  350M     0 100% /snap/gnome-3-38-2004/143
/dev/loop12      64M   64M     0 100% /snap/core20/2105
/dev/loop9       75M   75M     0 100% /snap/core22/1033
/dev/loop11      75M   75M     0 100% /snap/core22/1122
/dev/loop13      50M   50M     0 100% /snap/snapd/18357
/dev/nvme0n1p1  511M   20M  492M   4% /boot/efi
tmpfs           767M   24K  767M   1% /run/user/1000`
tmpfs           767M   24K  767M   1% /run/user/1000`

Thank you for your support !

gitbls, 2024-02-17:
OK, nothing odd there except a pantload of loop devices. Doubt that's the issue, just noting for now. Please try this:
* Copy/save the mini-nspawn sript here: https://github.com/gitbls/sdm/discussions/178
* `chmod 755 mini-nspawn`
* `sudo ./mini-nspawn 2023-05-03-raspios-bullseye-arm64-lite.img`
* If it works you should see:
```
* Human-readable partition list
Model:  (file)
Disk /ssdy/work/2023-12-05-raspios-bookworm-arm64-lite.img: 2738880512B
Sector size (logical/physical): 512B/512B
Partition Table: msdos
Disk Flags: 

Number  Start       End          Size         Type     File system  Flags
 1      4194304B    541065215B   536870912B   primary  fat32        lba
 2      541065216B  2738880511B  2197815296B  primary  ext4

* Machine-readable partition list
BYT;
/ssdy/work/2023-12-05-raspios-bookworm-arm64-lite.img:2738880512B:file:512:512:msdos::;
1:4194304B:541065215B:536870912B:fat32::lba;
2:541065216B:2738880511B:2197815296B:ext4::;
mount: /dev/loop0p2 mounted on /mnt/sdm.
mount: /dev/loop0p1 mounted on /mnt/sdm/boot/firmware.
Spawning container sdm on /mnt/sdm.
Press ^] three times within 1s to kill container.
```
You can either `exit` or press CTRL-] 3 times to exit.

* If it doesn't work, what's the output?

pierreCAMANALYTICS, 2024-02-18:
I got the expected output :
[image]

pierreCAMANALYTICS, 2024-02-18:
Ok Benn, I don't know what solved the problem but after many hard reboots, it actually works !! Thank you for your support, again you're doing  an amazing work !  Greetings from france

closed 2024-02-18 by pierreCAMANALYTICS

hajekPBH, 2024-02-20:
hi, same problem here, 2 different machines (both Linux Mint 20.2 base od Ubuntu 20.04) and i get error: 

**mount: /mnt/sdm: wrong fs type, bad option, bad superblock on /dev/loop26, missing codepage or helper program, or other error**

sdm v11.4

mini-nspawn output is ok

i work with current _2023-12-11-raspios-bookworm-armhf-lite.img_ image

thank you

gitbls, 2024-02-20:
> hi, same problem here, 2 different machines (both Linux Mint 20.2 based on Ubuntu 20.04) and i get error:
> 
> **mount: /mnt/sdm: wrong fs type, bad option, bad superblock on /dev/loop26, missing codepage or helper program, or other error mount: /mnt/sdm/boot: mount point does not exist ? Error mounting IMG '2023-12-11-raspios-bookworm-armhf-lite.img'**
> 
> i run command from documentation: `sudo sdm --customize --plugin user:"adduser=bls|password=mypassword" --plugin L10n:host --plugin disables:piwiz --regen-ssh-host-keys --restart 2023-12-11-raspios-bookworm-armhf-lite.img`
> 
> sdm v11.4
> 
> mini-nspawn output is ok without errors
> 
> i work with freshly downloaded latest _2023-12-11-raspios-bookworm-armhf-lite.img_ image
> 
> thank you

And are your Linux and Ubuntu systems running on X64 or aarch64?

hajekPBH, 2024-02-20:
> And are your Linux and Ubuntu systems running on X64 or aarch64?

thank you for your response, i am running on "x86_64"

gitbls, 2024-02-20:
> thank you for your response, i am running on "x86_64"

I just installed Mint 20.2, fully updated it, and am unable to replicate this issue using sdm V11.4.

You'll have to provide me with a reliable repro in order for me to investigate further.

Thx

ondrejhajek, 2024-02-20:
> > > > hi, same problem here, 2 different machines (both Linux Mint 20.2 based on Ubuntu 20.04) and i get error:
> > > > **mount: /mnt/sdm: wrong fs type, bad option, bad superblock on /dev/loop26, missing codepage or helper program, or other error mount: /mnt/sdm/boot: mount point does not exist ? Error mounting IMG '2023-12-11-raspios-bookworm-armhf-lite.img'**
> > > > i run command from documentation: `sudo sdm --customize --plugin user:"adduser=bls|password=mypassword" --plugin L10n:host --plugin disables:piwiz --regen-ssh-host-keys --restart 2023-12-11-raspios-bookworm-armhf-lite.img`
> > > > sdm v11.4
> > > > mini-nspawn output is ok without errors
> > > > i work with freshly downloaded latest _2023-12-11-raspios-bookworm-armhf-lite.img_ image
> > > > thank you
> > > 
> > > 
> > > And are your Linux and Ubuntu systems running on X64 or aarch64?
> > 
> > 
> > thank you for your response, i am running on "x86_64"
> 
> I just installed Mint 20.2, fully updated it, and am unable to replicate this issue using sdm V11.4.
> 
> You'll have to provide me with a reliable repro in order for me to investigate further.
> 
> Thx
> 
> Also, in the future, please start a new Issue (referencing an existing issue if relevant) rather than piggy-backing on an existing issue, especially one that's already closed. Thx!

ok, sorry, i started new issue #182 (there is a error that i have found id dmesg)