gitbls/sdm #177: parted plugin - Internal error: File '//sdm-cparse' not found in readparams

stb68, 2024-02-17:
Hello

I'm running version 11.4 on a Pi 3B+ running Bookworm 32 bit and I got the above error from the following burn command:

```
sudo sdm --burn /dev/sda --no-expand-root --hostname GatewayPi \
 --burn-plugin parted:"rootexpand=3724|addpartition=7680,ext4"   2023-12-11-raspios-bookworm-armhf-lite.img 
```

The command output shows the copy to card finishing and then:

```
3611295744 bytes (3.6 GB, 3.4 GiB) copied, 341.152 s, 10.6 MB/s
> Set new disk ID '4875d4d8' on '/dev/sda'
Disk identifier changed from 0x57c84f67 to 0x4875d4d8.

The partition table has been altered.
Calling ioctl() to re-read partition table.
Syncing disks.
> Disk ID 4875d4d8 set successfully with sfdisk
* Mount device '/dev/sda'
mount: (hint) your fstab has been modified, but systemd still uses
       the old version; use 'systemctl daemon-reload' to reload.
mount: /dev/sda2 mounted on /mnt/sdm.
mount: (hint) your fstab has been modified, but systemd still uses
       the old version; use 'systemctl daemon-reload' to reload.
mount: /dev/sda1 mounted on /mnt/sdm/boot/firmware.
> Flush accumulated burn log messages
> Force --no-expand-root for parted burn-plugin
> Force --regen-ssh-host-keys for --no-expand-root
> Set MBR partition Disk ID in cmdline.txt
> Set MBR partition Disk ID in fstab
> Disable unneeded RasPiOS firstboot service and /etc/init.d/resize2fs_once
> Disable regenerate_ssh_host_keys service; sdm FirstBoot will run it instead
> Set hostname 'GatewayPi'
> First System Boot Custom Boot Scripts disabled
> First System Boot automatic restart enabled
* Burn Completed
umount: /mnt/sdm/boot/firmware unmounted
umount: /mnt/sdm unmounted
* Run Post-Burn plugins
> Run Plugins Phase 'burn-complete'
> Run Plugin 'parted' (/usr/local/sdm/plugins/parted) Phase burn-complete with arguments: 'rootexpand=3724|addpartition=7680,ext4|burndev=/dev/sda|burnfilefile=|imgtype=Device'
? Internal error: File '//sdm-cparse' not found in readparams
/usr/local/sdm/plugins/parted: line 235: plugin_getargs: command not found
? Plugin 'parted' exited with failure
```

Do you have any ideas why this would happen?

Thanks

gitbls, 2024-02-17:
* On the Pi3B+ host, do `ls -l /etc/sdm`.
* If the directory exists, see if there's a file there named `cparse` but no file named `cparams`. If there is, do `sudo mv /etc/sdm/cparse /etc/sdm/cparams`. Fumble-fingered edit in the installer if this fixes it.
* If the directory does not exist, do this and then retry the sdm command. If you installed sdm into a directory other than /usr/local/sdm, edit this before running and change `sdmdir` setting to be correct
```
!/bin/bash                                                                                                                                                               

sudo="sudo"
sdmdir="/usr/local/sdm"

$sudo mkdir -p /etc/sdm
$sudo chmod 700 /etc/sdm
$sudo rm -f /etc/sdm/sdm-readparams /etc/sdm/cparams.old
$sudo cp -a $sdmdir/sdm-readparams /etc/sdm
$sudo rm -f /etc/sdm/cparams.old
$sudo mv /etc/sdm/cparams /etc/sdm/cparams.old 2>/dev/null
cat <<EOF | $sudo tee /etc/sdm/cparams >/dev/null                                                                                                                         
sdmdir:$sdmdir                                                                                                                                                            
src:$sdmdir                                                                                                                                                               
plugindebug:0                                                                                                                                                             
EOF                                                                                                                                                                       
echo "sdm config updated"
```

stb68, 2024-02-17:
Thank you, the mv command sorted it

gitbls, 2024-02-17:
Great. Thx for the update. The install script has been updated/unduffed.

closed 2024-02-25 by gitbls