gitbls/sdm #179: keymap not being set

stb68, 2024-02-17:
Hello

I have another issue that I've noticed for a number of my tests but haven't really looked at it until now.

During customize, I use **--plugin L10n:host** but the keyboard on the new Pi isn't set to a British format as the host is (so i get a '#' where I should get '£', etc).  Pretty sure I had this on earlier versions running on Mint and I'm still getting it on 11.4 running on a Pi with Bookworm 32 bit.

I've pulled the following out from the logs, hope it helps...

```
2024-02-16 21:07:20 > Command Line: /usr/local/bin/sdm --customize --showapt --plugin L10n:host --plugin 

2024-02-16 21:07:20 * Plugins selected:
2024-02-16 21:07:20    * L10n
2024-02-16 21:07:20        Args: host

2024-02-16 21:07:21 > Run Plugins Phase '0'
2024-02-16 21:07:21 > Run Plugin 'L10n' (/mnt/sdm/usr/local/sdm/plugins/L10n) Phase 0 with arguments: 'host'
2024-02-16 21:07:21 * Plugin L10n: Start Phase 0
2024-02-16 21:07:21 > Plugin L10n: Keys/values found:
2024-02-16 21:07:21    host: 
2024-02-16 21:07:21 > Plugin L10n: Read L10n configuration from host system
2024-02-16 21:07:21 > Plugin L10n: Load Localization (L10N) settings from running system
2024-02-16 21:07:21 > Plugin L10n:   Keymap:       gb
2024-02-16 21:07:21 > Plugin L10n:   Locale:       en_GB.UTF-8
2024-02-16 21:07:21 > Plugin L10n:   Timezone:     Europe/London
2024-02-16 21:07:21 > Plugin L10n:   WiFi Country: 
2024-02-16 21:07:21 > Plugin L10n: Save L10n configuration
2024-02-16 21:07:21 * Plugin L10n: Complete Phase 0

2024-02-16 21:07:52 > Run Plugins Phase '1'
2024-02-16 21:07:53 > Run Plugin 'L10n' (/usr/local/sdm/plugins/L10n) Phase 1 with arguments: 'host'
2024-02-16 21:07:53 * Plugin L10n: Start Phase 1
2024-02-16 21:07:54 % Plugin L10n: Keymap will be set during FirstBoot
2024-02-16 21:07:54 > Plugin L10n: Disable keyboard-setup service; will be re-enabled during FirstBoot
2024-02-16 21:07:55 > Plugin L10n: Set locale to 'en_GB.UTF-8'
2024-02-16 21:08:19 > Plugin L10n: Set timezone to 'Europe/London'
2024-02-16 21:08:21 * Plugin L10n: Complete Phase 1

2024-02-16 21:21:49 > Run Plugins Phase 'post-install'
2024-02-16 21:21:49 > Run Plugin 'L10n' (/usr/local/sdm/plugins/L10n) Phase post-install with arguments: 'host'
2024-02-16 21:21:49 > Run Plugin 'disables'

$  sudo journalctl -b -1 | grep FirstBoot
Dec 11 01:24:46 GatewayPi systemd[1]: systemd-firstboot.service - First Boot Wizard was skipped because of an unmet condition check (ConditionFirstBoot=yes).
Dec 11 01:24:46 GatewayPi systemd[1]: first-boot-complete.target - First Boot Complete was skipped because of an unmet condition check (ConditionFirstBoot=yes).
Dec 11 01:24:55 GatewayPi systemd[1]: Starting sdm-firstboot.service - sdm FirstBoot System Configuration...
Dec 11 01:24:55 GatewayPi systemd[1]: Started sdm-firstboot.service - sdm FirstBoot System Configuration.
Dec 11 01:24:55 GatewayPi root[670]: sdm FirstBoot: Start sdm FirstBoot Service
Dec 11 01:24:55 GatewayPi root[673]: sdm FirstBoot: Start Configuration with configuration file '/etc/sdm/auto-1piboot.conf'
Dec 11 01:24:55 GatewayPi root[676]: sdm FirstBoot: Configuration with configuration file '/etc/sdm/auto-1piboot.conf' Completed
Dec 11 01:24:55 GatewayPi root[679]: sdm FirstBoot: Set WiFi Country to 'GB'
Dec 11 01:25:12 GatewayPi root[976]: sdm FirstBoot: Waiting 10/120 seconds for system time synchronization
Feb 17 21:57:49 GatewayPi root[1092]: sdm FirstBoot: System time synchronization achieved
Feb 17 21:57:49 GatewayPi root[1098]: sdm FirstBoot: Run Generated FirstBoot Script '/etc/sdm/0piboot/010-configure-keyboard.sh'
Feb 17 21:58:00 GatewayPi root[1552]: sdm FirstBoot: Run Generated FirstBoot Script '/etc/sdm/0piboot/050-disable-bt.sh'
Feb 17 21:58:00 GatewayPi root[1557]: sdm FirstBoot: Disable bluetooth using /etc/modprobe.d/blacklist-sdm-bluetooth.conf
Feb 17 21:58:02 GatewayPi root[1579]: sdm FirstBoot: Skip Custom FirstBoot Scripts
Feb 17 21:58:05 GatewayPi root[1603]: sdm FirstBoot: Regenerate SSH Host Keys
Feb 17 21:58:13 GatewayPi root[1634]: sdm FirstBoot: set boot behaviour to B1
Feb 17 21:58:18 GatewayPi root[1680]: sdm FirstBoot: Re-enable getty@tty1
Feb 17 21:58:20 GatewayPi root[1705]: sdm FirstBoot: System will restart in 20 seconds
Feb 17 21:58:42 GatewayPi root[1728]: sdm FirstBoot: System restarting now
Feb 17 21:58:43 GatewayPi systemd[1]: Stopped sdm-firstboot.service - sdm FirstBoot System Configuration.
```

And the /etc/sdm/0piboot/010-configure-keyboard.sh file contains:
```
#!/bin/bash
source /etc/sdm/sdm-readparams
SUDO_USER=$myuser raspi-config do_configure_keyboard "$keymap" nonint
systemctl enable keyboard-setup
```

The image is Bookworm lite.  I wonder is this is something that worked on Bullseye but which has changed under Bookworm.

Thank you

gitbls, 2024-02-17:
Ack! A bug! Will fix.

gitbls, 2024-02-18:
If you want to fix ahead of the next release coming out, you can do:
```
sudo sed -i 's/\\$keymap/$value/'  /usr/local/sdm/plugins/L10n
```
Update will be out in the next couple of days.

stb68, 2024-02-25:
Yep, that's working.
Thanks

gitbls, 2024-02-25:
Thanks for verifying!

closed 2024-02-25 by gitbls