gitbls/sdm #180: [Systemctl service won't start]

pierreCAMANALYTICS, 2024-02-18:
Hello,

When trying to start a custome service, the service won't start :

pierreCAMANALYTICS, 2024-02-18:
Found the solution, by stopping **userconfig.service**

`sudo systemctl stop userconfig.service`

The service was stopping all other services that wanted to use **multi-user.target**

gitbls, 2024-02-18:
Try NOT stopping/disabling the `userconfig` service, but instead use `--plugin disables:piwiz`. This will disable the userconfig service and a few other things that you don't need.

Also, a couple of other things:
* It doesn't hurt anything, but RasPiOS comes with the `pi` user already in /etc/passwd (but with no password set), so you don't need to `adduser=pi`. It doesn't hurt, of course, to leave it there
* In your camanalytics.service file, since you have specified `User=root` you don't need the `sudo` on the `ExecStart`

gitbls, 2024-03-21:
Closing as resolved.

closed 2024-03-22 by gitbls