gitbls/sdm #181: /etc/skel sub-directories are not propogated to new user

stb68, 2024-02-19:
Hello and sorry, me causing you trouble again!

If I use copyfile to overwrite the /etc/skel/.bashrc before creating a user then the user gets this new version in their home directory, as I would expect.

However, if I use copydir to put a .config directory and contents into /etc/skel then this doesn't get copied to the new user.

I'm sure that I had this working before so maybe it's a recent issue or maybe it's because, before, I had the copyfile/copydir and user creation in the burn command rather than customize.

This is the start of my customize command...

```
sudo sdm --customize --plugin L10n:host --plugin disables:"piwiz|bluetooth|triggerhappy" \
 --regen-ssh-host-keys --apt-dist-upgrade --restart \
 --plugin network:"netman=nm|noipv6|wificountry=GB" \
 --plugin network:"nmconn=/home/stb/Images/myhotspot.nmconnection" \
 --plugin apps:"apps=neovim,tldr,shellcheck,exa,bat|name=commonapps" \
 --plugin apps:"apps=dnsutils,fail2ban,ufw|name=gwapps" \
 --plugin user:"deluser=pi" \
 --plugin copydir:"from=/home/stb/Images/skel/|to=/etc/skel|rsyncopts=-rpti" \
 --plugin copyfile:"from=/home/stb/Images/Bookworm/.bashrc|to=/etc/skel|chown=root:root|chmod=644|mkdirif" \
 --plugin user:"adduser=gw|prompt|redact" \
```

The .config and it's contents get created OK in /etc/skel and the /home/gw/.bashrc is the new version, but there's no /home/gw/.config.

OK, I have asked myself why I'm bothering with /etc/skel, since I'm unlikely to create any extra users, so maybe that would be a valid resolution!  But skel seems like the "correct" way to go and I have had cause to create a new user on another Pi and had to manually copy the config files over from my original user (oh, it was so much hard work!).

Clearly not a major issue, but might be more important to other users.

gitbls, 2024-02-19:
At the moment, it only copies files from /etc/skel. I agree it should do directories also, but you'll be responsible for ensuring protections are correct (the user plugin will set the owner, of course).

stb68, 2024-02-20:
Thanks

Yes, I'm setting the permissions as required on the source files and including the -p flag to rsync (the copydir plugin) to preserve them during the copy into the skel directory.

stb68, 2024-02-25:
Yep, that's working.
Thanks

gitbls, 2024-02-25:
Thx for verifying!

closed 2024-02-25 by gitbls