gitbls/sdm #182: Error mounting IMG (wrong fs type, bad option, bad superblock)

ondrejhajek, 2024-02-20:
Hi, 

i have problem like in this #176  

2 different machines (both Linux Mint 20.2 based on Ubuntu 20.04, x86_64) and i get error:

```
mount: /mnt/sdm: wrong fs type, bad option, bad superblock on /dev/loop26, missing codepage or helper program, or other error
mount: /mnt/sdm/boot: mount point does not exist
? Error mounting IMG '2023-12-11-raspios-bookworm-armhf-lite.img'
```

i run command from documentation: sudo sdm --customize --plugin user:"adduser=bls|password=mypassword" --plugin L10n:host --plugin disables:piwiz --regen-ssh-host-keys --restart 2023-12-11-raspios-bookworm-armhf-lite.img

sdm v11.4

mini-nspawn output is ok without errors

i work with freshly downloaded latest 2023-12-11-raspios-bookworm-armhf-lite.img image

thank you

ondrejhajek, 2024-02-20:
i have found error in linux dmesg:

`EXT4-fs (loop40): VFS: Can't find ext4 filesystem`

ondrejhajek, 2024-02-20:
@gitbls done, i successfully burnt with sdm img to medium and i was able sdm --mount this medium

[image]

gitbls, 2024-02-20:
Thx. That confirms for me that it's a bug in old Ubuntu and Mint releases, and will work around it.

ondrejhajek, 2024-02-20:
`sudo sdm --customize --plugin apps:"apps=@myapps" 2023-12-11-raspios-bookworm-armhf-lite.img`

i get this error:

```
* Mount IMG '2023-12-11-raspios-bookworm-armhf-lite.img'
mount: /mnt/sdm: wrong fs type, bad option, bad superblock on /dev/loop40, missing codepage or helper program, or other error
? IMG '' already attached to a loop device
```

gitbls, 2024-02-20:
OK, thx. Will instrument up more logging to see what's going on.

ondrejhajek, 2024-02-21:
here is output:

```
* Mount IMG '2023-12-11-raspios-bookworm-armhf-lite.img'
Before 'new' mount
mount: /mnt/sdm: wrong fs type, bad option, bad superblock on /dev/loop40, missing codepage or helper program, or other error
Before 'old' mount
? IMG '' already attached to a loop device
```

gitbls, 2024-02-21:
Strange. Investigating...

gitbls, 2024-02-21:
Please try this version.
[sdm-cparse.zip](https://github.com/gitbls/sdm/files/14361014/sdm-cparse.zip)

ondrejhajek, 2024-02-21:
there are some errors, but i think it works:)

in output there are czech words, translation:

odpojení selhalo: Takové zařízení nebo adresa neexistuje - unmount failed: this device or address does not exists
připojeno - mounted
odpojeno, odpojen - unmounted

```
* Mount IMG '2023-12-11-raspios-bookworm-armhf-lite.img'
Before 'new' mount
Before 'old' mount
% Delete dangling loop device '/dev/loop40'
  /dev/loop40 1996488704      0         1  0 /home/ondicz/sdm/2023-12-11-raspios-bookworm-armhf-lite.img   0     512
losetup: /dev/loop40: odpojení selhalo: Takové zařízení nebo adresa neexistuje
mount: /dev/loop40p2 připojeno do /mnt/sdm.
mount: /dev/loop40p1 připojeno do /mnt/sdm/boot/firmware.
After 'old' mount
* Start Configuration
> Command Line: /usr/local/bin/sdm --customize --plugin apps:apps=@myapps 
  2023-12-11-raspios-bookworm-armhf-lite.img
* Host Information
   Hostname: rimmer
   uname: Linux rimmer 5.4.0-172-generic #190-Ubuntu SMP Fri Feb 2 23:24:22 UTC 2024 x86_64 
  x86_64 x86_64 GNU/Linux
   os-release Name: Linux Mint 20.3
   Version: 20.3 (Una) (una)
   Like OS: ubuntu
   sdm Version: V11.4
* IMG Information
   Name: 2023-12-11-raspios-bookworm-armhf-lite.img
   Date: 2023-12-11
   RasPiOS Version: 12
   RasPiOS Architecture: 32-bit armhf
   os-release Version: 12 (bookworm)
% sdm will use systemd-nspawn on this 'x86-64' host
  Retry the command with --chroot if this fails
* Plugins selected:
   * apps
       Args: apps=@myapps
> IMG '2023-12-11-raspios-bookworm-armhf-lite.img' has 299532 1K-blocks (0,3GB, 0,3GiB) free 
> Copy sdm to /usr/local/sdm in the IMG
* Start Phase 0 image customization
> Run Plugins Phase '0'
> Run Plugin 'apps' (/mnt/sdm/usr/local/sdm/plugins/apps) Phase 0 with arguments: 'apps=@myapps'
* Plugin apps: Start Phase 0
> Plugin apps: Keys/values found:
   apps: @myapps
?  file '/home/ondicz/sdm/myapps' not found
? Plugin 'apps' exited with failure
umount: /mnt/sdm/boot/firmware odpojen
umount: /mnt/sdm odpojen
```

gitbls, 2024-02-21:
Great! I will make the "odpojenf" and the "Delete dangling loop devices" messages go away, so it will be pretty seamless. 

This will be in the next release, coming soon. I'll update this thread when it's available.

gitbls, 2024-02-21:
one more thing. Could you check and see if there are any unexpected, sdm-ish looking loop devices hanging around after you run sdm?
```
sudo losetup -a
```

Thx

ondrejhajek, 2024-02-21:
> one more thing. Could you check and see if there are any unexpected, sdm-ish looking loop devices hanging around after you run sdm?
> 
> ```
> sudo losetup -a
> ```
> 
> Thx

i see only my snap applications

gitbls, 2024-02-21:
Could I convince you to try this updated version just to make sure I didn't duff it?

[sdm-cparse.zip](https://github.com/gitbls/sdm/files/14366191/sdm-cparse.zip)

Thx!

ondrejhajek, 2024-02-21:
sure, here it is:

```
* Mount IMG '2023-12-11-raspios-bookworm-armhf-lite.img'
mount: /dev/loop40p2 připojeno do /mnt/sdm.
mount: /dev/loop40p1 připojeno do /mnt/sdm/boot/firmware.
* Start Configuration
> Command Line: /usr/local/bin/sdm --customize --plugin apps:apps=@myapps 
  2023-12-11-raspios-bookworm-armhf-lite.img
* Host Information
   Hostname: rimmer
   uname: Linux rimmer 5.4.0-172-generic #190-Ubuntu SMP Fri Feb 2 23:24:22 UTC 2024 x86_64 
  x86_64 x86_64 GNU/Linux
   os-release Name: Linux Mint 20.3
   Version: 20.3 (Una) (una)
   Like OS: ubuntu
   sdm Version: V11.4
* IMG Information
   Name: 2023-12-11-raspios-bookworm-armhf-lite.img
   Date: 2023-12-11
   RasPiOS Version: 12
   RasPiOS Architecture: 32-bit armhf
   os-release Version: 12 (bookworm)
% sdm will use systemd-nspawn on this 32-bit x86-64 host
  Retry the command with --chroot if this fails
* Plugins selected:
   * apps
       Args: apps=@myapps
> IMG '2023-12-11-raspios-bookworm-armhf-lite.img' has 299512 1K-blocks (0,3GB, 0,3GiB) free 
> Copy sdm to /usr/local/sdm in the IMG
* Start Phase 0 image customization
> Run Plugins Phase '0'
> Run Plugin 'apps' (/mnt/sdm/usr/local/sdm/plugins/apps) Phase 0 with arguments: 'apps=@myapps'
* Plugin apps: Start Phase 0
> Plugin apps: Keys/values found:
   apps: @myapps
?  file '/home/ondicz/sdm/myapps' not found
? Plugin 'apps' exited with failure
umount: /mnt/sdm/boot/firmware odpojen
umount: /mnt/sdm odpojen

```

gitbls, 2024-02-23:
This is all fixed 🤞in V11.5. Please test/verify. Thx!

ondrejhajek, 2024-02-25:
v11.5 is ok

gitbls, 2024-02-25:
Excellent. Thanks for helping me sort this out. Still strange that I coudn't repro it here on Mint 20.2 😵

closed 2024-02-25 by gitbls