gitbls/sdm #183: OrangePi OS support

dwatrous, 2024-02-22:
I've been using (and loving) sdm to build images for my raspberry pi zero 2 w. I'm interested in trying the http://www.orangepi.org/html/hardWare/computerAndMicrocontrollers/service-and-support/Orange-Pi-Zero-2W.html, and I wondered if this could work for that OS too?

The os is arch linux based and comes in an xz compressed file https://drive.google.com/drive/folders/1XR01J9-CXk_rtiSa1KdS9CR5rmtXk6Np

The device takes an SD card.

Do you think it's possible to use sdm for the orangepi?

gitbls, 2024-02-23:
I took a very brief look at the IMG. You can certainly use sdm for a subset of things, but there will be many things that just don't work and are critically important for sdm's operation. For instance:
* No `raspi-config` so no setting anything that it depends on
* No `apt`
* ***Tons*** of testing to determine what works, what doesn't work and needs to be fixed, etc.

Changes to *fully* support this OS would likely be fairly extensive and invasive, and not something I'm interested in.

That said, you could use sdm and build your own plugins to do the things that need to be done.

It wouldn't be a huge amount of work to do a plugin/set of plugins to solve your specific problem. In other words, use the sdm "engine" and infrastructure. Many of the plugins *should* work, but obviously need to be tested.

dwatrous, 2024-03-10:
Orange Pi also supports Ubuntu and Debian, which should be closer to the RaspiOS distribution.
http://www.orangepi.org/html/hardWare/computerAndMicrocontrollers/service-and-support/Orange-Pi-Zero-2W.html

I downloaded the bookworm image here: https://drive.google.com/drive/folders/1Lnosr3bmdbG6F9RsDjY_RCLJl2zT4l61. After unzipping, this provides an img file. I used Etcher to burn the image to an SD card. 

I noticed that there is an orangepi-config (similar to raspi-config): https://github.com/orangepi-xunlong/orangepi-build/tree/next

I'm exploring now whether this can be used in a headless way, like raspi-config.

I'll keep updating here and I may try to use sdm with this debian image to see how much progress I make.

gitbls, 2024-03-15:
Cool. Looking forward to hear your results.

gitbls, 2024-06-02:
Closing this issue due to no activity.

@dwatrous please re-open this issue or start a new one if you'd like to discuss/pursue further.

closed 2024-06-02 by gitbls