gitbls/sdm #187: `runatboot` scripts don't run (FirstBoot: Skip Custom FirstBoot Scripts)

garshythoel, 2024-02-25:
Hi! I have the following setup for my project

* I'm developing on a Macbook Pro M1 machine with a UTM Debian 12 Virtual Machine
* I'm using a 32bit bullseye image with a rpi zero w2

The `runatboot` doesn't seem to be working for me. I've confirmed all the other options work fine + the VM has the correct scripts copied.

I've also confirmed all the pre on-hardware boot settings are setup fine, i also confirmed in the sdm --explore image that the on_first_boot.sh is copied over correctly into `usr/local/sdm/1piboot/on_first_boot.sh`

I've also confirmed that the `copyfile` is working correctly but when i boot on the rpi, there's no mention of the script. For reference here are the logs for the boot
```
inku@sdminkboard:~ $ journalctl -b -1 | grep FirstBoot
Dec 04 18:57:22 sdminkboard.local systemd[1]: Starting sdm FirstBoot System Configuration...
Dec 04 18:57:22 sdminkboard.local systemd[1]: Started sdm FirstBoot System Configuration.
Dec 04 18:57:22 sdminkboard.local root[466]: sdm FirstBoot: Start sdm FirstBoot Service
Dec 04 18:57:22 sdminkboard.local root[481]: sdm FirstBoot: Start Configuration with configuration file '/etc/sdm/auto-1piboot.conf'
Dec 04 18:57:22 sdminkboard.local root[489]: sdm FirstBoot: Set i2c to '0'
Dec 04 18:57:23 sdminkboard.local root[516]: sdm FirstBoot: Set spi to '0'
Dec 04 18:57:23 sdminkboard.local root[538]: sdm FirstBoot: Configuration with configuration file '/etc/sdm/auto-1piboot.conf' Completed
Dec 04 18:57:23 sdminkboard.local root[545]: sdm FirstBoot: Set WiFi Country to 'US'
Dec 04 18:57:37 sdminkboard.local root[822]: sdm FirstBoot: Waiting 10/120 seconds for system time synchronization
Dec 04 18:57:49 sdminkboard.local root[897]: sdm FirstBoot: Waiting 20/120 seconds for system time synchronization
Feb 24 18:38:52 sdminkboard.local root[919]: sdm FirstBoot: System time synchronization achieved
Feb 24 18:38:52 sdminkboard.local root[927]: sdm FirstBoot: Run Generated FirstBoot Script '/etc/sdm/0piboot/010-configure-keyboard.sh'
Feb 24 18:38:53 sdminkboard.local root[931]: sdm FirstBoot: Configure keyboard to 'us'
Feb 24 18:39:04 sdminkboard.local root[1228]: sdm FirstBoot: Skip Custom FirstBoot Scripts
Feb 24 18:39:06 sdminkboard.local root[1249]: sdm FirstBoot: Regenerate SSH Host Keys
Feb 24 18:39:13 sdminkboard.local root[1278]: sdm FirstBoot: set boot behaviour to B1
Feb 24 18:39:15 sdminkboard.local root[1314]: sdm FirstBoot: Re-enable getty@tty1
Feb 24 18:39:16 sdminkboard.local root[1334]: sdm FirstBoot: System will restart in 20 seconds
Feb 24 18:39:38 sdminkboard.local root[1360]: sdm FirstBoot: System restarting now
Feb 24 18:39:38 sdminkboard.local systemd[1]: Stopped sdm FirstBoot System Configuration.
```

Here are the files in questions

./create-image.sh
```
#!/bin/bash

# since we're using this image in a VM, we need to copy it to sdm/1piboot there
sudo cp code/on_first_boot.sh /usr/local/sdm/1piboot/on_first_boot.sh

sudo sdm --customize \
    --logwidth 132 \
    --extend --xmb 2048 \
    --hostname sdminkboard.local \
	--plugin @plugins \
    --regen-ssh-host-keys \
    --restart base-image/[clean]2023-12-05-raspios-bullseye-armhf-lite.img

# add modules-load=dwc2,g_ether to boot/config.txt via sdm explore
```

./plugins
```
user:deluser=pi|adduser=inku|password=paper
L10n:host
disables:piwiz
bootconfig:dtoverlay=dwc2
apps:apps=@apps|name=default
copydir:from=code/|to=/home/inku/code/
raspiconfig:i2c=0|spi=0
copyfile:from=/usr/local/sdm/1piboot/on_first_boot.sh|chmod=755|to=/home/inku|runphase=postinstall|mkdirif=yes
runatboot:script=home/inku/on_first_boot.sh|output=/home/inku/firstboot.log
```

./apps
```
python3-venv
git
python3-dev
build-essential
python3-pip
vim
```

./on_first_boot.sh
```
#!/bin/bash

# on boot
# turn on btle
# check if polling service exists
    # if YES: restart polling service
    # if NO: create polling service and start

# inside polling service -> check if Device is registered and authorized
# if not initiate registration/verification

## setup ble service
echo "Resetting bluetooth to factory settings"
sudo systemctl stop bluetooth
sudo systemctl disable bluetooth
sudo rm -rf /var/lib/bluetooth
# there's some issue with having bluetoothd so disable and have hciconfig up
sudo hciconfig hci0 up

#make create-service
# Create the service file with the desired content
echo "[Unit]\nDescription=Poll inkserver periodically\n\n[Service]\nExecStart=usr/bin/python3 home/inku/code/poll_inkserver.py\nRestart=always\n\n[Install]\nWantedBy=multi-user.target" | sudo tee /etc/systemd/system/poll_inkserver.service
	
sudo ln -s /etc/systemd/system/poll_inkserver.service /home/inku/code/poll_inkserver_symlink.service
sudo systemctl enable poll_inkserver.service
sudo systemctl start poll_inkserver.service

# Start the polling service
# sudo python3 ble_comms_service.py
```

Any help would be appreciated, thanks!

gitbls, 2024-02-25:
There are two different and completely independent sdm mechanisms for running scripts at boot:
* Putting scripts into /usr/local/sdm/1piboot/\*.sh
* Using the `runatboot` plugin

The former (1piboot) is an old mechanism that was introduced early on in sdm development. It is, in my opinion, inflexible, and TBH I've been thinking of removing it. It didn't run on your system's first boot because you didn't burn it with the `--bootscripts` command line switch.

The latter (`runatboot` plugin) should have worked and run your script at first system boot, so let's dig into why that didn't happen.

The runatboot plugin takes a script ("home/inku/on_first_boot.sh" in your example), copies it into the IMG in $SDMPT/etc/sdm/assets/runatboot), and sets up $SDMPT/etc/sdm/0piboot/060-runatboot.sh to run that script during sdm FirstBoot.

In your specific example, the file "home/inku/on_first_boot.sh" was not found (the leading "/" is missing), and the `runatboot` plugin has a found-just-now, need-to-fix bug where it fails to report that the script was not found. 😵

So, could you add the missing "/" to your `runatboot` plugin line, and retry the customize, burn, and boot? My guess is that it will work. If it doesn't, then please provide the output from `journalctl | grep FirstBoot` as you did here, as well as the output from `ls -l /etc/sdm/runatboot` in the customized IMG.

And while you're updating things, remove /usr/local/sdm/1piboot/on_first_boot.sh. Per my notes above, it's not what you want to be using.

I will add a note to the `runatboot` documentation in Plugins.md regarding 1piboot/\*.sh if I decide to keep the 1piboot/\*.sh mechanism. At this point I'm leaning toward removing it as a result of your feedback.

Lastly, as far as a more complete end-to-end example, I'm all for that as well. Can you provide a more detailed (not code) description of what you think would be helpful?

Thanks for your detailed and thoughtful feedback. I really appreciate it.

garshythoel, 2024-02-25:
Hi!

Thank you for the detailed response - this helps my understanding of the tooling.

Unfortunately i had already started a fix when i got he notif of your message and didn't see the edit so i don't have the old image anymore. That said maybe we can still figure it out from here.

The only changes I've done are:
1. Remove the copy from host -> image in the `./create-image.sh` script.
2. Removed the `copyfile` plugin from the plugins file.
3. Added a `/` to the `runatboot` (also can you confirm that the `script` should point to `/path/on/host` ? = which is `/code/on_first_boot.sh` on the host)

This is my `plugins` file now
```
user:deluser=pi|adduser=inku|password=paper
L10n:host
disables:piwiz
bootconfig:dtoverlay=dwc2
apps:apps=@apps|name=default
copydir:from=code/|to=/home/inku/code/
raspiconfig:i2c=0|spi=0
runatboot:script=/code/on_first_boot.sh|output=/home/inku/firstboot.log
```

Here are the contents of `/etc/sdm/history` with this run:

```
root@sdm:/# cat /etc/sdm/history
2024-02-25 10:14:29 * Start Configuration
2024-02-25 10:14:29 > Command Line: /usr/local/bin/sdm --customize --logwidth 132 --extend --xmb 2048 --hostname sdminkboard.local --plugin @plugins 
2024-02-25 10:14:29   --regen-ssh-host-keys --restart base-image/[clean]2023-12-05-raspios-bullseye-armhf-lite.img
2024-02-25 10:14:29 * Host Information
2024-02-25 10:14:29    Hostname: debian
2024-02-25 10:14:29    uname: Linux debian 6.1.0-18-arm64 #1 SMP Debian 6.1.76-1 (2024-02-01) aarch64 GNU/Linux
2024-02-25 10:14:29    os-release Name: Debian GNU/Linux 12 (bookworm)
2024-02-25 10:14:29    Version: 12 (bookworm)
2024-02-25 10:14:29    sdm Version: V11.5
2024-02-25 10:14:29 * IMG Information
2024-02-25 10:14:29    Name: base-image/[clean]2023-12-05-raspios-bullseye-armhf-lite.img
2024-02-25 10:14:29    Date: 2023-12-05
2024-02-25 10:14:29    RasPiOS Version: 11
2024-02-25 10:14:29    RasPiOS Architecture: 32-bit armhf
2024-02-25 10:14:29    os-release Version: 11 (bullseye)
2024-02-25 10:14:29 % sdm will use qemu 'arm'
2024-02-25 10:14:29 % sdm will use systemd-nspawn on this 64-bit ARM aarch64 host
2024-02-25 10:14:29   Retry the command with --chroot if this fails
2024-02-25 10:14:29 * Plugins selected:
2024-02-25 10:14:29    * user
2024-02-25 10:14:29        Args: deluser=pi|adduser=inku|password=paper
2024-02-25 10:14:29    * L10n
2024-02-25 10:14:29        Args: host
2024-02-25 10:14:29    * disables
2024-02-25 10:14:29        Args: piwiz
2024-02-25 10:14:29    * bootconfig
2024-02-25 10:14:29        Args: dtoverlay=dwc2
2024-02-25 10:14:29    * apps
2024-02-25 10:14:29        Args: apps=@apps|name=default
2024-02-25 10:14:29    * copydir
2024-02-25 10:14:29        Args: from=code/|to=/home/inku/code/
2024-02-25 10:14:29    * raspiconfig
2024-02-25 10:14:29        Args: i2c=0|spi=0
2024-02-25 10:14:29 > IMG 'base-image/[clean]2023-12-05-raspios-bullseye-armhf-lite.img' has 2275184 1K-blocks (2.3GB, 2.2GiB) free 
2024-02-25 10:14:29 > Copy sdm to /usr/local/sdm in the IMG
2024-02-25 10:14:29 > Copy --plugin @file(s) to /mnt/sdm/etc/sdm/assets/pluglist
2024-02-25 10:14:29   plugins
2024-02-25 10:14:29 * Start Phase 0 image customization
2024-02-25 10:14:29 > Run Plugins Phase '0'
2024-02-25 10:14:29 > Run Plugin 'user' (/mnt/sdm/usr/local/sdm/plugins/user) Phase 0 with arguments: 'deluser=pi|adduser=inku|password=paper'
2024-02-25 10:14:29 * Plugin user: Start Phase 0
2024-02-25 10:14:29 > Plugin user: Keys/values found:
2024-02-25 10:14:29    deluser: pi
[... 107 lines cut ...]
2024-02-25 10:17:32 > Run Plugin 'disables' (/usr/local/sdm/plugins/disables) Phase post-install with arguments: 'piwiz'
2024-02-25 10:17:33 * Plugin disables: Start Phase post-install
2024-02-25 10:17:33 > Plugin disables: Disable running piwiz and userconfig on first system boot
2024-02-25 10:17:33 * Plugin disables: Complete Phase post-install
2024-02-25 10:17:33 > Run Plugin 'bootconfig' (/usr/local/sdm/plugins/bootconfig) Phase post-install with arguments: 'dtoverlay=dwc2'
2024-02-25 10:17:33 * Plugin bootconfig: Start Phase post-install
2024-02-25 10:17:33 * Plugin bootconfig: Complete Phase post-install
2024-02-25 10:17:33 > Run Plugin 'apps' (/usr/local/sdm/plugins/apps) Phase post-install with arguments: 'apps=@apps|name=default'
2024-02-25 10:17:33 * Plugin apps: Start Phase post-install
2024-02-25 10:17:33 * Plugin apps: Complete Phase post-install
2024-02-25 10:17:33 > Run Plugin 'copydir' (/usr/local/sdm/plugins/copydir) Phase post-install with arguments: 'from=code/|to=/home/inku/code/'
2024-02-25 10:17:34 * Plugin copydir: Start Phase post-install
2024-02-25 10:17:34 * Plugin copydir: Complete Phase post-install
2024-02-25 10:17:34 > Run Plugin 'raspiconfig' (/usr/local/sdm/plugins/raspiconfig) Phase post-install with arguments: 'i2c=0|spi=0'
2024-02-25 10:17:34 * Plugin raspiconfig: Start Phase post-install
2024-02-25 10:17:34 * Plugin raspiconfig: Complete Phase post-install
2024-02-25 10:17:34 > Run graphics post-install configuration
2024-02-25 10:17:34 > Plugin sdm-phase1: No known Display Manager found
2024-02-25 10:17:34 > Plugin sdm-phase1: Set RasPiOS to B1:'console no login' for sdm FirstBoot
2024-02-25 10:17:34 > Enable SSH
2024-02-25 10:17:34 > Plugin phase1: Enable SSH service via /boot/ssh
2024-02-25 10:17:34 > IMG 'base-image/[clean]2023-12-05-raspios-bullseye-armhf-lite.img' has 2063240 1K-blocks (2.1GB, 2.0GiB) free at end of image customization
2024-02-25 10:17:35 * Enter Shell Command Prompt
2024-02-25 10:17:35   'exit' to exit back to host system
```

garshythoel, 2024-02-25:
```
2024-02-25 10:14:29 * Plugins selected:
2024-02-25 10:14:29    * user
2024-02-25 10:14:29        Args: deluser=pi|adduser=inku|password=paper
2024-02-25 10:14:29    * L10n
2024-02-25 10:14:29        Args: host
2024-02-25 10:14:29    * disables
2024-02-25 10:14:29        Args: piwiz
2024-02-25 10:14:29    * bootconfig
2024-02-25 10:14:29        Args: dtoverlay=dwc2
2024-02-25 10:14:29    * apps
2024-02-25 10:14:29        Args: apps=@apps|name=default
2024-02-25 10:14:29    * copydir
2024-02-25 10:14:29        Args: from=code/|to=/home/inku/code/
2024-02-25 10:14:29    * raspiconfig
2024-02-25 10:14:29        Args: i2c=0|spi=0
```
interesting it doesn't look like it picks up the runatboot plugin at all ?

gitbls, 2024-02-25:
> ```
> 2024-02-25 10:14:29 * Plugins selected:
> 2024-02-25 10:14:29    * user
> 2024-02-25 10:14:29        Args: deluser=pi|adduser=inku|password=paper
> 2024-02-25 10:14:29    * L10n
> 2024-02-25 [...]

Yes, I saw that immediately and am looking into it. Thx!

garshythoel, 2024-02-25:
Awesome!

If it helps, here is the filetree on the host
```
inkboard-device/
┣ base-image/
┃ ┣ saved/
┃ ┣ [clean]2023-12-05-raspios-bullseye-armhf-lite.img
┃ ┗ [clean]2023-12-05-raspios-bullseye-armhf-lite.img.xz
┣ code/
┃ ┣ Makefile
┃ ┣ ble_comms_service.py
┃ ┣ inkdevice.py
┃ ┣ on_first_boot.sh
┃ ┣ poll_inkserver.py
┃ ┣ requirements.txt
┃ ┣ reset_bt_to_factory.sh
┃ ┗ wifiutils.py
┣ .gitignore
┣ apps
┣ create-image.sh
┣ forpi.img
┗ plugins
```

And the working directory is inside `/inkboard-device`

the `/code` directory gets copied over to the image and then lives at `/home/inku/code`

gitbls, 2024-02-25:
I notice on the `copydir` pugin you use `code/` instead of `/code/`. could you retry with the correct leading "/" and LMK how that goes?

Your leading "/" omission has pointed out a code improvement opportunity in the `copydir` plugin

gitbls, 2024-02-25:
> Awesome!
>
> If it helps, here is the filetree on the host
>
> ```
> inkboard-device/
> ┣ base-image/
> ┃ ┣ saved/
> ┃ ┣ [clean]2023-12-05-raspios-bullseye-armhf-lite.img
> ┃ ┗ [...]

If the `code` directory got copied, it's only because of luck. There's no guarantee on where the sdm processes will be running current directory-wise, so you need to provide the full /path/to/whatever ...everywhere in sdm

garshythoel, 2024-02-25:
ahhh - okay, here is the run w/ the updated `plugins`

Added the full paths to the script on the VM host (so `/media/share/inkboard-device/*` rather than `/*` )

```
user:deluser=pi|adduser=inku|password=paper
L10n:host
disables:piwiz
bootconfig:dtoverlay=dwc2
apps:apps=@apps|name=default
copydir:from=/media/share/inkboard-device/code/|to=/home/inku/code/
runatboot:script=/media/share/inkboard-device/code/on_first_boot.sh|output=/home/inku/firstboot.log
raspiconfig:i2c=0|spi=0
```

Seems like that did the trick at least in terms of setup - I'm gonna confirm with a proper firstboot

```
root@sdm:/# cat /etc/sdm/history
2024-02-25 10:36:30 * Start Configuration
2024-02-25 10:36:30 > Command Line: /usr/local/bin/sdm --customize --logwidth 132 --extend --xmb 2048 --hostname sdminkboard.local --plugin @plugins 
2024-02-25 10:36:30   --regen-ssh-host-keys --restart base-image/[clean]2023-12-05-raspios-bullseye-armhf-lite.img
2024-02-25 10:36:30 * Host Information
2024-02-25 10:36:30    Hostname: debian
2024-02-25 10:36:30    uname: Linux debian 6.1.0-18-arm64 #1 SMP Debian 6.1.76-1 (2024-02-01) aarch64 GNU/Linux
2024-02-25 10:36:30    os-release Name: Debian GNU/Linux 12 (bookworm)
2024-02-25 10:36:30    Version: 12 (bookworm)
2024-02-25 10:36:30    sdm Version: V11.5
2024-02-25 10:36:30 * IMG Information
2024-02-25 10:36:30    Name: base-image/[clean]2023-12-05-raspios-bullseye-armhf-lite.img
2024-02-25 10:36:30    Date: 2023-12-05
2024-02-25 10:36:30    RasPiOS Version: 11
2024-02-25 10:36:30    RasPiOS Architecture: 32-bit armhf
2024-02-25 10:36:30    os-release Version: 11 (bullseye)
2024-02-25 10:36:30 % sdm will use qemu 'arm'
2024-02-25 10:36:30 % sdm will use systemd-nspawn on this 64-bit ARM aarch64 host
2024-02-25 10:36:30   Retry the command with --chroot if this fails
2024-02-25 10:36:30 * Plugins selected:
2024-02-25 10:36:30    * user
2024-02-25 10:36:30        Args: deluser=pi|adduser=inku|password=paper
2024-02-25 10:36:30    * L10n
2024-02-25 10:36:30        Args: host
2024-02-25 10:36:30    * disables
2024-02-25 10:36:30        Args: piwiz
2024-02-25 10:36:30    * bootconfig
2024-02-25 10:36:30        Args: dtoverlay=dwc2
2024-02-25 10:36:30    * apps
2024-02-25 10:36:30        Args: apps=@apps|name=default
2024-02-25 10:36:30    * copydir
2024-02-25 10:36:30        Args: from=/media/share/inkboard-device/code/|to=/home/inku/code/
2024-02-25 10:36:30    * runatboot
2024-02-25 10:36:30        Args: script=/media/share/inkboard-device/code/on_first_boot.sh|output=/home/inku/firstboot.log
2024-02-25 10:36:30 > IMG 'base-image/[clean]2023-12-05-raspios-bullseye-armhf-lite.img' has 2275184 1K-blocks (2.3GB, 2.2GiB) free 
2024-02-25 10:36:30 > Copy sdm to /usr/local/sdm in the IMG
2024-02-25 10:36:30 > Copy --plugin @file(s) to /mnt/sdm/etc/sdm/assets/pluglist
2024-02-25 10:36:30   plugins
2024-02-25 10:36:30 * Start Phase 0 image customization
2024-02-25 10:36:30 > Run Plugins Phase '0'
2024-02-25 10:36:30 > Run Plugin 'user' (/mnt/sdm/usr/local/sdm/plugins/user) Phase 0 with arguments: 'deluser=pi|adduser=inku|password=paper'
2024-02-25 10:36:30 * Plugin user: Start Phase 0
2024-02-25 10:36:30 > Plugin user: Keys/values found:
2024-02-25 10:36:30    deluser: pi
2024-02-25 10:36:30    adduser: inku
2024-02-25 10:36:30    password: paper
2024-02-25 10:36:30 > Plugin user: Create user 'inku' home directory '/home/inku' so available for other plugins
2024-02-25 10:36:30 * Plugin user: Complete Phase 0
2024-02-25 10:36:30 > Run Plugin 'L10n' (/mnt/sdm/usr/local/sdm/plugins/L10n) Phase 0 with arguments: 'host'
2024-02-25 10:36:30 * Plugin L10n: Start Phase 0
2024-02-25 10:36:30 > Plugin L10n: Keys/values found:
2024-02-25 10:36:30    host: 
[... 110 lines cut ...]
2024-02-25 10:39:48 * Plugin bootconfig: Complete Phase post-install
2024-02-25 10:39:48 > Run Plugin 'apps' (/usr/local/sdm/plugins/apps) Phase post-install with arguments: 'apps=@apps|name=default'
2024-02-25 10:39:49 * Plugin apps: Start Phase post-install
2024-02-25 10:39:49 * Plugin apps: Complete Phase post-install
2024-02-25 10:39:49 > Run Plugin 'copydir' (/usr/local/sdm/plugins/copydir) Phase post-install with arguments: 
2024-02-25 10:39:49   'from=/media/share/inkboard-device/code/|to=/home/inku/code/'
2024-02-25 10:39:49 * Plugin copydir: Start Phase post-install
2024-02-25 10:39:49 * Plugin copydir: Complete Phase post-install
2024-02-25 10:39:49 > Run Plugin 'runatboot' (/usr/local/sdm/plugins/runatboot) Phase post-install with arguments: 
2024-02-25 10:39:49   'script=/media/share/inkboard-device/code/on_first_boot.sh|output=/home/inku/firstboot.log'
2024-02-25 10:39:49 * Plugin runatboot: Start Phase post-install
2024-02-25 10:39:49 * Plugin runatboot: Complete Phase post-install
2024-02-25 10:39:49 > Run graphics post-install configuration
2024-02-25 10:39:49 > Plugin sdm-phase1: No known Display Manager found
2024-02-25 10:39:49 > Plugin sdm-phase1: Set RasPiOS to B1:'console no login' for sdm FirstBoot
2024-02-25 10:39:50 > Enable SSH
2024-02-25 10:39:50 > Plugin phase1: Enable SSH service via /boot/ssh
2024-02-25 10:39:50 > IMG 'base-image/[clean]2023-12-05-raspios-bullseye-armhf-lite.img' has 2063228 1K-blocks (2.1GB, 2.0GiB) free at end of image customization
2024-02-25 10:39:50 * Enter Shell Command Prompt
2024-02-25 10:39:50   'exit' to exit back to host system
```

So maybe for documentation purposes just mentioning the need for `/full/path/to/file/on/host` might help make it idiot proof ?

I'll respond to this thread to see if the rest of it works but thank you for the prompt responses! i spent so much time on this yesterday haha

gitbls, 2024-02-25:
Looks like `runatboot` was processed, which is great. Agree on the doc update.

Sorry you spent a bunch of time on this, but I think it was worthwhile. You've turned up a couple of logging bugs (nothing impactful, just things that trigger my OCD 🤣), and a necessary documentation update.

Do LMK when you've got it running successfully.

garshythoel, 2024-02-25:
IT WORKS

```
inku@sdminkboard:~ $ journalctl | grep FirstBoot
Dec 04 18:57:22 sdminkboard.local systemd[1]: Starting sdm FirstBoot System Configuration...
Dec 04 18:57:22 sdminkboard.local systemd[1]: Started sdm FirstBoot System Configuration.
Dec 04 18:57:22 sdminkboard.local root[454]: sdm FirstBoot: Start sdm FirstBoot Service
Dec 04 18:57:22 sdminkboard.local root[459]: sdm FirstBoot: Start Configuration with configuration file '/etc/sdm/auto-1piboot.conf'
Dec 04 18:57:22 sdminkboard.local root[470]: sdm FirstBoot: Configuration with configuration file '/etc/sdm/auto-1piboot.conf' Completed
Dec 04 18:57:23 sdminkboard.local root[474]: sdm FirstBoot: Set WiFi Country to 'US'
Dec 04 18:57:38 sdminkboard.local root[738]: sdm FirstBoot: Waiting 10/120 seconds for system time synchronization
Dec 04 18:57:49 sdminkboard.local root[813]: sdm FirstBoot: Waiting 20/120 seconds for system time synchronization
Feb 25 11:10:35 sdminkboard.local root[838]: sdm FirstBoot: System time synchronization achieved
Feb 25 11:10:35 sdminkboard.local root[844]: sdm FirstBoot: Run Generated FirstBoot Script '/etc/sdm/0piboot/010-configure-keyboard.sh'
Feb 25 11:10:35 sdminkboard.local root[848]: sdm FirstBoot: Configure keyboard to 'us'
Feb 25 11:10:47 sdminkboard.local root[1140]: sdm FirstBoot: Run Generated FirstBoot Script '/etc/sdm/0piboot/060-runatboot.sh'
Feb 25 11:10:47 sdminkboard.local root[1144]: sdm FirstBoot: Run runatboot scripts
Feb 25 11:10:47 sdminkboard.local root[1145]: sdm FirstBoot: Run runatboot script 'on_first_boot.sh': /etc/sdm/assets/runatboot/on_first_boot.sh  >/home/inku/firstboot.log 2>&1
Feb 25 11:10:51 sdminkboard.local root[1214]: sdm FirstBoot: Skip Custom FirstBoot Scripts
Feb 25 11:10:52 sdminkboard.local root[1235]: sdm FirstBoot: Regenerate SSH Host Keys
Feb 25 11:11:14 sdminkboard.local root[1266]: sdm FirstBoot: set boot behaviour to B1
Feb 25 11:11:16 sdminkboard.local root[1302]: sdm FirstBoot: Re-enable getty@tty1
Feb 25 11:11:18 sdminkboard.local root[1322]: sdm FirstBoot: System will restart in 20 seconds
Feb 25 11:11:39 sdminkboard.local root[1344]: sdm FirstBoot: System restarting now
Feb 25 11:11:39 sdminkboard.local systemd[1]: Stopped sdm FirstBoot System Configuration.
```

Also getting good logs in the output
```
inku@sdminkboard:~ $ cat /home/inku/firstboot.log
Resetting bluetooth to factory settings
Synchronizing state of bluetooth.service with SysV service script with /lib/systemd/systemd-sysv-install.
Executing: /lib/systemd/systemd-sysv-install disable bluetooth
Removed /etc/systemd/system/bluetooth.target.wants/bluetooth.service.
Removed /etc/systemd/system/dbus-org.bluez.service.
[Unit]\nDescription=Poll inkserver periodically\n\n[Service]\nExecStart=usr/bin/python3 home/inku/code/poll_inkserver.py\nRestart=always\n\n[Install]\nWantedBy=multi-user.target
Failed to enable unit: File poll_inkserver.service: Bad message
Failed to start poll_inkserver.service: Unit poll_inkserver.service failed to load properly: Bad message.
See system logs and 'systemctl status poll_inkserver.service' for details.
```

Question for you - to speed up iteration speed for fixing my first boot script, can i just --explore the image and then edit the file at `$SDMPT/etc/sdm/assets/runatboot` ?

gitbls, 2024-02-25:
Excellent! 

Yes, you can edit the file in /etc/sdm/assets and reburn. Just keep everything straight b/c it's awfully easy to make the change there and neglect to carry it back to your "master source" and other similar oopsies.

Feel free to close this. I typically go through and close issues every few weeks if they're ready to be closed or there's been no action for some period of time.

garshythoel, 2024-02-25:
Awesome, thanks for the prompt support, incredibly helpful tool! especially since I'm not even sure if there's any other actively maintained alternative

closed 2024-02-25 by garshythoel