gitbls/sdm #192: Exit codes when "Image cannot be shrunk any further"

justin-robinson, 2024-03-06:
If you attempt to shrink an image that can't be shrunk anymore, the command exits with status 1. This was unexpected and requires us to suppress that error code so it doesn't kill our script. I'd argue this isn't an error since the result is what the user wanted, an image that's shrunk as much as possible. What are you thoughts?

```shell
# sudo sdm --shrink my.img
* Start shrink on IMG 'my.img'
> Check the file system
rootfs: 52111/215568 files (0.2% non-contiguous), 571790/861981 blocks
resize2fs 1.46.5 (30-Dec-2021)
% Image cannot be shrunk any further

# echo $?
1
```

## We have to do this as a workaroung
```shell
# shrink it (always return exit code 0 since sdm sets it to 1 if the image already is the minimum size)
sudo sdm --shrink $1 || true
```

gitbls, 2024-03-06:
Ugh. That one is easy. Please continue posting busted exit statusi, will fix them in the next release.

gitbls, 2024-03-22:
Fixed in V11.6. Please test/verify. Thx

gitbls, 2024-04-22:
Closing due to no activity. Please re-open or create a new issue if necessary.

closed 2024-04-22 by gitbls