gitbls/sdm #193: EZsdmInstaller does not include the rsync package

karoiz, 2024-03-29:
The SDM installer does not install the rsync package. 

If a copy plugin is used, the reason for failure is the lack of rsync. Example:
  --plugin copydir:"from=/home/user/repo|to=/home" 

Currently:

`pkgs="binfmt-support gdisk parted qemu-user-static systemd-container"`

Proposal:

`pkgs="binfmt-support gdisk parted qemu-user-static systemd-container rsync"`

gitbls, 2024-03-29:
Easy fix, but I'm curious what distro you're using that doesn't include rsync by default?

karoiz, 2024-03-30:
Plain Debian without a graphical (X) environment

```
user@build-debian:~$ cat /etc/debian_version
12.5
user@build-debian:~$ uname -a
Linux build-debian 6.1.0-18-amd64 #1 SMP PREEMPT_DYNAMIC Debian 6.1.76-1 (2024-02-01) x86_64 GNU/Linux
user@build-debian:~$
```

gitbls, 2024-03-30:
Thx. Fix will be in next release.

gitbls, 2024-04-15:
Fixed in V11.7.

closed 2024-04-16 by karoiz