gitbls/sdm #195: still prompting for keyboard and user at first boot

tim-oe, 2024-04-12:
just installed the latest version
using 2024-03-15-raspios-bookworm-arm64-lite.img
building image on same version
launched on a pi 3b+

build command
`sudo sdm --customize \
--plugin user:"adduser=tcronin|password=$TEC_PWD" \
--plugin user:"adduser=ansible|password=$ANS_PWD" \
--plugin user:"deluser=pi" \
--plugin L10n:host \
--regen-ssh-host-keys \
--plugin network:"netman=dhcpcd|wifissid=tec-wan|wifipassword=$WIFI_PWD|wificountry=US"  \
--restart img/bookworm.img`

burn command
`sudo sdm --burn /dev/sdc--hostname tec-test --expand-root`

when i launch the image i get a prompt to select keyboard
then to rename a user and set pwd.

the users created with the command do work

i've run it twice same result

am i missing some command parameter?

gitbls, 2024-04-12:
Adding `--plugin disables:piwiz` to your command line should fix that.

tim-oe, 2024-04-13:
yup that fixed it!
great utility making setting up a kube cluster a breeze.

closed 2024-04-13 by tim-oe