gitbls/sdm #196: copyfile source file collision

tim-oe, 2024-04-13:
just installed the latest version
using 2024-03-15-raspios-bookworm-arm64-lite.img
building image on same version
launched on a pi 3b+

tried with both the following:

separate plugins
`--plugin copyfile:"from=/mnt/clones/data/home/tcronin/.ssh/authorized_keys|to=/home/tcronin/.ssh|runphase=postinstall|chown=tcronin:tcronin|chmod=600" \
--plugin copyfile:"from=/mnt/clones/data/home/ansible/.ssh/authorized_keys|to=/home/ansible/.ssh|runphase=postinstall|chown=ansible:ansible|chmod=600" \`

and filelist
`--plugin copyfile:"filelist=./filelist" \`
filelist content
`from=/mnt/clones/data/home/ansible/.ssh/authorized_keys|to=/home/ansible/.ssh|chown=ansible:ansible|chmod=600|mkdirif|runphase=postinstall
from=/mnt/clones/data/home/tcronin/.ssh/authorized_keys|to=/home/tcronin/.ssh|chown=tcronin:tcronin|chmod=600|mkdirif|runphase=postinstall
`
both instances would copy the first from to each to target
even verified  reordering the files
also the filelist command required adding the mkdirif even with a preceding mkdir plugin creating the target dir.

full customize command:
`sudo sdm --customize \
--plugin user:"adduser=tcronin|password=$TEC_PWD" \
--plugin mkdir:"dir=/home/tcronin/.ssh|chown=tcronin:tcronin|chmod=700" \
--plugin copyfile:"from=/mnt/clones/data/home/tcronin/.ssh/authorized_keys|to=/home/tcronin/.ssh|runphase=postinstall|chown=tcronin:tcronin|chmod=600" \
--plugin user:"adduser=ansible|password=$ANS_PWD" \
--plugin mkdir:"dir=/home/ansible/.ssh|chown=ansible:ansible|chmod=700" \
--plugin copyfile:"from=/mnt/clones/data/home/ansible/.ssh/authorized_keys|to=/home/ansible/.ssh|runphase=postinstall|chown=ansible:ansible|chmod=600" \
--plugin user:"deluser=pi" \
--plugin L10n:"keymap=us|locale=en_US.UTF-8|timezone=America/Chicago" \
--regen-ssh-host-keys \
--plugin network:"netman=nm|wifissid=tec-wan|wifipassword=$WIFI_PWD|wificountry=US" \
--restart img/bookworm.img`

gitbls, 2024-04-13:
Investigating...

gitbls, 2024-04-13:
Please provide the redacted-as-necessary /etc/sdm/history from the IMG or console output from the customize command. Thx

tim-oe, 2024-04-13:
I think i found the issue from command output:
first copy:
`> Run Plugin 'copyfile' (/usr/local/sdm/plugins/copyfile) Phase 1 with arguments:
  'from=/mnt/clones/data/home/tcronin/.ssh/authorized_keys|to=/home/tcronin/.ssh|runphase=postinst
  all|chown=tcronin:tcronin|chmod=600'
* Plugin copyfile: Start Phase 1
> Plugin copyfile: Copy 'authorized_keys' from /etc/sdm/assets/copyfile to '/home/tcronin/.ssh'
> Plugin copyfile: Set owner 'tcronin:tcronin' on file '/home/tcronin/.ssh/authorized_keys'
> Plugin copyfile: Set mode '600' on file '/home/tcronin/.ssh/authorized_keys'
* Plugin copyfile: Complete Phase 1
`

second copy:
`> Run Plugin 'copyfile' (/usr/local/sdm/plugins/copyfile) Phase 1 with arguments:
  'from=/mnt/clones/data/home/tcronin/.ssh/authorized_keys|to=/home/tcronin/.ssh|runphase=postinst
  all|chown=tcronin:tcronin|chmod=600'
* Plugin copyfile: Start Phase 1
> Plugin copyfile: Copy 'authorized_keys' from /etc/sdm/assets/copyfile to '/home/tcronin/.ssh'
> Plugin copyfile: Set owner 'tcronin:tcronin' on file '/home/tcronin/.ssh/authorized_keys'
> Plugin copyfile: Set mode '600' on file '/home/tcronin/.ssh/authorized_keys'
* Plugin copyfile: Complete Phase 1
* 
`

key being:
Copy 'authorized_keys' from /etc/sdm/assets/copyfile 

below is the full redacted output:

tcronin@tec-pi-mgr:~/src/piImage $ src/sdm/image.sh
* Mount IMG 'img/bookworm.img'
mount: /dev/loop0 mounted on /mnt/sdm.
mount: /dev/loop1 mounted on /mnt/sdm/boot/firmware.
* Start Configuration
> Command Line: /usr/local/bin/sdm --customize --plugin user:adduser=tcronin|password= --plugin
  mkdir:dir=/home/tcronin/.ssh|chown=tcronin:tcronin|chmod=700 --plugin
  copyfile:from=/mnt/clones/data/home/tcronin/.ssh/authorized_keys|to=/home/tcronin/.ssh|runphase=
  postinstall|chown=tcronin:tcronin|chmod=600 --plugin user:adduser=ansible|password= --plugin
  mkdir:dir=/home/ansible/.ssh|chown=ansible:ansible|chmod=700 --plugin
  copyfile:from=/mnt/clones/data/home/ansible/.ssh/authorized_keys|to=/home/ansible/.ssh|runphase=
  postinstall|chown=ansible:ansible|chmod=600 --plugin user:deluser=pi --plugin
  L10n:keymap=us|locale=en_US.UTF-8|timezone=America/Chicago --plugin disables:piwiz
  --regen-ssh-host-keys --plugin network:netman=nm|wifissid=tec-wan|wifipassword=|wificountry=US
  --restart img/bookworm.img
* Host Information
   Hostname: tec-pi-mgr
   uname: Linux tec-pi-mgr 6.6.20+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.6.20-1+rpt1 (2024-03-07)
  aarch64 GNU/Linux
   os-release Name: Debian GNU/Linux 12 (bookworm)
   Version: 12 (bookworm)
   sdm Version: V11.6
* IMG Information
   Name: img/bookworm.img
   Date: 2024-03-15
   RasPiOS Version: 12
   RasPiOS Architecture: 64-bit aarch64
   os-release Version: 12 (bookworm)
* Plugins selected:
   * user
       Args: adduser=tcronin|password=
   * mkdir
       Args: dir=/home/tcronin/.ssh|chown=tcronin:tcronin|chmod=700
   * copyfile
       Args: from=/mnt/clones/data/home/tcronin/.ssh/authorized_keys|to=/home/tcronin/.ssh|runphase=postinstall|chown=tcronin:tcronin|chmod=600
   * user
       Args: adduser=ansible|password=
   * mkdir
       Args: dir=/home/ansible/.ssh|chown=ansible:ansible|chmod=700
   * copyfile
       Args: from=/mnt/clones/data/home/ansible/.ssh/authorized_keys|to=/home/ansible/.ssh|runphase=postinstall|chown=ansible:ansible|chmod=600
[... 218 lines cut ...]
* Plugin user: Complete Phase post-install
> Run Plugin 'L10n' (/usr/local/sdm/plugins/L10n) Phase post-install with arguments:
  'keymap=us|locale=en_US.UTF-8|timezone=America/Chicago'
> Run Plugin 'disables' (/usr/local/sdm/plugins/disables) Phase post-install with arguments:
  'piwiz'
* Plugin disables: Start Phase post-install
> Plugin disables: Disable running piwiz and userconfig on first system boot
* Plugin disables: Complete Phase post-install
> Run Plugin 'network' (/usr/local/sdm/plugins/network) Phase post-install with arguments:
  'netman=nm|wifissid=tec-wan|wifipassword=|wificountry=US'
* Plugin network: Start Phase post-install
> Plugin network: Keys/values found:
   netman: network-manager
   wifissid: tec-wan
   wifipassword:
   wificountry: US
> Plugin network: Set WiFi Country 'US' for sdm FirstBoot
> Plugin network: Enable SSH service via /boot/ssh
* Plugin network: Complete Phase post-install
> Run graphics post-install configuration
> Plugin sdm-phase1: No known Display Manager found
> Plugin sdm-phase1: Set RasPiOS to B1:'console no login' for sdm FirstBoot
> IMG 'img/bookworm.img' has 196848 1K-blocks (0.2GB, 0.2GiB) free at end of image customization
* Enter Shell Command Prompt
  'exit' to exit back to host system

gitbls, 2024-04-13:
Thanks! Will correct this for next release.

gitbls, 2024-04-15:
This is fixed in v11.7. Please test/verify. Thx

tim-oe, 2024-04-17:
works! thanks for the quick turn around.

closed 2024-04-17 by tim-oe

gitbls, 2024-04-17:
Perfect. And thanks for finding this rather nasty bug. Using the same source filename in two different directories never occurred to me. 😵