gitbls/sdm #197: Error on customizing and exploring image

debemdeboas, 2024-04-15:
I've started to get an error on a new install of Ubuntu 22.04 LTS under WSL.
The error happens whenever trying to --customize or --explore an image. I can, however, --mount it.

Using --explore:
[image]

Using --customize (trimmed for size):
[image]

I was running SDM just fine under another Ubuntu 22 instance last week, so I'm not 100% sure what the problem is. I've tried reinstalling SDM as well.

debemdeboas, 2024-04-15:
This is the output of `mini-nspawn.sh` (#178):

```console
$ sudo ./mini-nspawn.sh images/2024-03-15-raspios-bookworm-arm64-lite.img
* Human-readable partition list
Model:  (file)
Disk [REDACTED]/images/2024-03-15-raspios-bookworm-arm64-lite.img: 5406720s
Sector size (logical/physical): 512B/512B
Partition Table: msdos
Disk Flags:

Number  Start     End       Size      Type     File system  Flags
 1      8192s     1056767s  1048576s  primary  fat32        lba
 2      1056768s  5406719s  4349952s  primary  ext4

* Machine-readable partition list
BYT;
[REDACTED]/images/2024-03-15-raspios-bookworm-arm64-lite.img:5406720s:file:512:512:msdos::;
1:8192s:1056767s:1048576s:fat32::lba;
2:1056768s:5406719s:4349952s:ext4::;
mount: /dev/loop0p2 mounted on /mnt/sdm.
mount: /dev/loop0p1 mounted on /mnt/sdm/boot/firmware.
Spawning container sdm on /mnt/sdm.
Press ^] three times within 1s to kill container.
execv(/bin/bash, /bin/sh) failed: No such file or directory
Container sdm failed with error code 1.
```

gitbls, 2024-04-15:
Huh. Works fine on my Debian WSL system.

* Can you verify that qemu-user-static and binfmt-support are installed on your Ubuntu system?
  * `apt search qemu-user-static` and `apt search binfmt-support`
* What is the result if you repeat the sdm command and add `--chroot` to the command?

debemdeboas, 2024-04-15:
It was working fine here as well until today!

Here are the outputs:

```console
$ apt search qemu-user-static
Sorting... Done
Full Text Search... Done
qemu/jammy-updates,now 1:6.2+dfsg-2ubuntu6.18 amd64 [installed]
  fast processor emulator, dummy package

qemu-user/jammy-updates,now 1:6.2+dfsg-2ubuntu6.18 amd64 [installed]
  QEMU user mode emulation binaries

qemu-user-static/jammy-updates,now 1:6.2+dfsg-2ubuntu6.18 amd64 [installed]
  QEMU user mode emulation binaries (static version)

$ apt search binfmt-support
Sorting... Done
Full Text Search... Done
binfmt-support/jammy,now 2.2.1-2 amd64 [installed]
  Support for extra binary formats
```

When using --chroot, the result is the same but the message is a little different:

```console
$ sudo sdm --chroot --explore images/2024-03-15-raspios-bookworm-arm64-lite.img
* Mount IMG 'images/2024-03-15-raspios-bookworm-arm64-lite.img'
mount: /dev/loop0 mounted on /mnt/sdm.
mount: /dev/loop1 mounted on /mnt/sdm/boot/firmware.
% sdm will use qemu 'aarch64'
% sdm will use chroot per --chroot on this 64-bit x86-64 host
* Enter IMG 'images/2024-03-15-raspios-bookworm-arm64-lite.img'
chroot: failed to run command ‘/bin/bash’: No such file or directory
umount: /mnt/sdm/boot/firmware unmounted
umount: /mnt/sdm unmounted
```

gitbls, 2024-04-15:
Huh. What specific Ubunut 22.04 are you running? 22.04.3 or 22.04.6  (these are the 2 versions I see in the MSFT store)

debemdeboas, 2024-04-15:
I'm using Ubuntu 22.04.4 LTS but I also tried this under a clean Debian WSL install and had the same issue.

After a clean Windows restart everything seems to be working as normal. Maybe this was just another case of Windows being Windows? Not sure. In any case, I'll close this issue as it doesn't seem to be a problem with sdm.

Thank you for your time and for developing and maintaining this amazing tool!

closed 2024-04-15 by debemdeboas

gitbls, 2024-04-15:
Great that a reboot cleared it up. Windows being Windows, I should have suggested that 🤔