gitbls/sdm #198: wifi config isn't being supplied

tim-oe, 2024-04-18:
running v11.7
using 2024-03-15-raspios-bookworm-arm64-lite.img
building image on same version
launched on a pi 3b+

following customize command:

`sudo sdm --customize \
--plugin user:"adduser=tcronin|password=$TEC_PWD" \
--plugin mkdir:"dir=/home/tcronin/.ssh|chown=tcronin:tcronin|chmod=700" \
--plugin copyfile:"from=/mnt/clones/data/home/tcronin/.ssh/authorized_keys|to=/home/tcronin/.ssh|runphase=postinstall|chown=tcronin:tcronin|chmod=600" \
--plugin user:"adduser=ansible|password=$ANS_PWD" \
--plugin mkdir:"dir=/home/ansible/.ssh|chown=ansible:ansible|chmod=700" \
--plugin copyfile:"from=/mnt/clones/data/home/ansible/.ssh/authorized_keys|to=/home/ansible/.ssh|runphase=postinstall|chown=ansible:ansible|chmod=600" \
--plugin user:"deluser=pi" \
--plugin L10n:"keymap=us|locale=en_US.UTF-8|timezone=America/Chicago" \
--plugin disables:piwiz \
--regen-ssh-host-keys \
--plugin network:"netman=nm|wifissid=tec-wan|wifipassword=$WIFI_PWD|wificountry=US" \
--restart img/bookworm.img
`
the env vars are working as i see the correct values in the output

i see the following in the output (redacted)
lugin network: Create NetworkManager WiFi configuration
  with SSID: 'tec-wan' Password: 'x xxx xxx xxx' WiFi Country: 'US'

but looking in the image post customize 
/etc/NetworkManager/system-connections folder is empty
(that is where i found it with manual setup)

and no network config in
/etc/wpa_supplicant/wpa_supplicant.conf

gitbls, 2024-04-18:
When the network plugin was built it was done in such a way that it could be run on bullseye systems to customize bookworm IMGs. NetworkManager in bullseye didn't have the `--offline` feature, which enables one to build a .nmconnection without perturbing the running system.

In order to make this work, the network plugin builds the script /etc/sdm/xpiboot/005-nm-config.sh. When the system boots the first time and sdm-firstboot runs, it immediately runs any scripts found in /etc/sdm/xpiboot, which causes the wifi configuration to be set.

This methodology could be improved, so I've added it to my "have a look at this" list.

Can you confirm that /etc/sdm/xpiboot/005-nm-config.sh is configured correctly per your customize command line?

tim-oe, 2024-04-18:
it's there but there might be an issue with there being spaces in the pwd
script content (redacted):
#!/bin/bash
logger "sdm FirstBoot: Configure Network Manager WiFi connection 'tec-wan' with SSID tec-wan"
nmcli c add type wifi con-name tec-wan ifname wlan0 ssid tec-wan
nmcli c modify tec-wan wifi-sec.key-mgmt wpa-psk wifi-sec.psk x xxx xxx xxx xx

tim-oe, 2024-04-18:
yup just built the image and placed ' around the pwd value

the image booted with wifi, and there's nm script

host: etc/NetworkManager/system-connections $ ls
tec-wan.nmconnection

gitbls, 2024-04-18:
Ugh. Will fix that in the next couple of days. If it's blocking you, you should be able to edit /usr/local/sdm/plugins/network and change
```
nmcli c modify $wificname wifi-sec.key-mgmt wpa-psk wifi-sec.psk $wifipassword
```
to
```
nmcli c modify $wificname wifi-sec.key-mgmt wpa-psk wifi-sec.psk '$wifipassword'
```
I've verified that this changes the .nmconnection correctly, but didn't test whether NetworkManager correctly handles a password with spaces in it.

Presumably you have?

gitbls, 2024-04-18:
> yup just built the image, edited 005-nm-config.sh, placed ' around the pwd value
> 
> the image booted with wifi, and there's nm script
> 
> host: etc/NetworkManager/system-connections $ ls tec-wan.nmconnection

Perfect.

gitbls, 2024-04-22:
In V11.8 sdm will not defer the configuration to FirstBoot if NetworkManager is new enough (bookworm version or later), and the WiFi password setting has quotes around it.

gitbls, 2024-04-24:
Configuration code overhauled:
* Password quoting
* Don't defer nm configuration to FirstBoot if nmcli is new enough (Bookworm or later)

Please test/verify.

tim-oe, 2024-04-25:
no longer a script at /etc/sdm/xpiboot
but it got wireless ip.

thanks!

closed 2024-04-25 by tim-oe