gitbls/sdm #199: Change default aes cipher from aes-cbc-essiv:sha256 to aes-xts-plain64

jollycar, 2024-04-18:
In V11.7 you changed the default luks encryption cipher from `xchacha20,aes-adiantum-plain64` to  `aes-cbc-essiv:sha256`. This has better performance on the raspberry pi5, but I did some investigation and found that `aes-xts-plain64` is even more performant than `aes-cbc-essiv:sha256` (in write performance) and has some other advantages:

- AES-XTS addressing CBC mode vulnerabilities.
- Current standard in disk encryption systems.
- Improved compatibility with SSD drives.

 This is a proposal to upgrade the aes encryption to `aes-xts-plain64`

I tested this change (and confirmed working) in my fork of sdm: https://github.com/jollycar/sdm
The only change you need is `local cipher="aes-cbc-essiv:sha256"` `local cipher="aes-xts-plain64"` in `sdmcryptfs`

gitbls, 2024-04-18:
I just did a quick test and am not convinced. DEcryption is much more important than ENcryption, since decryption is done much more frequently than encryption, for the most part.

If you have data that proves otherwise, or can share other light on why this is worthwhile, please share.
```
pw~# cryptsetup benchmark -c aes-xts-plain64 
# Tests are approximate using memory only (no storage IO).
# Algorithm |       Key |      Encryption |      Decryption
    aes-xts        256b      1737.4 MiB/s      1824.9 MiB/s

pw~# cryptsetup benchmark -c aes-cbc-essiv:sha256 
# Tests are approximate using memory only (no storage IO).
# Algorithm |       Key |      Encryption |      Decryption
    aes-cbc        256b       918.4 MiB/s      1881.3 MiB/s
```

gitbls, 2024-04-26:
I'm adding a way to make the accepted cryptos include aes-* which addresses your request and hopefully any future ones that are found 🤣

gitbls, 2024-05-05:
This is changed in V12.0. Please test. Thx!

gitbls, 2024-05-15:
Closing due to no activity. @jollycar please re-open if needed. Thx

closed 2024-05-15 by gitbls