gitbls/sdm #200: ELI5 add authorized_keys file for ssh.

synaption, 2024-04-21:
Can someone please explain it like I'm 5.  I'm having a hard time getting the authorized_keys into my image.  I have been able to do it with --explore, but that's too much work.  I would prefer to use copyfile or a plugin of some sort.  

I use this command to make the image and it exits with error status 1.  

```
pi@CC:~/CCpi/cm4_flash $ sudo sdm --customize --plugin user:"setpassword=pi|password=raspberry" --plugin mkdir:"dir=/home/pi/.ssh|chown=pi:users|chmod=700" --plugin copyfile:"from=/home/pi/CCpi/cm4_flash/authorized_keys|to=/home/pi/.ssh|chown=pi:users|chmod=600|runphase=postinstall" --plugin L10n:host --plugin disables:piwiz --plugin raspiconfig:serial --regen-ssh-host-keys --restart 2024-03-15-raspios-bookworm-arm64-lite.img

* Mount IMG '2024-03-15-raspios-bookworm-arm64-lite.img'
mount: /dev/loop1 mounted on /mnt/sdm.
mount: /dev/loop2 mounted on /mnt/sdm/boot/firmware.
** IMG '2024-03-15-raspios-bookworm-arm64-lite.img' has already been customized. Redo? [y/n]? y
* Start Configuration
> Command Line: /usr/local/bin/sdm --customize --plugin user:setpassword=pi|password=raspberry 
  --plugin mkdir:dir=/home/pi/.ssh|chown=pi:users|chmod=700 --plugin 
  copyfile:from=/home/pi/CCpi/cm4_flash/authorized_keys|to=/home/pi/.ssh|chown=pi:users|chmod=600|

  runphase=postinstall --plugin L10n:host --plugin disables:piwiz --plugin raspiconfig:serial 
  --regen-ssh-host-keys --restart 2024-03-15-raspios-bookworm-arm64-lite.img
* Host Information
   Hostname: CC
   uname: Linux CC 6.6.20+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.6.20-1+rpt1 (2024-03-07) aarch64 
  GNU/Linux
   os-release Name: Debian GNU/Linux 12 (bookworm)
   Version: 12 (bookworm)
   sdm Version: V11.7
* IMG Information
   Name: 2024-03-15-raspios-bookworm-arm64-lite.img
   Date: 2024-03-15
   RasPiOS Version: 12
   RasPiOS Architecture: 64-bit aarch64
   os-release Version: 12 (bookworm)
* Plugins selected:
   * user
       Args: setpassword=pi|password=raspberry
   * mkdir
       Args: dir=/home/pi/.ssh|chown=pi:users|chmod=700
   * copyfile
       Args: from=/home/pi/CCpi/cm4_flash/authorized_keys|to=/home/pi/.ssh|chown=pi:users|chmod=600|runphase=postinstall
   * L10n
       Args: host
   * disables
       Args: piwiz
   * raspiconfig
       Args: serial
> IMG '2024-03-15-raspios-bookworm-arm64-lite.img' has 199544 1K-blocks (0.2GB, 0.2GiB) free 
> Copy sdm to /usr/local/sdm in the IMG
* Start Phase 0 image customization
> Run Plugins Phase '0'
> Run Plugin 'user' (/mnt/sdm/usr/local/sdm/plugins/user) Phase 0 with arguments: 
  'setpassword=pi|password=raspberry'
* Plugin user: Start Phase 0
> Plugin user: Keys/values found:
   setpassword: pi
   password: raspberry
* Plugin user: Complete Phase 0
> Run Plugin 'mkdir' (/mnt/sdm/usr/local/sdm/plugins/mkdir) Phase 0 with arguments: 
  'dir=/home/pi/.ssh|chown=pi:users|chmod=700'
* Plugin mkdir: Start Phase 0
> Plugin mkdir: Keys/values found:
   dir: /home/pi/.ssh
   chown: pi:users
   chmod: 700
> Plugin mkdir: Create directory '/home/pi/.ssh'
* Plugin mkdir: Complete Phase 0
> Run Plugin 'copyfile' (/mnt/sdm/usr/local/sdm/plugins/copyfile) Phase 0 with arguments: 
  'from=/home/pi/CCpi/cm4_flash/authorized_keys|to=/home/pi/.ssh|chown=pi:users|chmod=600|runphase
  =postinstall'
* Plugin copyfile: Start Phase 0
> Plugin copyfile: Keys/values found:
   from: /home/pi/CCpi/cm4_flash/authorized_keys
   to: /home/pi/.ssh
   chown: pi:users
   chmod: 600
   runphase: postinstall
? Plugin copyfile: Copy /home/pi/CCpi/cm4_flash/authorized_keys to /home/pi/.ssh already 
  specified
? Plugin 'copyfile' exited with failure status '1'
umount: /mnt/sdm/boot/firmware unmounted
umount: /mnt/sdm unmounted
```

PS I'd really prefer to use relative paths so I can distribute the script that makes the image and have it be correct no mater who is using the script or where they put it.

gitbls, 2024-04-21:
First, to your problem. I note that you're redoing the customization of an IMG, instead of starting with a freshly, uncustomized IMG. Normally not an issue,  but a _feature_ in copyfile in V11.7 is that redoing a customize and re-using the `copyfile` plugin results in copyfile saying "you've already included this file", which is exactly what's happening to you.

So, the immediate "fix" is to always start with a fresh IMG when doing a customize. 

The copyfile plugin got a minor overhaul in V11.7 and resulted in this problem. I'll correct this for the next release.

As to your PS about using relative paths, two points:
* Although the documentation says that full paths must be used, it's not factually correct. Using full paths helps reduce user errors, but it's not a restriction in sdm
* That said, using relative paths can result in user error type problems, and the best/safest way in my opinion to use relative paths is to establish a variable or envar that has the current path, and then use that when referencing relative paths.:
```
mydir=$PWD
sudo sdm --plugin copyfile:"from=$mydir/.ssh/authorized_keys" ...
   instead of / vs 
sudo sdm --plugin copyfile:"from=.ssh/authorized_keys" ...
```
Of course there's a large degree of personal preference involved, so you may decide that the above isn't necessary.

synaption, 2024-04-23:
That did the trick, thanks.  neat tool.  I wish I know about it sooner.

gitbls, 2024-04-23:
> That did the trick, thanks. neat tool. I wish I know about it sooner.

I think it's pretty cool too, which is why I keep working on it.

Tell all your friends about it! A study I just made up found that sdm Pi users are happier than non-sdm Pi users 🤣

gitbls, 2024-04-24:
This is fixed in V11.8

closed 2024-04-24 by synaption