gitbls/sdm #201: serial plugin hot to disable shell console

tim-oe, 2024-04-22:
i'm trying to enable serial interface  but want the shell console to be disabled.
the raspiconfig only has the single flag and stumbled on the bug associated with it

trying to use the serial plugin but it seems to not disable the shell console I'm passing the following

--plugin serial

with no arguments

i had to run raspi-config to disable the console shell.

gitbls, 2024-04-22:
Will have a look. The plugin with no arguments _should_ disable the console on the serial port.

gitbls, 2024-04-22:
Yep, nice buglet! Will be fixed in V11.8.

tim-oe, 2024-04-22:
thanka

gitbls, 2024-04-24:
Fixed in V11.8. Please verify.

tim-oe, 2024-04-28:
so finally was able to look at this.
i'm seeing the following in boot/firmware/config.txt
enable_uart=1
i'm seeing /dev/ttyS0 usually it shows as /dev/ttyAMA0 going through raspi-config
i cannot connect over serial and lsof is showing me nothing.

it seems to be working other than the different device name.

gitbls, 2024-04-28:
What Pi model are you using this on?

tim-oe, 2024-04-29:
verifying it on: Raspberry Pi Zero 2 W Rev 1.0

gitbls, 2024-04-29:
Thx. Will check it out.

gitbls, 2024-04-29:
I think ttyAMA* is not created by raspi-config any more, in favor of ttyS0

That said, I mis-read raspi-config a bit, so was not disabling the uart, hence enable_uart=1

This will be fixed in next release, and adding a new argument `enableuart`, in case you want the shell disabled, and the uart enabled. 🤷‍♂️

tim-oe, 2024-04-30:
I'm running the latest version of the OS.
https://downloads.raspberrypi.com/raspios_lite_arm64/images/raspios_lite_arm64-2024-03-15/

now i don't know all the ins and outs but in manually configuring from raspi-config i get the following:

serial disabled
console disabled

tcronin@tec-zero:~ $ ls -l /dev/ttyS0
crw-rw---- 1 root dialout 4, 64 Apr 29 20:50 /dev/ttyS0
tcronin@tec-zero:~ $ ls -l /dev/ttyAMA0
ls: cannot access '/dev/ttyAMA0': No such file or directory
tcronin@tec-zero:~ $ ls -l /dev/serial0
ls: cannot access '/dev/serial0': No such file or directory

/boot/firmware/config.text
enable_uart=0

tcronin@tec-zero:~ $ sudo lsof /dev/ttyS0
tcronin@tec-zero:~ $

serial enabled
console disabled

tcronin@tec-zero:~ $ ls -l /dev/ttyS0
crw-rw---- 1 root dialout 4, 64 Apr 29 20:56 /dev/ttyS0
tcronin@tec-zero:~ $ ls -l /dev/ttyAMA0
crw-rw---- 1 root dialout 204, 64 Apr 29 20:56 /dev/ttyAMA0
tcronin@tec-zero:~ $ ls -l /dev/serial0
lrwxrwxrwx 1 root root 7 Apr 29 20:56 /dev/serial0 -> ttyAMA0

/boot/firmware/config.text
enable_uart=1

tcronin@tec-zero:~ $ sudo lsof /dev/ttyS0
tcronin@tec-zero:~ $ sudo lsof /dev/ttyAMA0
tcronin@tec-zero:~ $

serial enabled
console enabled

tcronin@tec-zero:~ $ sudo lsof /dev/ttyS0
tcronin@tec-zero:~ $ sudo lsof /dev/ttyAMA0
COMMAND PID USER   FD   TYPE DEVICE SIZE/OFF NODE NAME
agetty  668 root    0u   CHR 204,64      0t0  132 /dev/ttyAMA0
agetty  668 root    1u   CHR 204,64      0t0  132 /dev/ttyAMA0
agetty  668 root    2u   CHR 204,64      0t0  132 /dev/ttyAMA0

gitbls, 2024-05-01:
Thanks. I started with a fresh, unadulterated IMG the same as the one you used on a Pi2W. When it booted:
* No /dev/ttyS0
* No /dev/ttyAMA*
* no UART in config.txt

Used raspi-config to enable and rebooted
* console=serial0,115200 in cmdline.txt
* enable_uart=1 in config.sys
* /dev/serial0 is a link to /dev/ttyS0

Used raspi-config to disable and rebooted
* cmdline.txt and config.txt reverted to no console=serial0 and enable_uart=0

Curiously, no sign of /dev/ttyAMA* during any of these steps. Wonder how you got it?

Then used a lightly customized IMG and burned it with `--plugin serial`
* Serial port was disabled

Next used that same lightly customized IMG and burned with `--plugin serial:enableshell`
* Serial port shell was correctly enabled as with raspi-config
* Again, no sign of /dev/ttyAMA*

So, I believe this will work correctly in the next sdm release, targeting it for later this week. 🤞

tim-oe, 2024-05-02:
most of my knowledge on this  is googling, so not sure why as long as i can use serial gpio.  Currently for a gps  hat for a time server im trying to set up.
some cursory googling gives me this, still not clear on why the different name for the same port
https://unix.stackexchange.com/questions/307390/what-is-the-difference-between-ttys0-ttyusb0-and-ttyama0-in-linux

thanks again for the quick responses

gitbls, 2024-05-05:
I believe this is corrected in V12.0. Please test.Thx!

gitbls, 2024-05-15:
Closing due to no activity. @tim-oe please reopen if needed. Thx

closed 2024-05-15 by gitbls