gitbls/sdm #207: The apt-addrepo plugin seems to ignore the name argument

x7airworker, 2024-05-11:
I am using the following build script:
```bash
sudo sdm --customize \
	--plugin L10n:"keymap=de|locale=de_DE.UTF-8|timezone=Europe/Berlin" \
	--plugin disables:"piwiz|bluetooth|wifi|triggerhappy" \
        --plugin apt-addrepo:"name=kubernetes|repo=deb [signed-by=/etc/apt/trusted.gpg.d/kubernetes-apt-keyring.gpg] https://pkgs.k8s.io/core:/stable:/v1.30/deb/ /|gpgkey=./kubernetes-apt-keyring.gpg" \
        --plugin apt-addrepo:"name=cri-o|repo=deb [signed-by=/etc/apt/trusted.gpg.d/cri-o-apt-keyring.gpg] https://pkgs.k8s.io/addons:/cri-o:/prerelease:/main/deb/ /|gpgkey=./cri-o-apt-keyring.gpg" \
	--plugin apps:"apps=gnupg2,curl,apt-transport-https,crii-o,kubelet,kubeadm,kubectl,helm|name=kubernetes" \
        --plugin system:"swap=disable|service-enable=crio.service" \
        --plugin runscript:"script=./install.sh" \
        --plugin apt-cacher-ng \
        --regen-ssh-host-keys \
        image.img
```

When I try to run it I receive the following error message:
```
> Plugin apt-addrepo: Add repo 'deb [signed-by=/etc/apt/trusted.gpg.d/cri-o-apt-keyring.gpg] 
  https://pkgs.k8s.io/addons:/cri-o:/prerelease:/main/deb/ /' to apt sources
? Plugin apt-addrepo: Repo 'sdm-sourced-repo' already exists; Add argument 'name=somename' to 
  this apps plugin invocation
  NOTE: This plugin is not idempotent; Restart with a fresh uncustomized IMG if this IMG is 
  previously customized
? Plugin 'apt-addrepo' exited with failure status '1'
```

To me it seems that the supplied name is ignored and instead replaced with "sdm-sourced-repo". Am I doing anything wrong or is this a bug?

gitbls, 2024-05-11:
Argh...a bug!  In the file /usr/local/sdm/plugins/apt-addrepo you'll see line 36:
```
        [ "$name" != "" ] && name="sdm-sourced-repo"
```
If you `sudoedit` this file and change `!=` to `==` on that line, it should work much better 🤔

This will be corrected in the next release of sdm, which should release next week.

x7airworker, 2024-05-11:
Thanks for the quick response.
This indeed fixes my issue, thank you very much.

closed 2024-05-11 by x7airworker

gitbls, 2024-05-11:
Thanks for confirming!