gitbls/sdm #208: odd results when using plugin bootconfig with section option

tinker2much, 2024-05-19:
This sdm command: 

```
sudo sdm \
--burn /dev/$DEVICE \
--expand-root \
--host piano \
--plugin copyfile:"from=$MYFILES/cardid|to=/home/pi" \
--plugin runatboot:"user=pi|sudoswitches=-H|script=/home/pi/bin/aptcache_on|output=/home/pi/aptcache_on.log" \
--plugin bootconfig:"section=[pi4]|dtoverlay=pwm-fan|dtparam=fan_temp0=55000,fan_temp0_hyst=5000,fan_temp0_speed=75|dtparam=fan_temp1=62500,fan_temp1_hyst=5000,fan_temp1_speed=128|dtparam=fan_temp2=70000,fan_temp2_hyst=5000,fan_temp2_speed=192|dtparam=fan_temp3=77500,fan_temp3_hyst=5000,fan_temp3_speed=255" \
$IMGDIR/${MYIMAGE}32.img

```
results in an odd config.txt (with one line - the last one - being repeated 4 times, instead of four separate lines):

```
pi@pi44:~/Downloads $ sudo sdm --explore /dev/sdb
* Mount device '/dev/sdb'
mount: (hint) your fstab has been modified, but systemd still uses
       the old version; use 'systemctl daemon-reload' to reload.
mount: /dev/sdb2 mounted on /mnt/sdm.
mount: (hint) your fstab has been modified, but systemd still uses
       the old version; use 'systemctl daemon-reload' to reload.
mount: /dev/sdb1 mounted on /mnt/sdm/boot/firmware.
* Enter Device '/dev/sdb'
root@sdm:/# /home/pi/bin/livelines /boot/firmware/config.txt
dtparam=audio=on
camera_auto_detect=1
display_auto_detect=1
auto_initramfs=1
dtoverlay=vc4-kms-v3d
max_framebuffers=2
disable_fw_kms_setup=1
disable_overscan=1
arm_boost=1
[cm4]
otg_mode=1
[all]
[pi4]
dtoverlay=pwm-fan
dtparam=fan_temp3=77500,fan_temp3_hyst=5000,fan_temp3_speed=255
dtparam=fan_temp3=77500,fan_temp3_hyst=5000,fan_temp3_speed=255
dtparam=fan_temp3=77500,fan_temp3_hyst=5000,fan_temp3_speed=255
dtparam=fan_temp3=77500,fan_temp3_hyst=5000,fan_temp3_speed=255
root@sdm:/# 

```

Here's the history: 

```
2024-05-19 13:17:49 > Flush accumulated burn log messages
2024-05-19 13:13:33 * Start Burn
2024-05-19 13:13:33 > Burn '/home/pi/Downloads/bookworm32.img' (6.5GB, 6.1GiB) to '/dev/sdb'...
2024-05-19 13:13:34 > Burn command line: /usr/local/bin/sdm --burn /dev/sdb --expand-root --host piano --plugin copyfile:from=/home/pi/Documents/mysdmfiles/cardid|to=/home/pi --plugin runatboot:user=pi|sudoswitches=-H|script=/home/pi/bin/aptcache_on|output=/home/pi/aptcache_on.log --plugin bootconfig:section=[pi4]|dtoverlay=pwm-fan|dtparam=fan_temp0=55000,fan_temp0_hyst=5000,fan_temp0_speed=75|dtparam=fan_temp1=62500,fan_temp1_hyst=5000,fan_temp1_speed=128|dtparam=fan_temp2=70000,fan_temp2_hyst=5000,fan_temp2_speed=192|dtparam=fan_temp3=77500,fan_temp3_hyst=5000,fan_temp3_speed=255 /home/pi/Downloads/bookworm32.img
2024-05-19 13:13:34 > dd command: dd if=/home/pi/Downloads/bookworm32.img of=/dev/sdb status=progress bs=16M iflag=direct
2024-05-19 13:17:46 > dd Copy completed
2024-05-19 13:17:47 > Set new disk ID '93d452b7' on '/dev/sdb'
2024-05-19 13:17:48 > Disk ID '93d452b7' set successfully with sfdisk
2024-05-19 13:17:50 * Resize the ext4 root file system
2024-05-19 13:17:50 % (Ignore 'on-line resizing required' message)
2024-05-19 13:17:51 > Set MBR rootfs partition ID '93d452b7' in cmdline.txt (was 1aa4489a)
2024-05-19 13:17:51 > Set MBR partition Disk IDs in fstab
2024-05-19 13:17:51 > Disable unneeded RasPiOS firstboot service and /etc/init.d/resize2fs_once
2024-05-19 13:17:51 > Disable regenerate_ssh_host_keys service; sdm FirstBoot will run it instead
2024-05-19 13:17:51 > Set hostname 'piano'
2024-05-19 13:17:51 > First System Boot Custom Boot Scripts disabled
2024-05-19 13:17:51 > First System Boot automatic restart enabled with a 5 second wait
2024-05-19 13:17:51 * Plugins selected:
2024-05-19 13:17:51    * copyfile
[... 66 lines cut ...]
2024-05-19 13:17:52   'section=[pi4]|dtoverlay=pwm-fan|dtparam=fan_temp0=55000,fan_temp0_hyst=5000,fan_temp0_speed=75|
2024-05-19 13:17:52   dtparam=fan_temp1=62500,fan_temp1_hyst=5000,fan_temp1_speed=128|dtparam=fan_temp2=70000,fan_temp
2024-05-19 13:17:52   2_hyst=5000,fan_temp2_speed=192|dtparam=fan_temp3=77500,fan_temp3_hyst=5000,fan_temp3_speed=255'
2024-05-19 13:17:52 * Plugin bootconfig: Start Phase 1
2024-05-19 13:17:52 * Plugin bootconfig: Complete Phase 1
2024-05-19 13:17:52 > Run Plugins Phase 'post-install'
2024-05-19 13:17:52 > Run Plugin 'copyfile' (/usr/local/sdm/plugins/copyfile) Phase post-install with arguments: 
2024-05-19 13:17:53   'from=/home/pi/Documents/mysdmfiles/cardid|to=/home/pi'
2024-05-19 13:17:53 * Plugin copyfile: Start Phase post-install
2024-05-19 13:17:53 * Plugin copyfile: Complete Phase post-install
2024-05-19 13:17:53 > Run Plugin 'runatboot' (/usr/local/sdm/plugins/runatboot) Phase post-install with arguments: 
2024-05-19 13:17:53   'user=pi|sudoswitches=-H|script=/home/pi/bin/aptcache_on|output=/home/pi/aptcache_on.log'
2024-05-19 13:17:53 * Plugin runatboot: Start Phase post-install
2024-05-19 13:17:53 * Plugin runatboot: Complete Phase post-install
2024-05-19 13:17:53 > Run Plugin 'bootconfig' (/usr/local/sdm/plugins/bootconfig) Phase post-install with 
2024-05-19 13:17:53   arguments: 
2024-05-19 13:17:53   'section=[pi4]|dtoverlay=pwm-fan|dtparam=fan_temp0=55000,fan_temp0_hyst=5000,fan_temp0_speed=75|
2024-05-19 13:17:53   dtparam=fan_temp1=62500,fan_temp1_hyst=5000,fan_temp1_speed=128|dtparam=fan_temp2=70000,fan_temp
2024-05-19 13:17:53   2_hyst=5000,fan_temp2_speed=192|dtparam=fan_temp3=77500,fan_temp3_hyst=5000,fan_temp3_speed=255'
2024-05-19 13:17:53 * Plugin bootconfig: Start Phase post-install
2024-05-19 13:17:53 * Plugin bootconfig: Complete Phase post-install
2024-05-19 13:17:53 * Burn Completed
2024-05-19 13:17:53 > Burn elapsed time: 00:00:04:20

```

gitbls, 2024-05-19:
Huh. That's cool...better 4 times than 0? Thx for the example and log...I've replicated it.

gitbls, 2024-05-19:
Ah, yes, this is a shortcoming of the way arguments are parsed. Each argument (e.g., `section`, `dtoverlay`, `dtparam`) can be used exactly _once_ in a plugin invocation. If an argument is used more than once, the last value gets used for all of them. 😵

The workaround is to use multiple bootconfig plugins (blah blah obviously not valid commands 🤣)
```
sudo sdm blah blah blah \
--plugin bootconfig:"section=[pi4]" \
--plugin bootconfig:"dtoverlay=pwm-fan" \
--plugin bootconfig:"dtparam=fan_temp0=55000,fan_temp0_hyst=5000,fan_temp0_speed=75" \
--plugin bootconfig:"dtparam=fan_temp1=62500,fan_temp1_hyst=5000,fan_temp1_speed=128" \
--plugin bootconfig:"dtparam=fan_temp2=70000,fan_temp2_hyst=5000,fan_temp2_speed=192" \
--plugin bootconfig:"dtparam=fan_temp3=77500,fan_temp3_hyst=5000,fan_temp3_speed=255" \
 blah blah blah
```
Plugin arguments are parsed by sdm by running through the list and assigning each found string to a variable with the name of the argument.  In this case, the variable `dtparam` is reassigned 4 times, with the final value being the value of the last one found.

Unfortunately, there's no easy way to fix this in sdm without likely creating collateral damage, but the above (somewhat ugly) workaround does the job.

which results in /boot/firmware/config.txt with:
```
[pi4]
dtoverlay=pwm-fan
dtparam=fan_temp0=55000,fan_temp0_hyst=5000,fan_temp0_speed=75
dtparam=fan_temp1=62500,fan_temp1_hyst=5000,fan_temp1_speed=128
dtparam=fan_temp2=70000,fan_temp2_hyst=5000,fan_temp2_speed=192
dtparam=fan_temp3=77500,fan_temp3_hyst=5000,fan_temp3_speed=255
```

tinker2much, 2024-05-19:
So, would the following work too, since "section" and "dtoverlay" are different? That would be a way to slightly shorten it? 

```
sudo sdm blah blah blah \
--plugin bootconfig:"section=[pi4]|dtoverlay=pwm-fan" \
--plugin bootconfig:"dtparam=fan_temp0=55000,fan_temp0_hyst=5000,fan_temp0_speed=75" \
--plugin bootconfig:"dtparam=fan_temp1=62500,fan_temp1_hyst=5000,fan_temp1_speed=128" \
--plugin bootconfig:"dtparam=fan_temp2=70000,fan_temp2_hyst=5000,fan_temp2_speed=192" \
--plugin bootconfig:"dtparam=fan_temp3=77500,fan_temp3_hyst=5000,fan_temp3_speed=255" \
 blah blah blah
```
I haven't dived into your parsing [ and won't ], so anything I suggest is likely to be annoying and certain to be ignorant. But why not??   ;-)

The following atomic style was where I started, then I realized I could compress things to the 4-line style above:

```
section=[pi4]
dtoverlay=pwm-fan
dtparam=fan_temp0=55000
dtparam=fan_temp0_hyst=5000
dtparam=fan_temp0_speed=75
dtparam=fan_temp1=62500
dtparam=fan_temp1_hyst=5000
dtparam=fan_temp1_speed=128
dtparam=fan_temp2=70000
dtparam=fan_temp2_hyst=5000
dtparam=fan_temp2_speed=192
dtparam=fan_temp3=77500
dtparam=fan_temp3_hyst=5000
dtparam=fan_temp3_speed=255
```
Since the **_pi_** is happy with me combining distinct variables under one dtparam, I assume that the **_pi_** would be happy to have me mash **_all_** of the dtparam variables into one line? 

```
section=[pi4]
dtoverlay=pwm-fan
dtparam=cooling_fan=on,fan_temp0=55000,fan_temp0_hyst=5000,fan_temp0_speed=75,fan_temp1=62500,fan_temp1_hyst=5000,fan_temp1_speed=128,fan_temp2=70000,fan_temp2_hyst=5000,fan_temp2_speed=192,fan_temp3=77500,fan_temp3_hyst=5000,fan_temp3_speed=255

```
So, since there's only one section, and one dtoverlay, and one dtparam, would **_you_** accept: 
```
sudo sdm blah blah blah \
--plugin bootconfig:"section=[pi4]|dtoverlay=pwm-fan|dtparam=cooling_fan=on,fan_temp0=55000,fan_temp0_hyst=5000,fan_temp0_speed=75,fan_temp1=62500,fan_temp1_hyst=5000,fan_temp1_speed=128,fan_temp2=70000,fan_temp2_hyst=5000,fan_temp2_speed=192,fan_temp3=77500,fan_temp3_hyst=5000,fan_temp3_speed=255" \
 blah blah blah
```

gitbls, 2024-05-19:
It appears so:
```
#!/bin/bash
sudo sdm $1 \
--customize \
--apt-options none \
--plugin bootconfig:"section=[pi4]|dtoverlay=pwm-fan|dtparam=cooling_fan=on,fan_temp0=55000,fan_temp0_hyst=5000,fan_temp0_speed=75,fan_temp1=62500,fan_temp1_hyst=5000,fan_temp1_speed=128,fan_temp2=70000,fan_temp2_hyst=5000,fan_temp2_speed=192,fan_temp3=77500,fan_temp3_hyst=5000,fan_temp3_speed=255" \
exit
```
yields a config.txt with:
```
[pi4]
dtoverlay=pwm-fan
dtparam=cooling_fan=on,fan_temp0=55000,fan_temp0_hyst=5000,fan_temp0_speed=75,fan_temp1=62500,fan_temp1_hyst=5000,fan_temp1_speed=128,fan_temp2=70000,fan_temp2_hyst=5000,fan_temp2_speed=192,fan_temp3=77500,fan_temp3_hyst=5000,fan_temp3_speed=255
```

tinker2much, 2024-05-20:
... but such a line doesn't seem to work on the pi. There's apparently a documented 98-character [line length limit](https://www.raspberrypi.com/documentation/computers/config_txt.html#file-format) (anything more is ignored). In my case it seems the long line is rejected, perhaps it breaks at a syntactically incorrect point. In any case the fan just runs and doesn't get PWM'd to different speeds per temperature. So maybe multiple lines can't be avoided...

Testing now, will report what finally works.

tinker2much, 2024-05-20:
Because I may use this block in multiple burns within the script, I made the lines into variables: 
```
FANPARMS420="from=$MYFILES/pwm-fan.dtbo|to=/boot/firmware/overlays|chown=root:root|chmod=755"
FANPARMS421="section=[pi4]|dtoverlay=pwm-fan"
FANPARMS422="dtparam=fan_temp0=55000,fan_temp0_hyst=3000,fan_temp0_speed=75"
FANPARMS423="dtparam=fan_temp1=62500,fan_temp1_hyst=3000,fan_temp1_speed=128"
FANPARMS424="dtparam=fan_temp2=70000,fan_temp2_hyst=3000,fan_temp2_speed=192"
FANPARMS425="dtparam=fan_temp3=77500,fan_temp3_hyst=3000,fan_temp3_speed=255"

sudo sdm \
...
--burn /dev/$DEVICE \
--expand-root \
--plugin   copyfile:"$FANPARMS420" \
--plugin bootconfig:"$FANPARMS421" \
--plugin bootconfig:"$FANPARMS422" \
--plugin bootconfig:"$FANPARMS423" \
--plugin bootconfig:"$FANPARMS424" \
--plugin bootconfig:"$FANPARMS425" \
...
$IMGDIR/${MYIMAGE}32.img
RC=$?

```
sdm is happy and the pi is happy, so I'm happy

gitbls, 2024-05-20:
Nice! I'll add a line-length check to the bootconfig plugin to help avoid errors, especially since "extra" characters are just ***ignored*** by the system.

tinker2much, 2024-05-20:
Or, to cut down on the number of lines in my script, I can do this? 

```
--plugin   copyfile:"$FANPARMS420" --plugin bootconfig:"$FANPARMS421" --plugin bootconfig:"$FANPARMS422" \
--plugin bootconfig:"$FANPARMS423" --plugin bootconfig:"$FANPARMS424" --plugin bootconfig:"$FANPARMS425" \
```

gitbls, 2024-05-20:
I would expect that to work.

Don't know if it's of interest to you or not, but you could also set up plugin @files, one per fan profile. For instance, here's fanprofile1.plugins:
```
bootconfig:dtparam=fan_temp0=55000,fan_temp0_hyst=3000,fan_temp0_speed=75
bootconfig:dtparam=fan_temp1=62500,fan_temp1_hyst=3000,fan_temp1_speed=128
bootconfig:dtparam=fan_temp2=70000,fan_temp2_hyst=3000,fan_temp2_speed=192
bootconfig:dtparam=fan_temp3=77500,fan_temp3_hyst=3000,fan_temp3_speed=255
```
and then use `--plugin @fanprofile1.plugins`. 

Keeps the command lines simpler at the expense of having more configuration files. I use plugin lists for my server configs:
```
apt-cacher-ng:tunnelenable=true
postfix:relayhost=mail.mydomain.com|mailname=mydomain.com|domain=mydomain.com|rootmail=myemail@somewhere.com
logwatch:config=/rpi/etc/logwatch|sendfrom=ps-logwatch<myname@outlook.com>|sendto=bls<myemail@somewhere.com>
ndm:localsrc=/l/work/ndm|config=/rpi/pisrv1/etc/dbndm.json
/l/work/mydomain/host-plugin:host=pisrv1
```

gitbls, 2024-06-02:
sdm 12.2 checks for config.txt lines too long. Closing this issue. Please re-open or start a new issue if required.

closed 2024-06-02 by gitbls