gitbls/sdm #212: Console colors are completely broken after sdm run

ykharuzhy, 2024-05-30:
I use Konsole and colors are not restored after run.

[image]

gitbls, 2024-05-30:
Thanks for trying sdm. Sorry to hear that you're having an issue.

Unfortunately, the problem you are seeing with `konsole` appears to be due to a deficiency in the konsole terminal emulator. From my testing, konsole returns the correct FG and BG colors, but does not return the cursor color. Further, when sdm presents the appropriate colors to konsole for restoring, they are not honored. 

This code works correctly when using the `xterm` terminal emulator and others, which properly and completely honor the color escape sequences.

To avoid this unpleasant mis-colorization sdm will be updated to enable the console colorizing to be disabled using a specific  setting (`--mcolors 0`). In this case sdm will not modify the terminal colors and they will remain at whatever they are prior to running sdm.

Addenda: Curiously, the following bash function (which you can add to your .bashrc if desired) does work on konsole. But this is the same code that sdm uses, and I've verified that sdm is doing the right thing. I don't understand why it works with this function and not in sdm, but the above workaround will be implemented in lieu of spending time chasing down the root cause, which is likely to be some konsole oddity.
```
function xtset() {
    #
    # set xterm color. can specify 3 sets of pairs (e.g., fg yellow bg black cursor red)
    # $1=fg|bg|cursor
    # $2=color to set
    # if 'xtset reset' is used, the terminal colors are reset to white bg and black fg
    # Colors are listed in /etc/X11/rgb.txt (or /usr/share/X11/rgb.txt)
    #
    function xtsetx() {
        case "$1" in
            fg) printf "\e]10;${2}\a"
                ;;
            bg) printf "\e]11;${2}\a"
                ;;
            cur*) printf "\e]12;${2}\a"
                  ;;
            *) echo "Usage: xtset fg|bg|cur*sor color..."
               ;;
        esac
    }
    [ "$1" == "reset" ] && xtsetx fg black && xtsetx bg white && return
    [ "$1" != "" ] && xtsetx "$1" "$2"
    [ "$3" != "" ] && xtsetx "$3" "$4"
    [ "$5" != "" ] && xtsetx "$5" "$6"
} ; declare -fx xtset
```

ykharuzhy, 2024-05-31:
Thanks. A switch to disable colorizing completely would be a good solution. Personally I use the sdm in an image building script for CI, so colors have no sense there.

gitbls, 2024-05-31:
> Thanks. A switch to disable colorizing completely would be a good solution. Personally I use the sdm in an image building script for CI, so colors have no sense there.

Are you using the `--batch` switch on your customize command? sdm won't colorize the terminal and operates non-interactively to completion.

ykharuzhy, 2024-05-31:
Yes, i use --batch but sdm uses colors anyway.

gitbls, 2024-05-31:
Could you provide the (appropriately redacted) command string that you are using? It shouldn't be coloring with --batch.

Thx

ykharuzhy, 2024-06-03:
Yes, I added --batch later. I rechecked with unmodified sdm (I disabled color changing in the code before) — you are right, with --batch background color is changed to blue during of phase 1 but restored successfully.

gitbls, 2024-06-03:
OK. `--batch` should preclude the color from changing at all. I've modified the code in sdm to honor `--batch` so it won't change at all.

This will be in the next update to sdm later this week.

gitbls, 2024-06-07:
`--ecolors 0` and `--mcolors 0` added in V12.3. Please test/verify. Thx

gitbls, 2024-07-03:
No activity so closing issue.

closed 2024-07-03 by gitbls