gitbls/sdm #213: Unable to logon to newly imaged NVMe

CordThomasLAS, 2024-05-31:
I ran the following two commands, then updated the boot order and rebooted my RPi 5

I am unable to log on using either of the user accounts I added (not to worry, the password will change)

Two questions - 

1. Can one add two users in the way I did?
2. Are there any character restrictions to the password in the user plugin? I didn't see anything documented

sudo sdm --customize --plugin user:"adduser=cord|password=m&TWq6SD#xM$6K4YunNB" --plugin user:"adduser=ansible|password=J$orK7q9Ai#Gumq%PTmU" --plugin L10n:host --plugin disables:piwiz --regen-ssh-host-keys --restart 2024-03-15-raspios-bookworm-arm64-lite.img

sudo sdm --burn /dev/nvme0n1 --hostname pi-las8 --expand-root 2024-03-15-raspios-bookworm-arm64-lite.img

The system boots up as pi-las8 but I cannot authenticate

gitbls, 2024-05-31:
With passwords like that it's no surprise that you can't login! 🤣

Seriously, the issue is that your passwords have dollar signs in them (`$`). As you may know, bash uses the dollar sign to indicate a variable.

Since you're using plugins on the command line those dollar signs are subject to bash interpretation, so your password got changed. For free!

One handy feature of the `user` plugin is the `log=/path/on/host/to/logfile` where it logs what it actually found. FYI, the log is written in Phase 0, but the actual work is done in Phase 1. A bit confusing, but once you know, you know.

I used the `log` argument and saw that your passwords were modified. Here's a snippet from the user plugin log. First entry was using your command line as is, second was after I implemented the first solution below.
```
Date                Operation        User/Group      IMG/Hostname

2024-05-31 14:37:00 Add User         cord            2024-03-15-raspios-bookworm-arm64-lite.img
                    password:m&TWq6SD#xMK4YunNB homedir:/home/cord 
                    groups:dialout,cdrom,floppy,audio,video,plugdev,users,adm,sudo,users,input,netdev,spi,i2c,gpio

2024-05-31 14:39:37 Add User         cord            2024-03-15-raspios-bookworm-arm64-lite.img
                    password:m&TWq6SD#xM$6K4YunNB homedir:/home/cord 
                    groups:dialout,cdrom,floppy,audio,video,plugdev,users,adm,sudo,users,input,netdev,sp
```
There are two solutions:
* Leave the command line as is, but quote the `$` as `\$`. This will prevent bash from mangling your passwords.
* Use `--plugfile @/path/to/pluglist` on the command line and create /path/to/pluglist with all your plugins in it:
```
user:adduser=cord|password=m&TWq6SD#xM$6K4YunNB
user:adduser=ansible|password=J$orK7q9Ai#Gumq%PTmU
L10n:host
disables:piwiz 
```
  With this, the lines are read and not subject to bash substitutions, so the `\` quotes are not needed, and in fact, will be treated as part of the password.

I use the `--plugfile @pluglist` for my own build, because it keeps the sdm command lines more manageable, and it's easier to scan the pluglist to see what's there, or not.

CordThomasLAS, 2024-05-31:
Thank you, very helpful

closed 2024-05-31 by CordThomasLAS

CordThomasLAS, 2024-06-02:
Sorry to bother, but now working on another NVMe and wanted to try plugfile option, but can't seem to get it to work. I searched the docs, checked the --help and also searched the source and can't seem to find how --plugfile is supposed to work:

sudo sdm --customize --plugfile /home/cord/plugfile.txt --regen-ssh-host-keys --restart 2024-03-15-raspios-bookworm-arm64-lite.img

gets me

sdm: unrecognized option '--plugfile'

Using sdm V12.2

gitbls, 2024-06-02:
Ooops...my bad!  It's `--plugin @/path/to/plugfile.txt` . Sorry about that! And, the file type doesn't need to be '.txt' it's just a Linux file. I use `myplugins` with no file type whatsoever.

CordThomasLAS, 2024-06-03:
Great, and I note that the plugfile has to be executable. That worked like a charm.

Cord Thomas
Legal Analysis Systems, Inc.
310.430.3724
***@***.******@***.***>

[A black background with blue letters  Description automatically generated]

gitbls, 2024-06-03:
> Great, and I note that the plugfile has to be executable. That worked like a charm.

I don't believe that it needs to be executable.  It's only read by sdm running as root, not executed.

CordThomasLAS, 2024-06-03:
Here's what I  am using

sudo sdm --customize --plugin /home/cord/plugfile --regen-ssh-host-keys --restart 2024-03-15-raspios-bookworm-arm64-lite.img

If i leave plugfile as rw, then I get this error:

? Plugin /home/cord/plugfile is not executable

gitbls, 2024-06-03:
Try this instead: 
```
sudo sdm --customize --plugin @/home/cord/plugfile  ...
```
You're missing the `@` at the front of the plugfile list path, so sdm assumes that the file is a plugin, and plugins must be executable. On the other hand, lists of plugins, denoted with `@`, are read as text files and do not need to be executable.